MCUdude/MegaCoreX #107: MegaCoreX seems not updated on PlatfromIO

tgtakaoka, 2021-01-25:
I can't update MegaCoreX to 1.0.7 on PlatfromIO. Am I doing something wrong?

```
$ platformio platform update atmelmegaavr
Platform atmelmegaavr
--------
Updating platformio/atmelmegaavr              1.3.0                              [Up-to-date]
Updating platformio/toolchain-atmelavr        1.70300.191015 @ ~1.70300.0        [Up-to-date]
Updating platformio/framework-arduino-megaavr 1.8.6 @ ~1.8.6                     [Up-to-date]
Updating platformio/framework-arduino-megaavr-megacorex 1.0.5 @ ~1.0.5                     [Up-to-date]
```

MCUdude, 2021-01-25:
Hi, great to see you back!

This is because the PlatformIO developers haven't updated the MegaCoreX package yet. There's nothing I can do about this, other than reminding them.

tgtakaoka, 2021-01-25:
I noticed this because of Event library. It's available on arduino-cli but got error on platformio :).
I was wondering if there is something like `platformio package update` to push a revised platform.
Hope they will update the MegaCoreX sometime soon.

MCUdude, 2021-01-28:
1.0.7 is now available for PlatformIO!

closed 2021-01-28 by MCUdude