MCUdude/MegaCoreX #108: Event.h timer users - the names seem inconsistent w/other users

SpenceKonde, 2021-01-29:
I was just writing an example with these (on oscillators) and at the end where I bring it back to practicality by using a prescaled clock to realized that there doesn't seem to be any consistency with the naming of the timer users in Event.h, The rest of the users list is consistent - the name of the user is the define from the header, minus the EVSYS_USER part, in lower case with a  _ between peripheral name and name of channel, and between that and the a and b if there is that. Essentially the name of the event user from the header... the way your libraries style things.

if that were continued here, we would have

```
      tca0_cnt_a      = 0x1A,
      tca0_cnt_b      = 0x1B,
#     if defined(TCA1)
      tca1_cnt_a      = 0x1C,
      tca1_cnt_b      = 0x1D,
#     endif
      tcb0_capt      = 0x1E,
      tcb0_count       = 0x1F,
      tcb1_capt      = 0x20,
      tcb1_count       = 0x21,
      tcb2_capt      = 0x22,
      tcb2_count       = 0x23,
#     if defined(TCB3)
      tcb3_capt      = 0x24,
      tcb3_count       = 0x25,
#     endif
#     if defined(TCB3)
      tcb4_capt      = 0x26,
      tcb4_count       = 0x27,
#     endif
#     if defined(TCD0)
      tcd0_input_a      = 0x28,
      tcd0_input_b      = 0x29,
#     endif
```

What we currently have is
```
      tca0_cnta      = 0x1A,
      tca0_cntb      = 0x1B,
      tca1_cnta      = 0x1C,
      tca1_cntb      = 0x1D,
      tcb0_capt      = 0x1E,
      tcb0_cnt       = 0x1F,
      tcb1_capt      = 0x20,
      tcb1_cnt       = 0x21,
      tcb2_capt      = 0x22,
      tcb2_cnt       = 0x23,
#     if defined(TCB3)
      tcb3_capt      = 0x24,
      tcb3_cnt       = 0x25,
#     endif
#     if defined(TCB3)
      tcb4_capt      = 0x26,
      tcb4_cnt       = 0x27,
#     endif
      tcd0_in_a      = 0x28,
      tcd0_in_b      = 0x29,
```
If I was inclined to change anything on purpose (this looks incidental?), I'd change the TCA ones to tcaN_event_a/tcaN_event_b, because ffs, you can do more than count them!

MCUdude, 2021-01-29:
Thanks, I'll do more of this.
The reason why I didn't do it on everything is that all these are available on all MegaCoreX targets anyway. but I do understand that this is different on the tinyAVRs, and I'm a big fan of cooperating on the source files, even though this doesn't really matter too much for MegaCoreX.

Off-topic: I've been communicating with a PlatformIO developer, and AVR-Dx support is currently on hold because the Arduino toolchain isn't up to date, and they're just distributing their binaries. The question is: Are "your" compilers tested and known to work on all platforms? And have you updated it with the latest tinyAVR pack too? If yes, then PlatformIO could be ready for tinyAVR-2 as well.

SpenceKonde, 2021-02-07:
Oh shiiiit! I remembered this message, and thought I had answetred, and came looking wondering if I'd missed a reply.... and I apparentrly hadn't ever sent my response here. 

Regarding AVR-Dx and the toolchain - The azduino3 package works on all platforms. I haven't personally tested it, but I know of people who used it and said it worked on x64 linux, x86 linux, and both 32-bit and 64-bit ARM, and mac. So that's all of them.. 

All the ATpacks in azduino3 toolchain package are updated, too. I pull em all when I do that. They are more than welcome to snag my toolchain packages and use them!. There is only one condition: They need to rehost the binaries themselves. I am currently serving them out of an AWS S3 bucket (in other words, I have to pay for bandwidth usage. Not very much, but if every platformio user suddenly started downloading them, I feel like that could get expensive in a jiffy....

Do note that tinyAVR 2-series isn't supported by megaTinyCore yet. I've got people on my ass about ATTinyCore, though, and I have more users on that old thing than the other two combined, so I need to focus on that for a while. Support for the 2-series on megaTinyCore is gonna be kind of annoying (needs completely new analogRead implementation) and guess who doesn't really care about analog. I cant say I'm very excited about the 2-series,in all honesty. About the only feature I care about is the alternate reset pin but they don't even have that for the 14-pin parts, only the 20+ pin ones..... and you lose the type D timer! I guess the pair of extra LUTs is nice too...

MCUdude, 2021-03-01:
Sorry for leaving this issue. I'm planning a new MegaCoreX release, and I want to completely finish the Event library.

Should we just agree that we change the names on the following enums`

```
tcaN_cnta -> tcaN_cnt_a
tcaN_cntb -> tcaN_cnt_b
```

I would prefer to still keep `tcb0_cnt` since I'm using `cnt` for everything that's count-related, including `tcaN_cnt_a`.

It would be nice if you could rename this in your repos too 👍

closed 2021-04-04 by MCUdude