MCUdude/MegaCoreX #109: A preventive measure against electrical short circuits.

mikrocoder, 2021-01-29:
Hello Hans,

just a precautionary question because I stumbled across this while building my Pin Lib.

Do you take care in your Pin Lib that the pins 18/19 are switched to input and the pullups are switched off when using the Wire Lib?
And when ending Wire.end the pins 22/23 are switched to input and its pullups are switched off. This was overlooked in the original code.

A preventive measure against electrical short circuits.
https://github.com/arduino/Arduino/issues/11221

Translated with www.DeepL.com/Translator (free version)

MCUdude, 2021-01-29:
This is only a potential issue on Arduino Nano Every, since PA2/PF2 and PA3/PF3 are shorted together on the PCB. The i2c interface needs pullup resistors, so it can only drive the lines low. 

The only theoretical issue is that PF2 and PF3 are forced high while PA2 and PA3 are in i2c mode. I'll push a commit that fixes this for the Nano Every.

closed 2021-01-29 by MCUdude

mikrocoder, 2021-01-29:
Hello Hans,

I was in the wrong twi.c.
Have corrected my code suggestion in the link above.

MCUdude, 2021-01-29:
I did realize this, so if you look at the latest commit, it should be correct

commit 2415f241 (2021-01-29): Disable PF2and PF3 on Nano Every
... on TWI init because these are hardwired to the i2c pins. Closes #109