MCUdude/MegaCoreX #113: Odd behaviour

miguelwon, 2021-03-23:
Hi MCUdude,

I'm having odd behavior with an ATMega4809 when trying to program it via UPDI. I'm able to burn a bootloader via UPDI and using an Arduino Nano (Nano+4.7K+10uF). In that conditions, I can program the ATMega4809 with FTDI. 
The problem is when I try to program directly with UPDI, i.e., with the same setup used to burn the bootloader but uploading with `no bootloader` option and `JTAG2UPDI` as a programmer. Everything compiles and no error is shown, but the program is not being uploaded. I'm testing with a simple blink example. This is the log:

```
/Applications/Arduino 2.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 2.app/Contents/Java/hardware -hardware /Users/miguelwon/Library/Arduino15/packages -hardware /Users/miguelwon/Documents/Arduino/hardware -tools /Applications/Arduino 2.app/Contents/Java/tools-builder -tools /Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -tools /Users/miguelwon/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 2.app/Contents/Java/libraries -libraries /Users/miguelwon/Documents/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=48pin_standard,resetpin=reset,BOD=2v6,clock=internal_16MHz,bootloader=no_bootloader -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460 -warnings=none -build-cache /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_cache_685124 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino 2.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 2.app/Contents/Java/hardware -hardware /Users/miguelwon/Library/Arduino15/packages -hardware /Users/miguelwon/Documents/Arduino/hardware -tools /Applications/Arduino 2.app/Contents/Java/tools-builder -tools /Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -tools /Users/miguelwon/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 2.app/Contents/Java/libraries -libraries /Users/miguelwon/Documents/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=48pin_standard,resetpin=reset,BOD=2v6,clock=internal_16MHz,bootloader=no_bootloader -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460 -warnings=none -build-cache /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_cache_685124 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board '4809' from platform in folder: /Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4
Using core 'coreX-corefiles' from platform in folder: /Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4
Detecting libraries used...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/cores/coreX-corefiles/api/deprecated -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/cores/coreX-corefiles -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/variants/48pin-standard /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/sketch/Blink.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/cores/coreX-corefiles/api/deprecated -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/cores/coreX-corefiles -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/variants/48pin-standard /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/sketch/Blink.ino.cpp -o /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
"/Applications/Arduino 2.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/cores/coreX-corefiles/api/deprecated -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/cores/coreX-corefiles -I/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/variants/48pin-standard /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/sketch/Blink.ino.cpp -o /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_cache_685124/core/core_6eb6bfbec527c59c671f15260da7d45e.a
Linking everything together...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x0 -mmcu=atmega4809 -o /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.elf /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/sketch/Blink.ino.cpp.o /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/../arduino_cache_685124/core/core_6eb6bfbec527c59c671f15260da7d45e.a -L/var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460 -lm
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.elf /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.eep
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.elf /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.hex
chmod +x /Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/scripts/create_disassembler_listing.sh
/Users/miguelwon/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/scripts/create_disassembler_listing.sh "/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-objdump" "--disassemble --source --line-numbers --demangle --section=.text" /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.elf /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.lst
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-size" -A /var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.elf
Sketch uses 776 bytes (1%) of program storage space. Maximum is 49152 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 6140 bytes for local variables. Maximum is 6144 bytes.
/Users/miguelwon/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/miguelwon/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf -v -patmega4809 -cjtag2updi -P/dev/cu.usbserial-1440 -Uflash:w:/var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.with_bootloader.hex:i -Ufuse0:w:0x00:m -Ufuse1:w:0x54:m -Ufuse2:w:0x01:m -Ufuse4:w:0x00:m -Ufuse5:w:0xC9:m -Ufuse6:w:0x06:m -Ufuse7:w:0x00:m -Ufu [... cut ...]
avrdude: reading input file "0xC5"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xC5:
avrdude: load data lock data from input file 0xC5:
avrdude: input file 0xC5 contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.
```

MCUdude, 2021-03-23:
That's strange! What happens if you re-arrange the avrdude command, so that the last thing that happens is that the hex file is written?

Here's your Avrdude command, but slightly modified where the flashing happens at the end.
```
/Users/miguelwon/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/miguelwon/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf -v -patmega4809 -cjtag2updi -P/dev/cu.usbserial-1440 -Ufuse0:w:0x00:m -Ufuse1:w:0x54:m -Ufuse2:w:0x01:m -Ufuse4:w:0x00:m -Ufuse5:w:0xC9:m -Ufuse6:w:0x06:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Ulock:w:0xC5:m -Uflash:w:/var/folders/vp/2f2lk6gj16v00znk4lc6yrkw0000gn/T/arduino_build_755460/Blink.ino.with_bootloader.hex:i
```

miguelwon, 2021-03-23:
I found that the command works if I replace `Blink.ino.with_bootloader.hex:i` with `Blink.ino.hex:i`

When I compile the sketch with Arduino IDE, two .hex files are generated (and saved in `arduino_build_755460` folder) : `Blink.ino.with_bootloader.hex` and `Blink.ino.hex`. 

I think the problem is that for some reason when I upload the sketch, Arduino IDE tries to write `Blink.ino.with_bootloader.hex:i`, instead of `Blink.ino.hex:i`, although I have defined the option `no bootloader`.

mcuee, 2022-10-16:
> The problem is when I try to program directly with UPDI, i.e., with the same setup used to burn the bootloader but uploading with no bootloader option and JTAG2UPDI as a programmer (I'm using Arduino IDE 1.8.13).

I am not so sure how this can happen. Maybe an IDE bug which may have been fixed.
Ref:
* https://github.com/MCUdude/MegaCoreX/issues/156#issuecomment-1196536701

mcuee, 2023-05-12:
I think this issue can be closed.

closed 2023-05-12 by MCUdude