MCUdude/MegaCoreX #114: #include <SD.h> doesn't compile

bigginsking, 2021-03-27:
Hi,
Trying to interface with an SD card and the SD library prevents the code from compiling. Here is the verbose output from Arduino 1.8.7:
`

In file included from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,

                 from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/SdFat.h:29,

                 from C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:20,

                 from C:\Users\Bill\Documents\Arduino\Test4809\Test4809.ino:2:

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:424:5: error: 'DDRD' was not declared in this scope

   {&DDRD, &PIND, &PORTD, 0},  // D0  0

     ^~~~

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:424:5: note: suggested alternative: 'VDD'

   {&DDRD, &PIND, &PORTD, 0},  // D0  0

     ^~~~

     VDD

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:424:12: error: 'PIND' was not declared in this scope

   {&DDRD, &PIND, &PORTD, 0},  // D0  0

            ^~~~

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:424:12: note: suggested alternative: 'PIN0'

   {&DDRD, &PIND, &PORTD, 0},  // D0  0

            ^~~~

            PIN0

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:425:5: error: 'DDRD' was not declared in this scope

   {&DDRD, &PIND, &PORTD, 1},  // D1  1

     ^~~~

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:425:5: note: suggested alternative: 'VDD'

   {&DDRD, &PIND, &PORTD, 1},  // D1  1

     ^~~~

     VDD

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:425:12: error: 'PIND' was not declared in this scope

   {&DDRD, &PIND, &PORTD, 1},  // D1  1

            ^~~~

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:425:12: note: suggested alternative: 'PIN0'

   {&DDRD, &PIND, &PORTD, 1},  // D1  1

            ^~~~

            PIN0

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:426:5: error: 'DDRD' was not declared in this scope

[... 480 lines cut ...]
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:443:5: note: suggested alternative: 'DD0'

   {&DDRC, &PINC, &PORTC, 5}   // C5 19

     ^~~~

     DD0

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:443:12: error: 'PINC' was not declared in this scope

   {&DDRC, &PINC, &PORTC, 5}   // C5 19

            ^~~~

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:443:12: note: suggested alternative: 'PIN0'

   {&DDRC, &PINC, &PORTC, 5}   // C5 19

            ^~~~

            PIN0

exit status 1
Error compiling for board ATmega4809.
`

MCUdude, 2021-03-28:
Hi!
You're using an old IDE version where the bundled SD library doesn't support the ATmega4809. Try updating to the latest IDE.

bigginsking, 2021-03-28:
Ah, yes, that fixes it, amateur here... thanks!

closed 2021-03-28 by bigginsking