MCUdude/MegaCoreX #115: Comparator library cannot invert without outputting to a pin.

SpenceKonde, 2021-03-31:
While porting the comparator library I realized that the invert bit in the mux register is controlled by the output setting! This means that the library cannot be used to produce and event with inverted output unless the output pin is also used. Is there a reason for this seemingly arbitrary limitation? I would think that the invert bit should be controlled by it's own boolean property - and that any inverting specified by the output setting would be implemented separately from inverting the event (by setting INVEN on the corresponding PINnCTRL register (which IMO would be excessive for this library)

MCUdude, 2021-04-08:
Whaaat? So this means that if you use the output of the comparator internally, and don't enable the output, the comparator can't invert its own "internal output"?

Is this documented somewhere (errata) or did you find it by coincidence?

SpenceKonde, 2021-04-08:
No no, the hardware supports it, but your library doesn't - the invert bit is controlled by the output field, and there's no option for invert without enabling output.... 
```
  enum output_t : uint8_t {
    disable = 0x00,
    enable  = 0x40,
    invert  = 0x80,
  }; 

````

MCUdude, 2021-04-08:
Ah, I see! I'll come up with a solution for this. Another thing I just realized was lacking was a way to read the state of the comparator without reading the physical output pin or triggering off an interrupt.

``` c++
Comparator.read(); // Read output of comparator, internally
```

How about re-doing the output_t enum to something like this?

``` c
  enum output_t : uint8_t {
    disable = 0x00,
    disable_invert = 0x80,
    enable  = 0x40,
    enable_invert  = 0xC0,
    invert  = 0xC0,
  }; 
```

BTW It would be nice to have a Zoom meeting with you soon! Perhaps to figure out if there are other libraries that need to be ported or created, further cooperation between cores, hardware/UPDI programmers, etc. Let me know if you have time sometime!

closed 2021-04-08 by MCUdude

SpenceKonde, 2021-04-09:
Oh yeah, we should do that soon, I think that's a good idea. 

I'm still operating on pandemic schedule, so can make almost any time as long as I don't need to leave the house. And my sleep schedule.... basically isn't a schedule. 

Oh - definitely check on the boards you m,aintain comparator for whether you even need to set the pin output.. On Dx and tiny, you do not!

MCUdude, 2021-04-09:
> Oh yeah, we should do that soon, I think that's a good idea.
> I'm still operating on pandemic schedule, so can make almost any time as long as I don't need to leave the house. And my sleep schedule.... basically isn't a schedule.

I'm usually available in the evenings. I know this might be a little too short of a notice, but how about today/tonight, 10pm my time, 4pm your time?

SpenceKonde, 2021-04-09:
Yeah, that works for me., I would like to catch you up on some of what I'd been working on (and be called out if what I'm doing is foolish :-P )

MCUdude, 2021-04-09:
Excellent, mail me a Zoom link, and I'll see you in ~2.5 hours! I'll write down a few things we'll have to discuss

commit 3f20e21b (2021-04-08): Add output enable and disable invert
Closes #115