MCUdude/MegaCoreX #120: Clock accuracy

NilsMinor, 2021-04-14:
Hey, 

I have a question on the different clock settings for the 4808 :) I am using an atmega4808 to measure a low frequency on an input pin (the code is here https://github.com/MCUdude/MegaCoreX/issues/91) with TCB1 configured. I use this information to control an PWM output (Pin PIN_PD2 ) that drives a motor. This system is closed so when driving the motor faster, I get more pulses. The pulses have a frequency of 25 to 35 Hz.

I have some problems in the accuracy and I ask if it is possible that this is correlated to the internal clock as it might be better to use an external clock? And what is the most accurate clock selection of the internal clocks? Currently I am driving the system with the internal 20 MHz clock. Will the accuracy increase if I use an external clock? Thanks for any suggestion :)

Best regards,
Nils

MCUdude, 2021-04-14:
Hi Nils!

There are two internal clocks in the 4808 you can use. One 20MHz and one 16MHz, that can be divided down. They _can_ be tuned by writing to calibration register, (CLKCTRL.OSC20MCALIBA, I haven't tried it), but the value usually varies between chips and batches, so an external clock source is almost always more accurate.

If you have a signal generator on your bench you can try to connect this to the CLKI input pin on your chip to see if the accuracy gets any better.

NilsMinor, 2021-04-14:
Hi @MCUdude,

thanks again for your help with the 4808 component. Do you think the clock accuracy of the internal 20MHz clock can result in an error of the input pluses measurement even at this low frequency of 25 - 35 Hz ? For the future we will use an external clock source.
Best regards,
Nils

MCUdude, 2021-04-14:
I'm not sure how much "inaccuracy" you see. The internal oscillator is usually off by a maximum of a couple of percent. This will only give you a couple of percent of measuring error at 25 - 36Hz

NilsMinor, 2021-04-15:
Yeah you are right, I will use an external oscillator in the future :) Thanks for your help :)

closed 2021-04-15 by NilsMinor