MCUdude/MegaCoreX #122: Reference to avr\ioXXX.h file

robmaxtech, 2021-05-07:
Hello, I'm using the MegaCoreX together with VisualMicro, so I get all the nice and comfortable Intellisense support of Visual Studio. I found that the PORTx, PINx_bm and similar definitions are not visible for VisualMicro, therefore the items are highlighted as "identifier not defined"; although the code compiles fine.
When I #include "...\Arduino\hardware\tools\avr\avr\include\avr\iom1608.h" (I'm using the ATMEGA1608 currently) in my INO file, the definitions are no more marked as error; but when compiling, I get the error "Attempt to include more than one <avr\ioXXX.h> file." 
There is a definition _AVR_IOXXX_H_ that avoids duplicate definition in every ioxxx.h file, but I don't understand where it gets initially defined when using a specific core. 

The problem doesn't exist with other cores, like original arduino core or e.g. the MiniCore.

OFC I could comment/uncomment the #include, but this is simply a bit annoying. Any idea how to get around this? Thanks!

MCUdude, 2021-05-07:
Hi!

I'm not sure how well Visual Micro works with this core. I've never tried it, and would recommend you to use VSCode + PlatformIO instead where you'll also have Intellisense.

What happens if you try to include <avr/io.h>?

```c

// Either this
#include <avr/io.h>

// Or this
#ifndef _AVR_IO_H_
#include <avr/io.h>
#endif
```

robmaxtech, 2021-05-07:
Thanks! Both includes don't work unfortunately; <avr/io.h> isn't found...
Just being curious: Where is the avr\ioXXX.h file included in general? Is this in the scope of a core or a general arduino thing?
I'll give VScode a try; I have seen it when it came out, but didn't follow it how the AVR integration has developed.

robmaxtech, 2021-05-08:
Thanks for pointing me to VS Code + PlatformIO! Installed it yesterday and had it up and running really quick! This is my future dev platform - it works great, with Intellisense, and the integration of the MegaCore is perfect. Thanks again!

closed 2021-05-08 by robmaxtech

MCUdude, 2021-05-08:
Great to hear you like it! I use PlatformIO for all my serious projects. I've put a lot of time and effort into making MegaCoreX PlatformIO the experience as nice as possible. Did you know you calculate/set fuses, burn bootloader and upload with [Pyupdi](https://github.com/mraardvark/pyupdi) using PlatformIO? It's quite neat!

robmaxtech, 2021-05-09:
Yes, I saw the fuse calculation and other features; excellent! Currently I use jtag2upti for uploading, together with a selfmade programmer based on the info of Spence Konde. Works like a charm also within platform.io! I definitely want to try also the simplified programmer with only the USB-to-serial chips; IMO it's important to spread the info so all Arduino fans can make use of the new ATMEL chips too. They are so great!