MCUdude/MegaCoreX #123: SPI Question

mikrocoder, 2021-06-19:
Hello,

I have here [https://forum.arduino.cc/t/houston-spi-has-a-problem-with-avr-gcc-11-1-0/876291](url) described here pure problem in connection with avr-gcc 11.1.0 and the Arduino SPI Lib. With your package SPI works with avr-gcc 11.1.0. If I compare your SPI lib with the Arduino lib they are clearly different. Was there a specific reason for this? Were the problems already known before?

Thanks.

MCUdude, 2021-06-19:
The MegaCoreX SPI library was forked off the official repo sometime in April 2019. Since then I haven't done much to it. I've not used anything newer than avr-gcc 7.3.0, so I have no idea why it works with avr-gcc 10.3 but not 11. You should open an issue over at [the official megaavr core](https://github.com/arduino/ArduinoCore-megaavr).

Here are all the commits I've pushed to the SPI library since then:
https://github.com/MCUdude/MegaCoreX/commits/master/megaavr/libraries/SPI/src

Here are all the commits pushed to the official library:
https://github.com/arduino/ArduinoCore-megaavr/commits/master/libraries/SPI/src

The only major difference I can see is that I've never adopted the newer Arduino API.

closed 2021-06-19 by MCUdude

mikrocoder, 2021-06-19:
Okay. Hopefully this will help me.  ;-)  Thanks in advance.