MCUdude/MegaCoreX #124: issue with Atmel ICE in UPDI mode

zakaster, 2021-07-04:
download success with `Atmel ICE` in `UPDI` mode but the program itself doesn't work
### environment
MCU: atmega1608 28pin
IDE: VScode + Platformio (both latest)
programmer: Atmel ICE

### platformio.ini
```
[platformio]
; Default build target
default_envs = Upload_UPDI
; default_envs = Upload_UART
; default_envs = fuses_bootloader

; Parameters used for all environments
[env]
platform = atmelmegaavr
framework = arduino

; Chip in use
board = ATmega1608
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L
; Oscillator type (internal or external)
board_hardware.oscillator = internal
; Arduino pinout variant
board_build.variant = 28pin-standard

; Unflag build flags
build_unflags =
; Extra build flags 
build_flags =

; Monitor port is auto detected. Override here
;monitor_port = 
; Serial monitor baud rate
monitor_speed = 9600

; Run the following command to upload with this environment
; pio run -e Upload_UPDI -t upload
[env:Upload_UPDI]
; Upload protocol for UPDI upload
; upload_protocol = xplainedmini_updi
upload_protocol = atmelice_updi
upload_flags =

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
; Upload protocol for serial uploads (using Optiboot)
upload_protocol = arduino
upload_flags =
; upload_port = COM5

; run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader
[env:fuses_bootloader]
; Upload protocol for used to set fuses/bootloader
upload_protocol = ${env:Upload_UPDI.upload_protocol}
upload_flags =
; Hardware settings
board_hardware.bod = 2.7v
board_hardware.eesave = yes
board_hardware.uart = no_bootloader
board_hardware.rstpin = reset 
```
### program
``` C
#include <Arduino.h>

void setup()
{
    Serial.begin(9600);
}

void loop()
{
    Serial.printf("hello, time = %ld\n", millis());
    delay(1000);
}
```
### compiling message with `Upload_UPDI`
```
Processing Upload_UPDI (platform: atmelmegaavr; framework: arduino; board: ATmega1608)
--------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/ATmega1608.html
PLATFORM: Atmel megaAVR (1.4.0) > ATmega1608   
HARDWARE: ATMEGA1608 16MHz, 2KB RAM, 16KB Flash
PACKAGES:
 - framework-arduino-megaavr-megacorex 1.0.8
 - toolchain-atmelavr 2.70300.201015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\Upload_UPDI\src\main.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\UART.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\UART0.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\UART1.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\UART2.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\UART3.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\UNO_compat.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\USBCore.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\WInterrupts.c.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\abi.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\api\Common.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\api\IPAddress.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\api\PluggableUSB.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\api\Print.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\api\RingBuffer.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\api\Stream.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\api\String.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\hooks.c.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\main.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\new.cpp.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\wiring.c.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\wiring_pulse.S.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\Upload_UPDI\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\Upload_UPDI\libFrameworkArduino.a
Linking .pio\build\Upload_UPDI\firmware.elf
Building .pio\build\Upload_UPDI\firmware.hex
Checking size .pio\build\Upload_UPDI\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.9% (used 161 bytes from 2048 bytes)
Flash: [===       ]  25.5% (used 4183 bytes from 16384 bytes)
============================================== [SUCCESS] Took 2.97 seconds ==============================================

Environment    Status    Duration
-------------  --------  ------------
Upload_UPDI    SUCCESS   00:00:02.974
============================================== 1 succeeded in 00:00:02.974 ============================================== 
```

### problem
everything works fine with platformio, no issue with compile and download
but simply serial-to-usb receives no incoming message (led doesn't blink and monitor receives no incoming data)
tried with `Atmel Studio` + `Atmel ICE` buring the same hex file, its the same problem
however if I change to use `default_envs = Upload_UART` in `platformio.ini` and burn it using `Atmel Stduio` + `Atmel ICE`, then everything works, I realize the hex file is somehow different from the one using `default_envs = Upload_UPDI`
tried dozens of times and its always the same, 
let me know if it helps to attached the hex file, many thanks

MCUdude, 2021-07-04:
Hi!

The "problem" with the newer megaAVR-0 series compared to the classic AVRs is is that the bootloader is located in the bottom part of the flash memory instead of on the top. This means that a hex file has to be compiled to make room for a bootloader at the bottom of the flash.

Can you try to run the following command:
`pio run -t fuses -e fuses_test_1`

and then upload using your Atmel ICE UPDI programmer?
`pio run -t upload -e Upload_UPDI`

Also, please paste this into your platformio.ini file:

```ini
[env:fuses_test_1]
; Upload protocol for used to set fuses/bootloader
upload_protocol = ${env:Upload_UPDI.upload_protocol}
upload_flags =
; Hardware settings
board_hardware.bod = 2.7v
board_hardware.eesave = yes
board_hardware.uart = no_bootloader
board_hardware.rstpin = reset 

[env:fuses_test_2]
; Upload protocol for used to set fuses/bootloader
upload_protocol = ${env:Upload_UPDI.upload_protocol}
upload_flags =
; Hardware settings
board_hardware.bod = 2.7v
board_hardware.eesave = yes
board_hardware.uart = uart0
board_hardware.rstpin = reset 
```

zakaster, 2021-07-04:
```
PS D:\projects\purifier display\firmware> pio run -t fuses -e fuses_test_1
Processing fuses_test_1 (platform: atmelmegaavr; framework: arduino; board: ATmega1608)
----------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/ATmega1608.html
PLATFORM: Atmel megaAVR (1.4.0) > ATmega1608   
HARDWARE: ATMEGA1608 16MHz, 2KB RAM, 16KB Flash
PACKAGES: 
 - framework-arduino-megaavr-megacorex 1.0.8   
 - tool-avrdude-megaavr 2.60300.210128 (6.3.0) 
 - toolchain-atmelavr 2.70300.201015 (7.3.0)   
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode

TARGET CONFIGURATION:   
------------------------
Target = atmega1608
Clock speed = 16000000L
Oscillator = internal
BOD level = 2.7v
Save EEPROM = yes
Reset pin mode = reset
------------------------

Selected fuses:
------------------------
[fuse0 / WDTCFG  = 0x00]
[fuse1 / BODCFG  = 0x00]
[fuse2 / OSCCFG  = 0x01]
[fuse4 / TCD0CFG = 0x00]
[fuse5 / SYSCFG0 = 0xC9]
[fuse6 / SYSCFG1 = 0x06]
[fuse7 / APPEND  = 0x00]
[fuse8 / BOOTEND = 0x00]
[lfuse / LOCKBIT = 0xC5]
------------------------

Setting fuses...

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e9427 (probably m1608)
avrdude: reading input file "0x00"
avrdude: writing fuse0 (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of fuse0 written
avrdude: verifying fuse0 memory against 0x00:
avrdude: load data fuse0 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse0 data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of fuse0 verified
avrdude: reading input file "0x00"
avrdude: writing fuse1 (1 bytes):

Writing | ################################################## | 100% 0.01s

[... 201 lines cut ...]

avrdude: 1134 bytes of flash written
avrdude: verifying flash memory against .pio\build\Upload_UPDI\firmware.hex:
avrdude: load data flash data from input file .pio\build\Upload_UPDI\firmware.hex:
avrdude: input file .pio\build\Upload_UPDI\firmware.hex contains 1134 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.06s

avrdude: verifying ...
avrdude: 1134 bytes of flash verified

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.

=========================================== [SUCCESS] Took 3.26 seconds ===========================================

Environment    Status    Duration
-------------  --------  ------------
Upload_UPDI    SUCCESS   00:00:03.260
=========================================== 1 succeeded in 00:00:03.260 ===========================================
```

wow dude, it seems to be working now, but i'm totally confused, your `env:fuses_test_1` is exactly the same as the default `env:fuses_bootloader` and where's the magic?

MCUdude, 2021-07-04:
To me, it looks like the ATmega1608 had the fuses set for bootloader use. You can always try using `env:fuses_bootloader` but the result should be the same. Unlike classic AVRs, you'll have to set the right fuses in order to upload using a programmer.

Does both these work now?

Using a programmer:
``` bash
# Set fuses
pio run -t fuses -e fuses_test_1
# Upload using Atmel ICE
pio run -t upload -e Upload_UPDI

# Set fuses and burn bootloader (change to something else than uart0 if needed)
pio run -t bootloader -e fuses_test_2
#Upload using USB to serial adapter
pio run -t upload -e Upload_UART
```

zakaster, 2021-07-04:
yes, tested both, work like a charm, thank you very much for helping dude.

closed 2021-07-04 by zakaster