MCUdude/MegaCoreX #125: Thinary Nano Every support

asferreira, 2021-07-17:
Thinary has a Nano Every "clone" board based on ATmega4808. There is a package in https://github.com/Thinary/ThinaryArduino that adds support to these boards to the Arduino IDE and it seems to work fine.
Unfortunatelly, I wasn't able to use it in PlatformIO. I've tried to use the MegaCoreX ATmega4808 and I could build and upload the firmware but it didn't seem to work properly.
Is it possible to add support to this board in the MegaCoreX package?  

Thanks in advance.

SpenceKonde, 2021-07-17:
I agree, it would be really nice to have support for those available from a
proper core. Thinary is the best of the chinese clone manufacturers. Their
nano clones have micro USB, and they make ones with 328PB's in addition to
the normal P's......

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

MCUdude, 2021-07-17:
@asferreira does the Thinary 4808 work with MegaCoreX in Arduino IDE? You say the firmware didn't seem to work properly. Have you tried to select your preferred settings in the IDE menu and ran "Burn Bootloader"? This sets the correct fuses. It might be related to the bootloader fuse bits being set when they shouldn't.

asferreira, 2021-07-17:
No, it doesn't work in Arduino IDE also. 
In my tests I created a simple sketch that blinks the builtin led. It worked using the thinary package found in github on the Arduino IDE, but it didn't work with MegaCoreX on Arduino IDE or PlatformIO. I rewrote the sketch to use port registers instead of the Arduino PIN number and the program did work with MegaCoreX on both platforms. I also managed to work with the serial (USB) monitor, which was accessible using Serial2. 
The only setting I had to change was the programmer, which I set to JTAG2UPDI.

SpenceKonde, 2021-07-18:
That sounds to me like it just uses a different pin mapping.... ?

asferreira, 2021-07-18:
Indeed! Although the pin's labels printed on the board match with original Arduino Nano Every board mapping, they are mapped to different pins in the MCU. I just figured it out after looking into the pin_arduino.h file in the Thinary's github repo.

MCUdude, 2021-07-18:
What happens if you use PIN_PC2 as the pin number, and not 13?

MCUdude, 2021-07-19:
I've pretty much finished implementing the Thinary Nano 4808 pinout. However, I'd like to re-map the serial numbering so it mimics the Nano Every better. I've ordered a Nano 4808, but it will take several weeks before it arrives.

I found this Nano 4808 pinout over at the Avrfreaks forum:
[image]

I'd like to remap the serial pins to something like this. Thoughts?

| Arduino serial port | Hardware serial port | Pins | Pinswap
|--------------------|-----------------------|----------|-----|
| `Serial`                    | `HWSERIAL2`             | PF0 + PF1 | PF4 + PF5
| `Serial1`                  | `HWSERIAL1`            | PC0 + PC1 |  |
| `Serial2`                 | `HWSERIAL0`        | PA0 + PA1 | PA4 + PA5 |

```c
/*
  pins_arduino.h - Pin definition functions for Arduino
  Part of Arduino - http://www.arduino.cc/

  Copyright (c) 2007 David A. Mellis

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General
  Public License along with this library; if not, write to the
  Free Software Foundation, Inc., 59 Temple Place, Suite 330,
  Boston, MA  02111-1307  USA
*/

#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <avr/pgmspace.h>
#include "timers.h"

#define MEGACOREX
#define MCUDUDE_MEGACOREX

#define NANO_4808_PINOUT
#define NONA4808_PINOUT
#define __AVR_ATmegax08__

// Arduino pin macros
#define PIN_PF0 0
#define PIN_PF1 1
#define PIN_PA0 2
#define PIN_PA1 3
#define PIN_PA2 4
#define PIN_PA3 5
#define PIN_PA4 6
#define PIN_PA5 7
#define PIN_PA6 8
#define PIN_PA7 9
#define PIN_PC3 10
#define PIN_PC0 11
#define PIN_PC1 12
#define PIN_PC2 13
#define PIN_PD0 14
#define PIN_PD1 15
#define PIN_PD2 16
#define PIN_PD3 17
#define PIN_PF2 18
#define PIN_PF3 19
[... 315 lines cut ...]
// These serial port names are intended to allow libraries and architecture-neutral
// sketches to automatically default to the correct port name for a particular type
// of use.  For example, a GPS module would normally connect to SERIAL_PORT_HARDWARE_OPEN,
// the first hardware serial port whose RX/TX pins are not dedicated to another use.
//
// SERIAL_PORT_MONITOR        Port which normally prints to the Arduino Serial Monitor
//
// SERIAL_PORT_USBVIRTUAL     Port which is USB virtual serial
//
// SERIAL_PORT_LINUXBRIDGE    Port which connects to a Linux system via Bridge library
//
// SERIAL_PORT_HARDWARE       Hardware serial port, physical RX & TX pins.
//
// SERIAL_PORT_HARDWARE_OPEN  Hardware serial ports which are open for use.  Their RX & TX
//                            pins are NOT connected to anything by default.
#define SERIAL_PORT_MONITOR       Serial
#define SERIAL_PORT_HARDWARE      Serial1
#define SERIAL_PORT_USBVIRTUAL    Serial

void initVariant() __attribute__((weak));
void initVariant() { }

#endif

```

asferreira, 2021-07-19:
I’ve been using the same pinout picture as reference to my tests with this board. It seems to be correct, so far!
I have no objection regarding the serial pins remapping you proposed. In fact, the remapping defined in the Thinary’s package is quite weird, I didn’t get used with the serial monitor at Serial1.

MCUdude, 2021-07-20:
@asferreira I've just [pushed a branch with support for the Nano 4808](https://github.com/MCUdude/MegaCoreX/tree/nano-4808). Can you test it and see if UART, i2c, and SPI work as expected? I don't have any hardware to test with just yet.

[image]

asferreira, 2021-07-20:
I’ve manually installed the zip file from the branch nano-4808 on the Arduino IDE, but the nano 4808 board hasn’t showed up in the board list. I also noticed that the Arduino Nano Every board hasn’t appeared either.
The only boards available in the menu are: ATmega4809, ATmega4808, ATmega3209, ATmega3208, ATmega1609, ATmega1608, ATmega809, and ATmega808.

asferreira, 2021-07-20:
Forget about my last email. I’ve discovered I’ve done wrong. The Nano 4808 appears in the pinout menu for the ATmega4808. Sorry for the mistake.
I’ll run some tests and will return with the results soon!

asferreira, 2021-07-21:
I’ve just finished some basic tests and the board worked fine for most of them. Serial and I2C are working fine. Just one annoying thing about the serial interface it that when your program uses the serial monitor, it becomes very difficult to upload a new firmware to the board. I had to reset the board many times and try to upload again until avrdude could write the code. Unfortunately it seems to be expected due to the architecture of the us/serial interface used on the board.
I’ve also observed some strange behaviors that I’m not sure if they were expected or not:
   - The pins 21, 22, 23, and 24 didn’t  worked with digitalWrite() or analogWrite(), but they do work using the port registers directly.
   - While reading or changing the state of the pins 18 through 20, I noticed that the state was changed on the following pin. For instance, when I set pin 18 to high, nothing happened to it, instead, was the pin 19 that went to high.  As it happened before, this behavior didn’t happen when using port registers.
 

Is there any specific test you would like to be done? Let me know If you need something else!

MCUdude, 2021-07-21:
There was a typo in the timer pin mapping that may have caused the weird behavior you experienced. Can you pull the latest change and try again?

> Just one annoying thing about the serial interface it that when your program uses the serial monitor, it becomes very difficult to upload a new firmware to the board. I had to reset the board many times and try to upload again until avrdude could write the code. 

I've seen users complain about this elsewhere too. It's just how they have implemented the jtag2updi firmware. Right now, it's only possible to upload to the nano 4808 using jtag2updi through Arduino IDE. I should probably make it possible to choose any programmer since the UPDI prin is exposed anyways. If I were to design the board, I'd probably just remove the jtag2updi chip altogether and use a serial bootloader just like on classic Nanos.

MCUdude, 2021-07-21:
I think I've solved the issue where some pins were offset. But does `Serial.print` print to the serial monitor, or do you have to use Serial1 or Serial2?

> I had to reset the board many times and try to upload again until avrdude could write the code. Unfortunately it seems to be expected due to the architecture of the us/serial interface used on the board.

The Nano 4808 pinout variant is using JTAG2UPDI as the default programming option. If you want to use a different programmer for it, connect one to the UPDI pin and upload using the programmer. You can quite easily DIY your own microUPDI programmer if you have an on Leonardo, Micro or Pro Micro lying around.

asferreira, 2021-07-21:
I’ve downloaded and installed the updated version of the nano-4808 branch and now all pins are working. Serial monitor is mapped as Serial object. I didn’t need to use Serial1 or 2.

MCUdude, 2021-07-21:
Awesome! I'll wait for my board to arrive to do some testing myself, and then I'll merge the branch into master.

asferreira, 2021-07-21:
Great! 

Thank you very much!

SpenceKonde, 2021-07-21:
I gotta say,the products of Thinary seem put up a mighty fight to snatch a resounding defeat from the jaws of what could have been a glorious victory.

MCUdude, 2021-08-01:
I just got my Nano 4808 in the mail, and everything seems to work correctly according to the pinout. I'm closing this issue and merging the nano-4808 branch into master.

tresvdoble, 2021-11-07:
The servo.h library does not work with this card, is it planned to put this library that is compatible with the 4808?

adelin-mcbsoft, 2022-11-17:
I'd like to make a suggestion for this particular board (Thinary MK II), which I've been testing lately and I'm quite satisfied with.
So all in all, the board performs as expected with MegaCoreX.

The only (known) hardware bug, is that you cannot write it, if the Serial writes data to the USB faster than 1 second without delay.
Which means that, unless you have a separate programmer (_other than the JTAG2UPDI present on board_) to write it, you could consider the board virtually bricked if you made the mistake to write to Serial continuously.

As per my previous experience with other boards, I know that you have to pull-up or down a GPIO to enable writing mode (for example, on Espressif Chips you have to pull-down GPIO_0 to enter in writing mode).
Though I didn't dig further (yet), I'd expect that this board also expects some GPIO up or down for writing.

So I was thinking that this could be very helpful, as the Serial class could be modified in such a way, that it does not print/write (output) anything as soon as it finds that a specific GPIO (the one that enables writing in our case) has its state modified (up or down, as needed).

This way, there would be no output to the Serial USB anymore as soon as the GPIO state changed, and the board can be successfully written. No more bricked boards.