MCUdude/MegaCoreX #126: Event system and Arduino Pin numbers

aikopras, 2021-08-17:
I started playing with the Event System to decode manchester coded input. I configured the Manchester coded input pin as Event generator, and TCB0 as Event user. The timer is configured to raise an interrupt, after which the ISR reads the input signal. 

Things work very well, and using the Event System is (thanks to the provided library) quite simple.

There is one thing that (for me) is a bit annoying, and that is that the Event System / Library expects as input Port and Pin numbers, such as gen0::pin_pa0. For me it would be easier if instead of pin_pa0 the "standard" Arduino Pin number (2) could be used.

The question is thus if/how this is possible. I know that I could write some switch or if statements to create such mappings, but that solution would be variant specific.

MCUdude, 2021-08-20:
Great to hear you find the Event system useful and the library easy to use!

I'm planning to rewrite part of the Event library in cooperation with @SpenceKonde, and Arduino pin numbers are one thing that will be implemented. The other things that need to be done are mostly about easier integration, so other libraries easier can build upon the Event library.

@SpenceKonde has used the library more than I have (even though I'm the author), so he will be providing a list of necessary changes.

More info here: 
https://github.com/SpenceKonde/megaTinyCore/issues/480

MCUdude, 2021-08-20:
After looking a bit deeper into to the "Arduino pin as generator" issue, we can't just pass an Arduino pin number to set_generator(), because the object(Event0...Event7) has to support that pin as a generator. A solution to this is to let the library select the object for you, like so.

```c++
// Use Arduino pin PIN_PA0 as event generator, and let the library find
// the lowest channel number that supports PA0 as event generator
Event& myEvent = Event::set_generator_pin(PIN_PA0);

// Set PF2 as event user (yet to implement Arduino pin
// numbers for users
myEvent.set_user(user::evoutf_pin_pf2);

// Start Event generator
myEvent.start();

// Check channel number in use, just for fun
Serial.printf("PIN_PA0 uses Event channel %d\n", myEvent.get_channel_number());

// Check object in use
if(&myEvent == &Event0)
  Serial.printf("PA0 uses Event channel 0");
```

aikopras, 2021-08-21:
First I would like to thank you (as well as SpenceKonde) for the fantastic things that you do. I can give many examples where your work helped me a lot, both with older processors, like the Mega16 (MightyCore) as well as the newer processors (MegaCoreX, DxCore).

I'm also happy to see that I'm not the only one with this request, and that SpenceKonde and you are considering rewriting the Event library.

Yes, I do realise that channels and ports are tight together, which indeed requires the library to decide which event channel will be used for a given pin. To a certain extend the library user needs to be aware of that, and understand that (like for example USARTS) there is not complete freedom in choosing any pin you like. 

To be honest, I find the pin numbering for Arduino boards (like the Nano Every) quite confusing, but the numbering you propose for the (48, 40, 32 and 28) pin variants much more logical. 

PS: Are SpenceKonde and you also considering to change the Logic library such that Arduino pin numbers can be used?

MCUdude, 2021-08-21:
Thanks for the kind words! I've been doing this for a while now, and I've definitely learned a lot from it! In my opinion, it's very liberating to know that pretty much any hardware that has an AVR can be programmed with Arduino using either mine or SpenceKonde's cores.

> To be honest, I find the pin numbering for Arduino boards (like the Nano Every) quite confusing, but the numbering you propose for the (48, 40, 32 and 28) pin variants much more logical.

Yes, the Uno Wifi Rev2 and especially the Nano Every has a pretty wild pinout because they try to make them "as ATmega328P compatible" as possible. What still surprise me is that they have not at all provided all the features the Nano Every really has to offer. The official core only implements two serial ports, but the Every has four. The Nano Every also has way more analog inputs than the official Arduino core provide. Since I personally never really use Arduino shields, I don't rely on pinouts that mimics the functionality of ATmega328P based Arduinos. That's why I've made all the pinouts as straight forward as possible, and why Spekce Konde followed the same exact pattern as me with DxCore.

> Yes, I do realise that channels and ports are tight together, which indeed requires the library to decide which event channel will be used for a given pin. To a certain extend the library user needs [...]

If you look at the Event system in the datasheet, you'll pretty quickly realize that hretty much every application that utilizes the Event system has to be talor made to even get the Event system to work. Writing a high level library that would magically just take care of everything is pretty much impossible, especially since the Event system is so quirky. Arduino as event generators and event users makes total sense, and is absolutely possible to implement in a straight-forward and logical way. However, I'm not so sure it's that easy when it comes to all the other peripherals like timer, logic etc. But I'm open to suggestions on how it could be implemented!

> PS: Are SpenceKonde and you also considering to change the Logic library such that Arduino pin numbers can be used?

I'm not sure I'll add Arduino pins to the Logic library, and it would be way more work that with the Event library, since none of the input pins can be moved around. My experience is that if you want to use the Logic hardware you'll first have to figure out what you want to accomplish, then pick one or more logic block to use, and you're pretty much stuck with the pinout and output pins the block(s) has.

aikopras, 2021-08-21:
I'm not sure if I should better react here or on SpenceKonde's issues page.
Anyway, at that page the question was raised about what requirements a modified Event library should satisfy. I  can't say what others want to do, but maybe my ideas help. If not, please ignore.

Context: I'm building (as hobbyist and open source) decoders for modeltrains. Such decoders often rely on the DCC standard for receiving messages. The DCC message is manchester biphase encoded. Signal transitions (a pin interrupt) take place roughly every 100 microseconds, after which the pin ISR starts a timer which fires 75 microseconds later (the second Interrupt) to read the signal's value. Of course other decoding approaches are possible.
 
The event system can be quite useful to connect the pin Interrupt to the timer.
The advantage is not only better / more predictable performance, but it saves the first ISR that gets triggered after the signal transition on the input pin. The saving of an ISR is particularly important with the new megaX and DX core processors, since their interrupt system was changed (from one interrupt per INTx pin at the 328 and others to one interrupt per port). SpenceKonde has a nice description of this [https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/PinInterrupts.md,](url). If I measured well this new interrupt system takes (using attachInterrupt()) roughly 6 microseconds to get started, compared to roughly 1 microsecond with the 328/2560). For my application the overhead that these two ISRs generate every 100 microseconds is substantial.

However, things get worse since I also have a "feedback bus", where a master polls all (128) slaves. One polling puls takes roughly 100us, and after each puls a slave has to update its pulse counter. 

So this brings me to a third interrupt every 100 microseconds. Evey 100 microseconds I seem to loose roughly 15 to 20 microseconds to the overhead of three standard attachInterrupt() routines. Thus for me the real motivation to use the event system is to get at least similar interrupt efficiency with the new cores as with the old cores. 

Of course I could replace attachInterrupt() and write my own pin interrupt routines. The danger of doing that, however, is that if the user of my library does use attachInterrupt(), a conflict occurs. For me a much cleaner approach seems to let the Event system generate an interrupt, and write myself an Event interrupt routine instead. For me this sounds much safer then replacing attachInterrupt().
Instead of the Event system also the Logic library seems interesting, since it seems that also the CCL can be used to generate interrupts for which I can write the ISR handling routine myself. So for me the CCL seems interesting to avoid attachInterupt().

I hope this helps.

aikopras, 2021-08-21:
> Thanks for the kind words! 

For me it is fantastic what people like you do for the community!!! 

> it's very liberating to know that pretty much any hardware that has an AVR can be programmed with Arduino using either mine or SpenceKonde's cores.

Yes, and the cores provided by you and SpenceKonde are better than the standard cores.
Btw, one wish that I still have for some older cores of you: it would be nice to have Fast IO also there.  

> Arduino as event generators and event users makes total sense, and is absolutely possible to implement in a straight-forward and logical way. However, I'm not so sure it's that easy when it comes to [...]

Is your question how to connect timers etc. to the Event Library? But that seems trivial: `Event2.set_user(user::tcb0)`. What more is needed? For many people the challenge will be the writing of the timer initialisation code. There are some libraries out there, but my feeling is that many of them focus on PWM generation. If you want something simple, like the case I described in my previous post, I just needed single shot mode.

SpenceKonde, 2021-08-21:
Hans - YES EXACTLY! 

This is what I have been saying since I first commented on thls library around when it got merged into DxCore....   and which I started shouting much more loudly about since I made that issue, because that's the first time I tried to actually use it to write some *INCREDIBLY BASIC* event system code in a non-part-specific way to make one of the simplest use cases happen) and realized that it simply wasn't possible without ugly brute force solutions if I wanted the code to run everywhere. 

In order to be able to do anything in the general case, you cannot use the Event# objects, because you don't know which one to use! (even if you did, there is no way short of a switch case statement or lookup table to get from the integer 1 to the Event object named "Event1"), the only way this shit is gonna fly is if the library is capable of handing us an event channel with the specified properties. 

I strongly disagrtee with the claim that this is a fools errand because the EVSYS peripherals are different from eachother or something. With the exception of the tinyAVR 0/1-series (aka "public beta"), and the way that 2-series pins wrap around while the 48/64 pin parts just have some second class channels that can't do any pin everts, they're all same exact thing, just different lists of users and generators.
Getting rid of the "how do I go from a pin to an EventN object?" barrier would get rid of one of the big things that is keeping people from using this new functionality. Anyone wanting to do anything that requires an event channel needs to solve this problem first or make a single-part-only library, and this problem is a much bigger deal to solve than the rest of what they'd probably want to do with the event channel. And so unsurprisingly nobody has written libraries that use event channels to do awesome things, or even to do lame things. 

The reason Arduino isn't making their hardware work to it's full potential is nothing new. The mega2560 can do differential ADC.... with 200x gain on some pin combos! But Arduino doesn't give you any facilities for that (My cores do, as it happens, because I knew if I didn't do something about it, it would never happen for 2-series, and 2-series is all about the ADC). The Arduino people have *never* have done that kind of thing - that's what the community contributes :-P . They are not gifted electrical engineers nor competent embedded developers. They made an "good enough" IDE with an incredibly easy learning curve. which is a really hard UX nut to crack, and they have that board manager system to install toolchains. I know of plenty of people installing the IDE to get board manager *just to install toolchains* that they call using other methods - that's another hard thing they've dcone. And. and they do an incredibly good job of outreach and evangelism. But their hardware has always been of mediocre design (and it hasn't gotten better), and their cores, well.... there's a reason we have our own copy of the core ;-)

aikopras, 2021-08-21:
> The reason Arduino isn't making their hardware work to it's full potential is nothing new

Yes, I completely agree. Seems that Arduino, and many of its developers, go for "generic code", which runs more or less on each and every platform, ranging from Arduino Uno to 32 bit RISC processors and ESP. The performance cost of making everything "generic" may be acceptable on high end processors, but killing for slower processors, such as the ATMega's. By not taking advantage of the novel functions in these newer ATMega processors, we get none of their advantages (like the Event system) but all of the disadvantages (like the performance penalty of attachInterrupt()). Why would someone design new hardware for a 4809 or 128DA, if the performance at the end will be lower.

It would be great if some "standard" libraries would be there to use novel functions, such as the Event library.

MCUdude, 2021-08-21:
I'm about to push my current work to a new branch so you guys may have a chance to look at it. The code compiles and runs fine, and the README is also updated with the new functions. I'll have to at least create one new example to illustrate some of the new functionality, but it's already getting pretty late here.

We can now let the library choose a suitable Event channel for you, and Arduino pins can now be used as event generators and users. The following code compiles and works fine:
```c++
  Event& myEvent = Event::set_generator_pin(PIN_PA0);
  myEvent.set_user_pin(PIN_PC2);
  myEvent.set_user_pin(PIN_PD2);
  myEvent.start();
```

> Btw, one wish that I still have for some older cores of you: it would be nice to have Fast IO also there.

I've looked into adding `digitalWriteFast()` and `digitalReadFast()` to my classic AVR cores, but I haven't been able to make it compile down to a single instruction without basing the whole thing on a ton of macros. I'd much rather prefer to use the pins_arduino.h pin mapping if possible.

> Is your question how to connect timers etc. to the Event Library? But that seems trivial: Event2.set_user(user::tcb0). What more is needed? For many people the challenge will be the writing of the [...]

It is trivial to pass user::tcb0, but Spence mentioned something about passing the TCB0 struct itself, not just a constant like user::tcb0 is. I'm not sure how that could be implemented in a straight-forward way.

@SpenceKonde; functionality vise, is there anything obvious you think is missing now?

The updated library can be found here:
https://github.com/MCUdude/MegaCoreX/tree/New-Event-library/megaavr/libraries/Event

aikopras, 2021-08-24:
Hi Hans

I've just been playing with it a while, and it works very nice. For my application the code is now just a few lines:

```
  Event& myEvent = Event::set_generator_pin(15);
  myEvent.set_user_pin(5);
  myEvent.set_user(user::tcb0);
  myEvent.start();
```

I only need to know the Arduino pin numbers, which nicely integrates with the remainder of my library code.
What I also enjoy that the Event system reacts on inverting the pin input (using PINnCTRL).

Thanks a lot, this really helps to make my code nicer and smoother. I hope others will enjoy this as well; attached my sketch as example.

If I have additional questions or feedback I'll let you know.
Thanks, Aiko

```
/***********************************************************************|
| megaAVR event system library                                          |
|                                                                       |
| Simple_event_with_timer.ino                                                      |
|                                                                       |
| A library for interfacing with the megaAVR event system.              |
| Developed in 2021 by MCUdude                                          |
| https://github.com/MCUdude/                                           |
|                                                                       |
| In this example we use the following pins and timers:                 |
| - PD2 (Arduino pin 15): event generator                               |
| - PB2 (Arduino pin 5): event user (follows PD2's state)               |
| - TCB0: event user                                                    |  
| - PIN_PE0: In main loop to indicate if the timer is running           |
| - Pin PE1: Set and cleared by the timer                               |                  |
|                                                                       |
| See Microchip's application note AN2451 for more information.         |
|***********************************************************************/
#include <Event.h>

#define TIME_US 50    // Time (in microseconds) before the timer triggers

void init_timer(void) {
  noInterrupts();
  // Step 1: Calculate from TIME_US the right value for TOP
  #define TOP F_CPU / 1000000  * TIME_US
  // Step 2: fill the registers.
  // We use single shot mode
  // Reset the main timer control registers
  // This may be needed since the Arduino core creates some presets
  TCB0.CTRLA = 0;
  TCB0.CTRLB = 0;
  TCB0.EVCTRL = 0;
  TCB0.INTCTRL = 0;
  // Initialise the control registers 
  TCB0.CTRLA = TCB_ENABLE_bm;                    // Enable the Timer/Counter type B peripheral
  TCB0.CTRLB = TCB_CNTMODE_SINGLE_gc;            // Single-Shot mode
  // Event system: enable input capture events
  // Start counter only on positive edge
  // Enable noise cancelation
  TCB0.EVCTRL = TCB_CAPTEI_bm | TCB_FILTER_bm;
  // TCB0.EVCTRL = TCB_CAPTEI_bm | TCB_EDGE_bm;
  // Enable CAPT interrupts by writing a ‘1’ to the ENABLE bit in the Interrupt Control (TCBn.INTCTRL) register.
  TCB0.INTCTRL |= TCB_CAPT_bm;
  // Write a TOP value to the Compare/Capture (TCBn.CCMP) register.
  TCB0.CCMP = TOP;
  interrupts();
}

void setup() {
  // Pins for debugging
  pinMode(PIN_PE0, OUTPUT);          // Timer is running
  pinMode(PIN_PE1, OUTPUT);          // Is set and cleared within timer ISR
  digitalWriteFast(PIN_PE0, LOW);    // Initialise
  digitalWriteFast(PIN_PE1, LOW);
  init_timer();
  //
  // Event system, with Arduino pin numbers and an unknown event number
  // PD2 = Arduino pin 15
  // Uncomment to invert the input pin (TCB starts on falling edge)
  // PORTD.PIN2CTRL |= (1 << PORT_INVEN_bp );     // Invert the Interrupt
  Event& myEvent = Event::set_generator_pin(15);
  myEvent.set_user_pin(5);
  myEvent.set_user(user::tcb0);
  myEvent.start();
}

// The ISR should be called explicitly
ISR(TCB0_INT_vect) {
  TCB0.INTFLAGS |= TCB_CAPT_bm;      // We had an interrupt. Clear!
  digitalWriteFast(PIN_PE1, HIGH);
  digitalWriteFast(PIN_PE1, LOW);
}
    
void loop() {
  // PE0 indicaqtes if TCB0 is running
  digitalWriteFast(PIN_PE0, TCB0.STATUS);
}

```

MCUdude, 2021-08-24:
@aikopras Thanks for taking the time to test out the revised Event library.
Great to hear the latest revision makes it easier to integrate with Arduino projects.

But by looking at your code, it's pretty clear to me that a TCB library also would have been nice to have; written "in the same flavor" as the Event library. 

I'll wait for @SpenceKonde's feedback before merging the updated library into master though. He promised he would get back to me about this.

aikopras, 2021-08-25:
> But by looking at your code, it's pretty clear to me that a TCB library also would have been nice to have; written "in the same flavor" as the Event library.

Sounds quite interesting. Despite the complaints that others sometimes have, in my opinion the new TCB timers have some interesting features that the old timers didn't have. This is even more true for the DxCore. 
Can you share any ideas how this may look like?

aikopras, 2021-09-14:
I've been working now for a while with the new Event Library on 4808 and 4809 chips, and I must say that I really like it. It is extremely simple, but therefore very powerful. Since I'm including it in one of my libraries, I would enjoy if this new event library will be included in the next release of MegaCoreX.

I've also tried to "blindly copy" this library to the DxCore, but that didn't work. I guess it is something simple, but I didn't dive deeper. What I noticed, however, is that the Timer Event users on the AVR DA and DB series have a slightly different name: instead of tcb* it is now tcb*_capt (there is also a new and interesting tcb*_cnt). To enable portability between both cores, wouldn't it be an idea to have a #define that works on both cores?

Thanks a lot for developing this great library!!

MCUdude, 2021-09-14:
@aikopras great to hear you like the library! The plan is to merge it into master, but I was _really_ hoping that @SpenceKonde could provide some feedback and/or suggestions on how to further improve the library. Ideally, the new library will be ported to megaTinyCore and DxCore, so a user could expect the same features regardless of which core is being used.

@SpenceKonde have you lost interest? You're not responding to anything I write these days...

SpenceKonde, 2021-09-14:
No I have not lost interest, I just had made a rule to not work on anything else of consequence until I had a usable version of megaTinyCore in the 2.4;x versions out,and which I expected would be my main project after ordering my june order of PCBs (placed in august), Both of those got shoved out of the way when a freelance client needed some super urgent work done.. Anyway. I figured that it would take at most a few weeks to release 2.4..since we had just gotten SerialUPDI working with lhe latest optimizations.  but here we are in september, with that goal having just finally been achieved. This is on my list of things to work on., I want to get the new modernAVR Wire library squared away in the github version of megaTinyCore (two issue to clarify with author, and minor change that makes no difference on tinyAVR. but I want (really, I need) to be able to directly copy files between the cores. rather than going through some porting process. That update adds both Wire1 support and support for using both master and slave simultaneously (either in dual mode if supported or on the same pair of pins), and does everything in less flash (but more ram if you want master+slave - since the master transaction methods just queue up stuff, rather than actually changing the bus state,so , nothing precludes, in such a multimaster system,there being a slave TWI event between beginTransaction and endTransaction and hence you can't share the buffer). This is a change that people have been asking for in both cores both via email and github, I don't really understand WHY people want it so badl, but it is a very common request which I was totally out of my league trying to implement). Considering the history of Wire.h library implementationsi in my cores thee is need for a ton of real world testing, so I need to get the new wire library in the github version of both cores (megaTinyCore's will be in now, but DxCore gets one "safe" release with a few bugfixes, before the release that will actually include the new Wire library just like 2.4.2 mTC. So I'll deal with those bugs, then I release and put a version that I am confident in out for users,  But that is a change that I want in the hands of people as long as possible before it is in a release both on megaTinyCore and DxCore) because it is intended to be a drop-in, smaller replacement., yet CI tells us nothing about its correctness, and there is so much hardware and so many interactions that could break in so many subtle ways....So I want to get the completed version for megaTinyCore (which will naturally be ready to drop into next DxCore release at that point, assuming (it's author has done some testing at leas.) 

After that I get to put a little time into modern AVRs which is where I hope to get  Event library examination in at long last, and fix a couple of other hot issues like the analogWrite problem and squeeze out a DxCore patch, that fixes a few important bugs and adds the expanded device suipport top Serial UPDI, potentuially updated event depending on how it's looking, release, check in new Wire, and run to ATTinyCore and get 2.0.0 (80% done I think) fucking released - it's a huge restructuring of the core on every level., that core existed under my auspices before I had clue what hell I was doing, and it's bad in so many stupid ways. I have owed them a release for 9 months now , and for most of that time all the pins_arduino files have been redone, there's sa new implementation of and people who only use ATTinyCore have been starting to wonder what happened to me. And it's most recent release (as well as every one prior to that) not something I'm particularly proud of. (I was considerably worse than I am now then)..

Have you been watching the features announced on recent releases of DxCore and megaTinyCore? There is a bunch of stuff that I imagined you would want to steal for mega 0-series parts.... tuned oscillators not just for accuracy but to over/underclock when precision of ext clock not required. , Runtime changable PWM pins for TCA (I didn't bother to figure out some way to implement it for the apparently randomly scattered TCB's if it's even possible with acceptable overhead. Considering what crapola timers they are it probabnly isn't)., then analogReadEnh() (enhancedAnalogRead() and the functions associated witjh that), the for for the bug with long tones at high frequency,. and the bug with delayMicros() screwing up when the function gets inlined (unless you;ve modified it, it relies on the call and return overhead to get the time right (also, I don't think it was updated on official cores to account for the slightly altered timing of call, nor does it cover a bunch of releant frequencies that are suddenly within reach through tuning.

MCUdude, 2021-09-19:
In short terms, The Event library is very similar to the previous version, but it now supports Arduino pins in as generators and users, and you can get hold of the right object by either asking for a channel number, event user or event generator. This is what makes the library flexible.

@SpenceKonde Wire/i2c issues? I haven't got any Github issues regarding this. Am I just lucky, or is this something that only affects the tiny0/1/2's? I'm usually _very_ hesitant to replace "backbone libraries" like the Wire library is. If it breaks, a load of Github issues will pile up! But it it's worth it, of course!

> Have you been watching the recent releases of DxCore and megaTinyCore? 

Yes, I've regularly had a look at them from time to time. There are absolutely nice features I should get around to implement, but I do disagree on the implementation part of some of them. The ADC is a powerful and "important" peripheral, and all the "advanced" features could IMO be split out in a library instead, perhaps a static one for improved efficiency and code size. 

Have you experienced any issues regarding all the pin related compile time checks? I implemented this in MicroCore a while ago, but [I had to remove it](https://github.com/MCUdude/MicroCore/commit/bcfb8a4459350eb7d43347459bdcf9d61cfa97ff#diff-676d8782c776d073ad27c87da7d8585d583e78686f5b64069628d496f0d01458) due to the compiler being difficult.

SpenceKonde, 2021-10-19:
Finally having a chance to look at this this week for 1.3.7 DxCore, now that I've finished with a huge blob of fixes and improvements over there (take a look at what I did to the boards.txt - the file is B E A U T I F U L now (the initial impetus was to make editing easier with text editors that have the "minimap" scrollbar functionality.

You haven't been getting complaints about I2C?!! The baud rates were totally fucked, still. We went through two complete rewrites of the master baud rate calculation since the last time that came up and we were in communication about it, before arriving at one that works over on my cores. I think that brings the total rewrite count for baud rate calculation up to at least 5. 

I think at least two of the 5 open issues about I2C on my cores are user error, as the totally new implementation and the old one had the same behavior. One thing that is very clear is that people are not very good at debugging I2C or knowing how to approach I2C issues. The maximally ungraceful response to the bus being hung up by something else always directs attention to the core (nobody ever thinks to measure the voltage on the SCL and SDA lines - Isn't that the first thing you do? Then you unplug the one that's  being held in the wrong state from one side orr the other, and observe that it jumps back up to normal on the Arduino side while the other device continues holding it low, and conclude "Crap, I must be doing something that's making the device unhappy"). 

I also was *constantly* getting requests for dual mode and master+slave functionality. Do you not get those? @MX682X has blessed the Arduino community with what is essentially a complete rewrite of Wire with drastically reduces compile size, AND does dual mode and master+slave which is gonna be killer.  

I will give a big long general update via email tonight, and will also be looking at this

As for the compiletime pin checks - for functions that do NOT require it (like the fast digital I/O) I only do the check if the pin is compiletime known (do you have LTO enabled? You must have LTO enabled for these checksto work reliably, apparently; on my modern cores I *only* support LTO. ATTinyCore 2.0.0 will also force it on)). Except for really clumsy and stupid bugs, nope, no problems at all with them. They have worked beautifully. Every time I have a chance to recognize input is constant and that it's bad, I will badCall() it and turn it into a compile error. 

My philosophy is always "anything that can be recognized as guaranteed not to work at compile time, I want it to be a compile error, because we have no exception handling here, and debugging takes 10-20x longer than fixing a compile error. 

Also, we've gotta figure out a consistent way to format the files - of course, that's "holy war" territory, so that we can easily merge changes between them and see what was changed. It would be **awesome** to end up with a single version of Event for all the parts that support it instead of one version for Dx, one version for tiny, and one for mega zero series. I wound up reformatting the copy I downloaded so I could diff it. 

Can I ask what the comments before every function are, or rather what the spec is for it and how yhey're used? It looks likethere's some sort of auto-documentation generated fromthem, but I don't know what it is.

MCUdude, 2021-11-28:
@SpenceKonde I'm still waiting for your response

aikopras, 2021-12-04:
Whereas the new library works great on MegaCoreX, I can't get the DxCore version running. On DxCore I had to restore to the event.h and event.cpp files from March, to get at least some event functionality running. The newer versions (including from November 12) no longer seem to compile. 

@SpenceKonde: Although the issue doesn't relate to MegaCoreX but to DxCore, I post it here to keep the discussion at 1 place. I hope that is OK.

SpenceKonde, 2021-12-04:
Well, this is the next project since I've got the serial changes working on
both cores now

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2021-12-06:
And now, at 2:50 AM on sunday night, I've finally gotten the last of the serial changes in.... this is the last thing to get wosettled.

MCUdude, 2021-12-06:
@SpenceKonde are you happy with the current state of the Event library, functionality-wise? If yes, I'll probably just create a few more examples and merge the whole thing into master.

SpenceKonde, 2021-12-06:
I just finished implementing tinyAVR 0/1-series version of assign_generator, assign_generator pin, and I think some other place where I had a TODO comment, and added the DD-series defines and fixed a few copy-paste errors, and it now compiles on Dx and. I have *no idea* if it actually works though. 

https://github.com/SpenceKonde/DxCore/blob/master/megaavr/libraries/Event/src/Event.cpp

SpenceKonde, 2021-12-06:
Updated to fix some issues that came up on megaTinyCore (note that it requires latest version of megaTinyCore for that to work)

SpenceKonde, 2021-12-06:
This is the sort of thing I was thinking about for other peripherals.... 

```
gen::generator_t Event::gen_from_peripheral(&TCB_t timer, uint8_t event_type) {
  uint8_t addr = (uint8_t)(uint16_t) &timer;
  #if !defined(MEGATINYCORE) || MEGATINYCORE_SERIES == 2
    addr >> 3; //now it's 0, 2, 4, 6, or 8. 
    #if !defined(DXCORE)
      if (eventType > 1) 
    #else 
      if (event_type)
    #endif
      addr = 0;
    else {
      addr+= event_type & 0x01; // 0x00 = capt, 0x01 = ovf
      addr += 0xA0;
    }
  #else 
    /* megatinycore */
  #endif
  return (gen::generator_t) addr;
}
user::user_t Event::user_from_peripheral(&TCB_t timer, uint8_t user_type) {
  uint8_t addr = (uint8_t)(uint16_t) &timer;
  #if !defined(MEGATINYCORE)
    #if defined(DXCORE)
      addr >> 3; //now it's 0, 2, 4, 6, or 8. 
      if (eventType > 1) 
        addr = 0; // invalid user requested.
      else {
        addr += event_type & 0x01; // 0x00 = capt, 0x01 = ovf
        #if defined(__AVR_DA__)
          addr += 0x1F;
        #else
          addr += 0x1E
        #endif
      }
    #else 
      if (event_type)
        addr = 0;
      else {
        addr >> 4; //now it's 0, 1, 2, 3, or 4. 
        addr += 0x13; 
      }
    #endif
  #else 
    // handle tinyavr
  #endif
  return (user::user_t) addr;
}
```
and you shoudl be able to cover different kinds of peripherals like this, right? 

```
    static gen::generator_t gen_from_peripheral(&TCB_t timer, uint8_t event_type);
    static gen::generator_t gen_from_peripheral(&AC_t comp, uint8_t event_type);
    static gen::generator_t gen_from_peripheral(&TCA_t timer, uint8_t event_type);
    static gen::generator_t gen_from_peripheral(&ZCD_t zerocross, uint8_t event_type);
    static gen::generator_t gen_from_peripheral(&CCL_t logic, uint8_t event_type);
```

SpenceKonde, 2021-12-06:
Okay, I've got versions in that pass compile-test checked in for DxCore and megaTinyCore now for what I think are the important ones.

MCUdude, 2021-12-06:
I'll review your addition and merge the changes into MegaCoreX, and do some chip-spesific changes if necessary. You should probably add some example sketches and add them to the README as well.

SpenceKonde, 2021-12-06:
Unless you object, I propose bumping the version number in library.properties to 1.2.0.

MCUdude, 2021-12-07:
Sounds good! I'll spend some time this evening. BTW why pass the structs as pointers and not references?
If a reference is used, the user can use `Event::gen_from_peripheral(TCB3, some_event_type);` instead of `Event::gen_from_peripheral(&TCB3, some_event_type);`.

BTW would you like to explain what the `event_type` parameter is and how the user should know what to insert there? To be honest, it has been a while since I've worked with the Event system implementation, and it's something that easily slips off your mind when you don't use it regularly because it tends to be a bit tricky to wrap your head around.

SpenceKonde, 2021-12-07:
I've just committed what I hope is final (not sure I'll have time for examples)

I updated the API reference in the docs though.

event_type and user_type are just for peripherals that have more than one available - like how a TCA can generate something 5 different events and on Dx and 2-series, consume two different kinds., and the CCLs are all one peripheral, There's now a table in the documentation too. 

I noticed my note in the docs about the long_soft_event; It was trivial enough to implement that I did that too - it's not particularly complicated as far as asm goes, and I wanted to make sure I had the option to do at least 2 clocks of continuously asserted software event, since you want to be able to force an event through the filter on a CCL, and you can't do that with a normal softevent, no matter what, because the compiler would render it as sts, a 2-word 2-clock instruction. Even if you got 2 in a row, the soft_event would be halfway to the next write, and I generally want to be able to hold a soft-event for at least long enough for it to not just make it through the filter, but still be in the original state after it comes out the other side of the filter

I also added the USART to the list, just for getting the event input(I'm pretty sure you can use that to remap any event channel (and hence any pin) to a serial port's RX. I didn't have a chance to try it out yet (was too busy with the rest of the serial stuff.  (The serial implementation I'm using now supports half duplex UART, RS485 mode, and (via a tiny bit of manual effort, or use of a helper macro) USART (clocked) or MSPI mode.), and I made up for the size penalty by reimplementing the ISRs in assembly to get rid of the duplicated prologue and epilogues that save and restore registers that the function doesn't use, same trick I used on attachInterrupt).

aikopras, 2021-12-07:
Great, Thanks :-)

There are a few minor issues:
1) missing semicolon at end of sentence in event.cpp (DxCore):
Line 925: `_long_soft_event(channel_number, length)` 
Line 1140:  `addr += 0x1B // DA or mega0`

2) Missing aliases
The following works on DxCore, but not MegaCoreX
  `myEvent.set_user(user::tcb1_capt);  // DxCore`

The following works on MegaCoreX, but not DxCore
  `myEvent.set_user(user::tcb1);        // MegaCoreX`

A solution could be to add some tcb related aliases to MegaCoreX.

If I find more, I let you know. Next to compiling it also actually works ;-) (at least for my libraries)

MCUdude, 2021-12-10:
@SpenceKonde I'm currently looking at your `user_from_peripheral()` and `gen_from_peripheral()`, and I don't get why you have made them so complicated. It's pretty much impossible to follow what all the hex values and bit shifts are for. Why couldn't we do something like this instead? Why all the bit shifts and stuff when we're going to return a known gen:: or user:: based purely on the device type and input parameters?

```cpp
gen::generator_t Event::gen_from_peripheral(AC_t& comp)
{  
  #if defined(__AVR_ATmegax08__) || defined(__AVR_ATmegax09__)
    return gen::ac0_out;
  #elif defined(DXCORE)
    if(&comp == &AC0)
      return gen::ac0_out;
    else if(&comp == &AC1)
      return gen::ac1_out;
    else if(&comp == &AC2)
      return gen::ac2_out;
  #endif  
}
```

MCUdude, 2021-12-11:
>Well, because I must have been doing something wrong when I was trying to do it with references, and once I made it work with pointers, I didn't go backl to references and try to figure out why it wasn't working for me., and I just didn't think of testing them directly. 

> I think in the case of the TCB'sthe sort of trickery I use could reduce code size a tiny bit, but I'm not certain, and for the parts that care (which only have 2 TCBs at most), it won't save space. 

> But yeah, that alternative never crossed my mind. If that works and is the same or smaller compiled, than by all means! 

> If that works, then the best implementation I think would be this - no part-specific knowledge is needed whatsoever. All it needs to know is if AC0, AC1 and AC2 exist, and the only assumption it makes is that for ACn,  n < 3 and there is an ACn-1 and so on.

``` c++
gen::generator_t Event::gen_from_peripheral(AC_t& comp)
{
#if defined(AC0)
  #if defined(AC1) ||  defined(AC2)
    if (&comp == &AC0) 
  #endif
    return gen::ac0_out;
  else 
  #if defined(AC1)
    if (&comp == &AC1) 
  #endif
    return gen::ac1_out;
  else 
  #if defined(AC2)
    if(&comp == &AC2)
      return gen::ac2_out;
  #endif
  else 
#endif
    return (gen::generator_t) -1; //is there a better way to indicate something that will be rejected by anything that takes a generayor?
}
```

Yes, we should probably use this implementation for all `gen_from_peripheral` and `user_from_peripheral`. Another benefit is that the code is much easier to read and understand.

I've also modified your `assign_generator()` implementation too. Why even bother trying to deal with Arduino pin numbers and channel-specific events when the function takes a gen::generator_t enum as a parameter anyways? My implementation only has one parameter and is way less complex, and still does what users expect it to do.
https://github.com/MCUdude/MegaCoreX/blob/New-Event-library/megaavr/libraries/Event/src/Event.cpp#L386

I've also done some changes to `get_generator_channel()`. Now gen::/genN:: enums and Arduino pins are handled differently. Previously it was impossible to get a reference to the channel object of an Arduino pin was set as the generator.

SpenceKonde, 2021-12-11:
Agree - if that works, that's better than my approach (though the CCL won't change much) 

I don't know what you mean by "when the function takes a gen::generator_t enum as a parameter anyways?" 

The pins aren't such a big deal; there's a proper function for that. How do you handle an "assign" for an RTC channel now though? Should I document that we do not support automatic discovery of available channels for the RTC generators? 

Okay, I'm going to adapt my gen/user from peripheral functions and merge your functions back into my copy, (you realize that my intent was that the files I was adding would be portable to megaAVR 0 as well as DxCore and megaTinyCore, right? It would be lovely if we could have a single copy of the evvent library that worked everywhere; not only is it a single copy of the library being maintained, but it enforces a uniform API - which is where most of the payoff is reaped. 

The version that I'm reconstituting will have the aliases that @aikopras requested too.

MCUdude, 2021-12-11:
> I don't know what you mean by "when the function takes a gen::generator_t enum as a parameter anyways?"

You can't pass `gen0::rtc_div8192` or anything else other than gen:: to the `assign_generator()`.

If you try to pass something else, `Event::assign_generator(gen0::rtc_div8192);`, the compiler won't let you without casting. That's why I didn't care about Arduino pins or the RTC. It's only meant for generators that can be assigned to _any_ channel, not specific ones like the RTC and digital pins requires.

> Okay, I'm going to adapt my gen/user from peripheral functions and merge your functions back into my copy, (you realize that my intent was that the files I was adding would be portable to megaAVR 0 as well as DxCore and megaTinyCore, right? It would be lovely if we could have a single copy of the event library that worked everywhere; not only is it a single copy of the library being maintained, but it enforces a uniform API - which is where most of the payoff is reaped.

It's nice to only have _one_ copy of the library, but you and I have vastly different coding styles, and I'm not really that excited about dense and somewhat cryptic code.

* I would obviously prefer my own coding style. It's not as dense, but I feel it's more readable and easier to follow. I am mainly thinking about brackets on separate lines and a nicely formatted Doxygen-styled summary above the function so that "modern" text editors can give the user a nice preview.
* Macros kills code readability. It's just a fact. I do understand that it has to be there, but we should strive to make macros as easy to follow as possible. `#if (!defined(MEGATINYCORE) || MEGATINYCORE_SERIES != 2)` is a good example of a macro that is difficult to follow and makes your brain itch. Not MEGATINYCORE or not the tiny-2 series? What is this even supposed to mean? IMO, code readability is a must, even if it makes for more verbose code.

SpenceKonde, 2021-12-11:
Well, the pins are easy enough to handle. It is awkward about the RTC though. 

I strongly dislike the opening brackets being on their own line. It's sort of strange to me that while I don't like opening brackets getting their own line, I only very rarely omit the brackets altogether (maybe the fact that it costs 0-1 line instead of 1-2 lines is part of that?), and when I do, it's because the conditional code is something particularly clear - short simple test, the code that is conditionally run is something that obviously is being run conditionally like a return statement when checking inputs for errors, where it's hard to not understand what is going on. 

And yeah, that macro is weird. And it's repeated all over the place in my code, yup. Why? Because the tinyAVR 2-series has the event system of the Dx, with only minor differences. Almost all code can be shared. I could define something... clearer than that test in one place, (which would just be defined as that but in only one place, instead of scattered everywhere. 

 The tiny0/1 on the other hand.... that's a total shitshow. Look at the struct: 

```
typedef struct EVSYS_struct
{
    register8_t ASYNCSTROBE;  /* Asynchronous Channel Strobe */
    register8_t SYNCSTROBE;  /* Synchronous Channel Strobe */
    register8_t ASYNCCH0;  /* Asynchronous Channel 0 Generator Selection */
    register8_t ASYNCCH1;  /* Asynchronous Channel 1 Generator Selection */
    register8_t ASYNCCH2;  /* Asynchronous Channel 2 Generator Selection */
    register8_t ASYNCCH3;  /* Asynchronous Channel 3 Generator Selection */
    register8_t reserved_1[4];
    register8_t SYNCCH0;  /* Synchronous Channel 0 Generator Selection */
    register8_t SYNCCH1;  /* Synchronous Channel 1 Generator Selection */
    register8_t reserved_2[6];
    register8_t ASYNCUSER0;  /* Asynchronous User Ch 0 Input Selection - TCB0 */
    register8_t ASYNCUSER1;  /* Asynchronous User Ch 1 Input Selection - ADC0 */
    register8_t ASYNCUSER2;  /* Asynchronous User Ch 2 Input Selection - CCL LUT0 Event 0 */
    register8_t ASYNCUSER3;  /* Asynchronous User Ch 3 Input Selection - CCL LUT1 Event 0 */
    register8_t ASYNCUSER4;  /* Asynchronous User Ch 4 Input Selection - CCL LUT0 Event 1 */
    register8_t ASYNCUSER5;  /* Asynchronous User Ch 5 Input Selection - CCL LUT1 Event 1 */
    register8_t ASYNCUSER6;  /* Asynchronous User Ch 6 Input Selection - TCD0 Event 0 */
    register8_t ASYNCUSER7;  /* Asynchronous User Ch 7 Input Selection - TCD0 Event 1 */
    register8_t ASYNCUSER8;  /* Asynchronous User Ch 8 Input Selection - Event Out 0 */
    register8_t ASYNCUSER9;  /* Asynchronous User Ch 9 Input Selection - Event Out 1 */
    register8_t ASYNCUSER10;  /* Asynchronous User Ch 10 Input Selection - Event Out 2 */
    register8_t ASYNCUSER11;  /* Asynchronous User Ch 11 Input Selection - TCB1 */
    register8_t ASYNCUSER12;  /* Asynchronous User Ch 12 Input Selection - ADC1 */
    register8_t reserved_3[3];
    register8_t SYNCUSER0;  /* Synchronous User Ch 0 - TCA0 */
    register8_t SYNCUSER1;  /* Synchronous User Ch 1 - USART0 */
    register8_t reserved_4[28];
} EVSYS_t;
``` 
megaTinyCore already has a bunch of #defines to rename them to match the rest of the world, but it's still wacky in lots of ways. 

But yeah - I'm writing for three different targets - tinyAVR 0/1, tinyAVR 2, and Dx-series. though tinyAVR2 and Dx-series are very nearly identical. (and mega 0-series is almost identical to Dx afaik).

On the topic of macros and readability - since you have the PIN_Pxn defines in your core too, why don't we just check for those instead of part numbers? (eg, in the pin user section). Seems clearer than checking pincounts.

MCUdude, 2021-12-11:
> I strongly dislike the opening brackets being on their own line. It's sort of strange to me that while I don't like opening brackets getting their own line, I only very rarely omit the brackets [...]

I realize now that the Logic library is already formatted in your favor, so OK, let's keep your formatting style, even though my personal preferences differ.

> And yeah, that macro is weird. And it's repeated all over the place in my code, yup. Why? Because the tinyAVR 2-series has the event system of the Dx, with only minor differences. Almost all code [...]

> On the topic of macros and readability - since you have the PIN_Pxn defines in your core too, why don't we just check for those instead of part numbers? (eg, in the pin user section). Seems clearer than checking pincounts.

Good idea! Could something like this work?

<details>

```cpp
#if defined(MEGACOREX) || defined(DXCORE)
// Features present on all generator channels
namespace gen {
  enum generator_t : uint8_t {
    disable       = 0x00,
    off           = 0x00,
    updi_synch    = 0x01,
    rtc_ovf       = 0x06,
    rtc_cmp       = 0x07,
    ccl0_out      = 0x10,
    ccl1_out      = 0x11,
    ccl2_out      = 0x12,
    ccl3_out      = 0x13,
    ac0_out       = 0x20,
#if defined(AC1)
    ac1_out       = 0x21,
#endif
#if defined(AC2)
    ac2_out       = 0x22,
#endif
    adc0_ready    = 0x24,
    usart0_xck    = 0x60,
    usart1_xck    = 0x61,
    usart2_xck    = 0x62,
#if defined(USART3)
    usart3_xck    = 0x63,
#endif
    spi0_sck      = 0x68,
    tca0_ovf_lunf = 0x80,
    tca0_hunf     = 0x81,
    tca0_cmp0     = 0x84,
    tca0_cmp1     = 0x85,
    tca0_cmp2     = 0x86,
    tcb0_capt     = 0xA0,
    tcb0          = 0xA0,
    tcb1_capt     = 0xA2,
    tcb1          = 0xA2,
    tcb2_capt     = 0xA4,
    tcb2          = 0xA4,
#if defined(TCB3)
    tcb3_capt     = 0xA6,
    tcb3          = 0xA6,
#endif
#if defined(TCB4)
    tcb4_capt      = 0xA8,
    tcb4           = 0xA8,
#endif
#if defined(__AVR_DA__) || defined(__AVR_DB__)
    ccl4_out      = 0x14,
    ccl5_out      = 0x15,
    zcd0_out      = 0x30,
    zcd1_out      = 0x31,
#if defined(ZCD2)
    zcd2_out      = 0x32,
#endif
#if defined(USART4)
    usart4_xck    = 0x64,
[... 905 lines cut ...]
  };
};
namespace gen2 {
  enum generator_t : uint8_t {
    pin_pa0      = 0x0A,
    pin_pa1      = 0x0B,
    pin_pa2      = 0x0C,
    pin_pa3      = 0x0D,
    pin_pa4      = 0x0E,
    pin_pa5      = 0x0F,
    pin_pa6      = 0x10,
    pin_pa7      = 0x11,
    updi         = 0x12,
#if (PROGMEM_SIZE > 8192 && MEGATINYCORE_SERIES == 1)
    ac1_out      = 0x13,
    ac2_out      = 0x14,
#endif
  };
};
```

</details>

Side note: This is illegal in terms of "Non-Arduino C++ rules": https://github.com/SpenceKonde/DxCore/blob/606d6c6210f74e13ba6ea296df448d0b485d8da1/megaavr/libraries/Event/src/Event.cpp#L239-L245
`assign_generator_pin(uint8_t port, uint8_t port_pin)` should be defined first, and then used in `assign_generator_pin(uint8_t pin_number)`. This would probably not compile with PlatformIO.

SpenceKonde, 2021-12-11:
Yeah I'm in the process of switching to testing PIN_Pxn defines  in my copy. 

I think a `TINY_0_OR_1_SERIES` will be more readable than the particular construction you cited

Thanks, good catch - hadn't realized it hadn't been declared yet. 

also, am I missing something, or did we both not notice a chance to massively simplify set_user_pin.... 
if the port and port pin aren't NOT_A_PIN, and the port pin is either 2 or 7, we can just add the port number to the user number for evouta on pa2, or with 0x80 if port pin is 7, instead of a couple of dozen comparisons....

There is no part for which there is a port with a pin 2 or 7 on that doesn't have the option to do event out on it!
```
int8_t Event::set_user_pin(uint8_t pin_number) {
  uint8_t port = digitalPinToPort(pin_number);
  uint8_t port_pin = digitalPinToBitPosition(pin_number);

  int8_t event_user = -1;
  if (port != NOT_A_PIN && port_pin != NOT_A_PIN) {
    uint8_t evout_user = (uint8_t) user::evouta_pin_pa2;
    if (port_pin == 2) {
      event_user = (user::user_t)(evout_user + port); 
    } 
    #if !defined(TINY_0_OR_1_SERIES)
      else if (port_pin == 7) {
        event_user = (user::user_t)(evout_user + port); 
      }
    #endif
    else {
      return -1;
    }
    set_user((user::user_t)event_user);
  }
  return event_user;
}
```

SpenceKonde, 2021-12-11:
Can't combine the Dx's and mega0's as one would like to for the users - the DA's and DB's are off by 1 from eachother! (seriously microchip? )

MCUdude, 2021-12-11:
> Yeah I'm in the process of switching to testing PIN_Pxn defines in my copy.

Great! Let me know if it works or not. The code is certainly easier to read, that's for sure. I've just pushed this very change to my repo. https://github.com/MCUdude/MegaCoreX/commit/03a8fac14a78676253ef86e6c9065c555fe1491f

> I think a TINY_0_OR_1_SERIES will be more readable than the particular construction you cited

I would actually prefer `TINY0_SERIES`, `TINY1_SERIES` and `TINY2_SERIES`. Then we can let the macro logic do the work:

```c++
#if defined(TINY0_SERIES) || defined(TINY1_SERIES)
  // Ugly workaround code
#elif defined(TINY2_SERIES)
  // Elegant code
#endif
```

> also, am I missing something, or did we both not notice a chance to massively simplify set_user_pin....
if the port and port pin aren't NOT_A_PIN, and the port pin is either 2 or 7, we can just add the port number to the user number for evouta on pa2, or with 0x80 if port pin is 7, instead of a couple of dozen comparisons....

Great catch! I didn't think of this. It's getting late here, so I'll have to continue tomorrow, but it would be great if you could give it a try!

> Can't combine the Dx's and mega0's as one would like to for the users - the DA's and DB's are off by 1 from eachother! (seriously microchip? )

where exactly?

SpenceKonde, 2021-12-12:
I've3 checked in the latest versions of my code. Testing was not particularly thorough. 

1.4.0 and 2.5.0 are going out tonight come hell or high water, so that what has been added will be available to users - a TON of changes have gone in recently, and I want to get them into the hands of the people - and I'm moving and will not be able to devote time to this like I want to. So let's get something out, and if there are disasters, quick fixes can be pushed out, but no big new things for a while. (and when big new things are able ot happen again, ATTinyCore's final chapter is next in line).

aikopras, 2021-12-14:
I've tested both the MegaCoreX and DxCore variants and both work well (for my specific libraries). Thanks!!!

MCUdude, 2021-12-14:
Thanks for taking your time to check @aikopras! I only have a few minor things left before it's ready to be merged into the master branch.

MCUdude, 2021-12-17:
@SpenceKonde I think `clear_user` will work on any tinyAVR. It's missing PORTMUX handling for the tinyAVRs.

```c++
void Event::set_user(user::user_t event_user) {
  // Figure out what user register to write to based on the passed parameter
  uint8_t event_user_mask = event_user & 0x7F;
  #if defined(TINY_0_OR_1_SERIES)
    volatile uint8_t *user_register = &EVSYS_ASYNCUSER0 + (volatile uint8_t &)event_user_mask;
  #else
    volatile uint8_t *user_register = &EVSYS_USERCCLLUT0A + (volatile uint8_t &)event_user_mask;
  #endif
    // Connect user to the channel we're working with
    *user_register = channel_number + 1;
    // Set PORTMUX pin swap for EVOUT if selected as channel generator
  #if defined(TINY_0_OR_1_SERIES)
    if (event_user >= 0x08 && event_user <= 0x0A) {
      PORTMUX.CTRLA |= 1 << (event_user - 0x08);
    }
  #else
    if (event_user & 0x80) {
      #if defined(MEGACOREX)
        PORTMUX_EVSYSROUTEA |= (1 << ((event_user & 0x7F) - 0x09));
      #elif defined(__AVR_DA__)
        PORTMUX_EVSYSROUTEA |= (1 << ((event_user & 0x7F) - 0x0E));
      #elif defined(__AVR_DB__)
        PORTMUX_EVSYSROUTEA |= (1 << ((event_user & 0x7F) - 0x0D));
      #elif defined(__AVR_DD__)
        PORTMUX_EVSYSROUTEA |= (1 << ((event_user & 0x7F) - 0x0D));
      #elif MEGATINYCORE_SERIES == 2
        PORTMUX_EVSYSROUTEA |= (1 << ((event_user & 0x7F) - 0x0D));
      #endif
    }
  #endif
}

void Event::clear_user(user::user_t event_user) {
  // Figure out what user register to write to based on the passed parameter
  uint8_t event_user_mask = event_user & 0x7F;
  volatile uint8_t *user_register = &EVSYS_USERCCLLUT0A + (volatile uint8_t &)event_user_mask;

  // Disconnect from event generator
  *user_register = 0x00;

  // Clear PORTMUX pin swap for EVOUT if selected as channel user
  if (event_user & 0x80)  {
    #if defined(__AVR_ATmegax08__) || defined(__AVR_ATmegax09__)
      PORTMUX_EVSYSROUTEA &= ~(1 << ((event_user & 0x7F) - 0x09));
    #elif defined(__AVR_DA__)
      PORTMUX_EVSYSROUTEA &= ~(1 << ((event_user & 0x7F) - 0x0E));
    #elif defined(__AVR_DB__)
      PORTMUX_EVSYSROUTEA &= ~(1 << ((event_user & 0x7F) - 0x0D));
    #endif
  }
}
```

MCUdude, 2021-12-19:
The latest Event system library has now been merged into master.

closed 2021-12-19 by MCUdude