MCUdude/MegaCoreX #127: analogWrite Not Working

btribley, 2021-08-29:
MegaCoreX version 1.0.8,  ATMega1608 28-pin Standard, PWM on pin PC0, physical pin 2, macro PIN_PC0

Writing any value makes no difference, the pin is always LOW.  I have tried adding 
analogWriteFrequency(1); 
to no avail. Do I have to enable TCA0? It's a dual-use pin and it looks like its primary function is USART0_Tx. I haven't done any pin swapping or Pin Control register settings. 

Thanks in advance for any help.

MCUdude, 2021-08-29:
Which microcontroller or board are you using?

MCUdude, 2021-08-29:
By default, there's no PWM on PC0 or any PORTC pin really. You _can_ move all the TCA0 PWM output pins to a different port, but you'll have to write to the low-level registers directly

From README
### PWM output
PWM output, `analogWrite()`, is available for the following pins:

| Pinout            | Number of PWM pins | Available PWM pins                 |
|-------------------|--------------------|------------------------------------|
| *28 pin standard* | 8                  | 2, 3, 12, 13, 14, 15, 16, 17       |
| *32 pin standard* | 8                  | 2, 3, 12, 13, 14, 15, 16, 17       |
| *40 pin standard* | 8                  | 8, 9, 10, 11, 12, 13, 30, 31       |
| *48 pin standard* | 9                  | 13, 14, 15, 16, 17, 18, 19, 38, 39 |
| *Uno WiFi*        | 6                  | 3, 5, 6, 9, 10, 27                 |
| *Nano Every*      | 5                  | 3, 5, 6, 9, 10                     |

[image]

btribley, 2021-08-29:
Thank you, Hans. I read "available" to mean "enabled", very bad. Looking at the data sheet once again TCA0 by default is on PORTA0..5 .

MCUdude, 2021-08-29:
It is correct that TCA0 defaults to PORTA, but MegaCoreX pin swaps them to PORTC to make the pinout more convenient

btribley, 2021-08-29:
MCUdude this is fascinating. I have not done any pin swapping, using MegaCoreX as installed, standard Arduino IDE. What would I need to do to activate the PWM output on PC0? One issue is that I'm using the other PORTC pins that are mapped to TCA0 so this may not be a good solution at this point anyway.

SpenceKonde, 2021-08-30:
You'd need to either modify the core to initialize PORTMUX.TCAROUTEA to PORTMUX_TCA0_PORTC_gc instead of PORTMUX_TCA0_PORTD_gc, and change the digitalPinToTimer table in the variant file, or modify that portmux register yourself and write replacement for analogWrite(). 

DxCore, which supports the Dx-series parts, where the relevant parts of TCA0 and PORTMUX are nearly identical) actually checks PORTMUX.TCAROUTEA when you call analogWrite() and will output PWM on any port that it's pointed at (It sounds extravagant, but the register is maximally convenient for this, so it's a lot less wasteful of flash and clock cycles than it sounds.

btribley, 2021-08-30:
Thanks for the explanation.  I had wanted to use DxCore to standardize the architecture with the other project I just finished but unfortunately the low pin count parts in the smaller package weren't out six months ago when I started designing the hardware... 

I do have a homebrew solution that will do a PWM on PC0 so for the project I'm covered. But I don't like the solution and would like to learn more about how to play with stuff like this. I was blindsided when I realized that even though I had read the README I still missed something this big. 

Hans says just above that the timer is already on PORTC. So what you are saying is that in this case all I would need to do is change the 28-Pin Standard variant table to remap the digitalPinToTimer to PC0? Reading the data sheet it seems that if a timer is used then five pins are mapped, which for me is impossible as all I need is one PWM output. And the other PORTC pins are being used right now for other things and they work fine.  

If I made this change I'd like to make a new variant table and name it something like 28 pin PC0 PWM but that may be too extensive a change in comparison to just adjusting the current file and making notes and comments everywhere so if I upgrade the core I make the change again...

SpenceKonde, 2021-08-30:
The chart says it's set to PORTD not PORTC, though.

SpenceKonde, 2021-08-30:
Also, only the pins you ask for PWM from will output PWM, the rest can do something else, like any other special function that a pin has.

btribley, 2021-08-30:
I am going to look at this tonight. Thanks for the info.

btribley, 2021-08-31:
I found where PORTD is specified and looked through analogWrite() in wiring_analog.c . Found this in wiring_analog.c in analogWrite:
```
    /* Get timer */
    uint8_t digital_pin_timer = digitalPinToTimer(pin);

```
which depends on this array in pins_arduino.h:
`const uint8_t digital_pin_to_timer[] = {`
BUT digital_pin_to_timer90 is in an ifdef:
`#ifdef ARDUINO_MAIN`
and my VS Code with manually inserted include files and -D definitions taken from the verbose Arduino IDE build says that ARDUINO_MAIN is not defined...

Given some time I can parse this out to understand the implications of making changes to the  I think for now I am going with my own PWM for the simple case I need.

btribley, 2021-09-01:
That didn't work. Going to try to #define MILLIS_USE_TIMERB0  . PIN_PC0 is included there. Unfortunately PD0:5 are already used.

SpenceKonde, 2021-09-02:
#define TCA0_PINS PORTMUX_TCA0_PORTD_gc // TCA0 output on PD[0:5]

needs to change to portc. And digitalPinToPWM definitions would need to be adjusted to change all the PDn to PCn, and you would also have to change the digitalPinToTimer table. 

Defining MILLIS_USE_TIMERB0 won't work, the define needs to be specified when the core is compiled, not just when your sketch is compiled so nothing you can do in your sketch will work - that needs to be defined in something included by Arduino.h or as a command line argument (which is what my cores do with the millis timer menu, I don't recall if MegaCoreX has a millis timer menu.

SpenceKonde, 2021-09-05:
No as I have explained above that it not correct, I have too many problems on my own cores to keep combing over here and doing Hans's work for him. Sorry I'm done here, I didn't have enough time to post what I already had, and it's become clear you need constant hand holding to make progress. I do not have time for that. All I'm oging to say in my cklosing shot is if you;d use a Dx-series part this stuff would have been trivial to do, .Hans has fallen nearly a year behiund my cores on functionality and shows no interesting in reversing that trend so what am I going to do???

MCUdude, 2021-09-05:
> Hans has fallen nearly a year behind my cores on functionality and shows no interesting in reversing that trend so what am I going to do???

I'll assume you're referring to MegaCoreX? How about involving me or at least inform me about what you're currently working on, and what's ahead? I know you've "Arduino wrapped" more features than I have, but I would probably have done some of them differently, but now it's too late. I've also tried to look at your most recent Github commits, but that's also turned out to be a dead-end because of all the "keeping the CI happy" commits.

I'd love to continue improving MegaCoreX, but I can't just pick up all the new features you implement in your cores without having any influence on them. If you want to cooperate, I'm ready. I'm still waiting for your thoughts regarding the Event system library (#126).

btribley, 2021-09-22:
PIN_PD0 works with the firmware as-is. I have other hardware changes to make, so that's what I will do.

closed 2021-09-22 by btribley

adelin-mcbsoft, 2022-11-23:
Could anyone help me with the necessary configuration/registers to make analogWrite() work with Thinary mkII (ATmega4808) ?
AFAIR, it has PWM on pins 9, 10 (but not on 11, though I'd wish), however, I tried them and it does not work.
It turns fully on at values higher than 128, and fully off at values lower than 127... it acts like a normal GPIO digital pin.

MCUdude, 2022-11-23:
@adelin-mcbsoft the Thinary Nano 4808? No, this doesn't have PWM on pin 9 and 10:

[image]

adelin-mcbsoft, 2022-11-24:
Oh, my bad, from [this list](https://github.com/MCUdude/MegaCoreX#pwm-output) I (wrongly) assumed it has the same PWM pins as the original Nano Every.

So, according to the image from your reply and the legend taken from the other boards) I understand that the PWM pins are those in purple? (marked with `TCAx`, `TCBx` -> `PA2`, `PA3`, `PD4`, `PD5`, `PD0`, `PD1`, `PD2`, `PD3`) ?

_[Later Edit]:_
I also found the same information in the image below, which match the _"32 pin standard output"_ from [that list](https://github.com/MCUdude/MegaCoreX#pwm-output), so I guess those are the ones...

[image]

Well, unlucky charm for me, since I swaped a Nano for Thinary, hoping that it'll still have PWM output on the same pins.
Let's see what software PWM and `digitalWriteFast()` can do... 😁

Anyway, thank you for your help!