MCUdude/MegaCoreX #128: analogRead(A9) 255

yourSpace24, 2021-09-07:
Hi,

when i read the ADC value from Analog pin: A9 -  (PF3 on a TQFP32 Atmega4808 pin23) i always get 255 even when nothing is connected.

If it is floating or put to ground always 255. All the other ADC pins working fine tested it on multiple cpus. 

does someone have an idea ? I use the latest MegaCoreX 1.0.9 on Arduino.

cheers

Michael

SpenceKonde, 2021-09-07:
does it reproduce when PF3 is grounded (through a resistor, something in the 100-1000 range should be good)? If yes, do you confirm that ~0v is measured on the pin through an external device? 

The fact that 255 is the number makes me suspicious that a -1 might be getting treated as a signed byte somew2here along the line. 

A pin connected to nothing should never be used as a test of behavior except under very rare use cases - literally any value could be measured for it's voltage without violating any reasonable assumptions, providedd that value was between ground and the supply voltage.

MCUdude, 2021-09-07:
Thanks for reporting! I'm able to reproduce the issue.

https://github.com/MCUdude/MegaCoreX/blob/358e102c5afd74ca2ae212ec707cd4ffa5322ebc/megaavr/cores/coreX-corefiles/wiring_analog.c#L54-L59

The problem is that `digitalPinToAnalogInput` returns the analog pin _channel_ not the analog pin number. For 32-pin parts, you only have 12 analog pins, but the highest analog channel number is 15. This is also an issue for the Nano 4808 pinout as well. A9 isn't the only pin that's not working; A10 and A11 are also affected.

I'll fix it right away, but there won't be a new boards manager release just yet. There are some library improvements I'd like to finish first. Meanwhile, you can replace `NUM_ANALOG_INPUTS` on line 57 with `15` instead.

closed 2021-09-07 by MCUdude

yourSpace24, 2021-09-08:
Hi Hans,

perfect it works. Thanks so much for the quick fix.

cheers
Michael

commit 07cd4c70 (2021-09-07): Fix analogRead issue for 32-pin parts
 Closes #128
commit 1337691a (2021-09-07): Fix analogRead issue for 32-pin parts
 Closes #128