MCUdude/MegaCoreX #132: Serial baudrates with 20 MHz not correct?

RudolfAtHome, 2021-11-18:
My board with the ATmega4808 generates correct output in the serial monitor at 16 MHz. Garbage occurs at 20 MHz.
I tested this at 38400 bps. There might be an error in the calculation of baud_settings, there is "only" calculated with int32_t.

MCUdude, 2021-11-18:
You probably need to set the fuses/re-burn the bootloader. It is the fuse bits that determine which clock source to use internally; a 16 or 20 MHz clock. The clock division is done in software

RudolfAtHome, 2021-11-18:
In my opinion, this is not a good solution!
  - The instructions say that the module with the ATmega4808 does not have a bootloader.
  - In the case of the Arduino Nano Every with ATmega4809, this was very easy to adapt in a package file.
 The "clock division" with integer arithmetic is strange because of the rounding errors.

MCUdude, 2021-11-18:
I can't change the "Burn Bootloader" label in the Arduino IDE. That's just how it is. The Nano Every runs by default at 16 MHz.

What happens if you select these settings and run "Burn Bootloader"?
[image]

SpenceKonde, 2021-11-18:
It is a constant frustration that we have to deal with confusion because the menu says "bun bootloader" when what it means is "set fuses and burn bootloader (if any)", and we don't really have a way to change that. I gave up and made my core set any fuse that couldn't sbrick the chip  and that the core had a menu for t. 

I got sick of people arguing with me about words I didn't write and don't have the power to change such that I gave in and made it set the OSCCFG on tinyAVR a and OSCCFG  SYSFG0 an on Dx,  (BOOTSIZE/BOOTEND and CODESIZE/APPEND,  and will add SYSCFG1 for next release because I forgot to turn that off  when I discovered you couldn't trash dual rail DB's by setting them to single supply.),  - anything with configuration options for in the core (ie, with a submenu) is set on uploads, and all fuses are still written on bootload. That has worked  without complaint. one person was concerned  initially, but only because he didn't realize just how tmuch harder it is to brick these parts. until he undetstood that only fusers that can't brick the part, adn

closed 2022-02-22 by MCUdude