MCUdude/MegaCoreX #133: Serial.printf() doesn't work with more than 1 substitution.

SpenceKonde, 2021-11-26:
I am using the serial.printf code in my cores which I believe you originally suggested I use. 

A user has reported that it only works when a single value is getting formatted. 
https://github.com/SpenceKonde/DxCore/issues/199

```
Monitor.printf("\r\n[2]elapsTime[%d], eSecs[%d], milliSecs[%d]\r\n", elapsTime, eSecs, milliSecs);
```
is expected to print 
```
[2]elapsTime[9602], eSecs[9], milliSecs[602]
```
He says he's seeing this instead
```
[2]elapsTime[9602], eSecs[0], milliSecs[9]
```
He iniriallt was encountering it with the full version of printf enabled (to print floats), but it reproduces even with the default. 

Have you seen this before? Is it a known limitation?

MCUdude, 2021-11-26:
Thanks for reporting, I'm able to reproduce the issue.
I've not used printf with MegaCoreX that much, but I remember it was working when I first implemented it.

I've used printf ALOT with MightyCore, and never had any issues.

I'll look into it.

MCUdude, 2021-11-26:
> It's a simple fix though; use the correct formatter for the datatype:

| Formatter | Datatype       |                              |
|-----------|--------------|------------------|
| `%d`        | signed 16-bit | `int` or `int16_t` |
| `%u`        | unsigned 16-bit | `unsigned int` or `uint16_t` |
| `%ld`        | signed 32-bit | `long` or `int32_t` |
| `%lu`        | unsigned 32-bit | `unsigned long` or `uint32_t` |

closed 2021-11-26 by MCUdude