MCUdude/MegaCoreX #135: Making Library compatible with Adafruit SSD1306 for oled

bhaveshk023, 2021-12-14:
I want to drive oled with 4809 controller. Not able to use Adafruit_SSD1306 and Adafruit_GFX libraries.
While compiling the code getting error for SPI and WIRE libraries. 
Can you please help me out and tell me how can I use these above mentioned libraries with Atmega 4809 controller. 

Getting below error.
C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp: In constructor 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BusIOBitOrder, uint8_t, SPIClass*)':

C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:24:58: error: no matching function for call to 'SPISettings::SPISettings(uint32_t&, BusIOBitOrder&, uint8_t&)'

   _spiSetting = new SPISettings(freq, dataOrder, dataMode);

                                                          ^

In file included from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.h:9:0,

                 from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:1:

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note: candidate: SPISettings::SPISettings()

   SPISettings() { init_AlwaysInline(4000000, MSBFIRST, SPI_MODE0); }

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note:   candidate expects 0 arguments, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note: candidate: SPISettings::SPISettings(uint32_t, BitOrder, uint8_t)

   SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note:   no known conversion for argument 2 from 'BusIOBitOrder {aka _BitOrder}' to 'BitOrder'

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(const SPISettings&)

 class SPISettings {

       ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(SPISettings&&)

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp: In constructor 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BusIOBitOrder, uint8_t)':

C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:71:58: error: no matching function for call to 'SPISettings::SPISettings(uint32_t&, BusIOBitOrder&, uint8_t&)'

   _spiSetting = new SPISettings(freq, dataOrder, dataMode);

                                                          ^

In file included from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.h:9:0,

                 from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:1:

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note: candidate: SPISettings::SPISettings()

   SPISettings() { init_AlwaysInline(4000000, MSBFIRST, SPI_MODE0); }

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note:   candidate expects 0 arguments, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note: candidate: SPISettings::SPISettings(uint32_t, BitOrder, uint8_t)

   SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note:   no known conversion for argument 2 from 'BusIOBitOrder {aka _BitOrder}' to 'BitOrder'

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(const SPISettings&)

 class SPISettings {

       ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(SPISettings&&)

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

MCUdude, 2021-12-14:
What happends if you modify Adafruit_SPIDevice.h and add ` || defined(MEGACOREX)`?

https://github.com/adafruit/Adafruit_BusIO/blob/649bc9f93c00a19d4217281f0a6b1a2556fbd7bb/Adafruit_SPIDevice.h#L12-L19

bhaveshk023, 2021-12-14:
Still giving error after modifying Adafruit_SPIDevice.h

Below is the error : 
In file included from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_I2CDevice.h:5:0,

                 from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_I2CDevice.cpp:1:

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\Wire\src/Wire.h: In member function 'bool Adafruit_I2CDevice::_read(uint8_t*, size_t, bool)':

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\Wire\src/Wire.h:64:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int, int)

     uint8_t requestFrom(int, int, int);

             ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\Wire\src/Wire.h:62:13: note: candidate 2: virtual uint8_t TwoWire::requestFrom(uint8_t, size_t, bool)

     uint8_t requestFrom(uint8_t, size_t, bool);

             ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp: In constructor 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BusIOBitOrder, uint8_t, SPIClass*)':

C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:24:58: error: no matching function for call to 'SPISettings::SPISettings(uint32_t&, BusIOBitOrder&, uint8_t&)'

   _spiSetting = new SPISettings(freq, dataOrder, dataMode);

                                                          ^

In file included from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.h:9:0,

                 from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:1:

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note: candidate: SPISettings::SPISettings()

   SPISettings() { init_AlwaysInline(4000000, MSBFIRST, SPI_MODE0); }

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note:   candidate expects 0 arguments, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note: candidate: SPISettings::SPISettings(uint32_t, BitOrder, uint8_t)

   SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note:   no known conversion for argument 2 from 'BusIOBitOrder {aka _BitOrder}' to 'BitOrder'

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(const SPISettings&)

 class SPISettings {

       ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(SPISettings&&)

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp: In constructor 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, int8_t, int8_t, int8_t, uint32_t, BusIOBitOrder, uint8_t)':

C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:71:58: error: no matching function for call to 'SPISettings::SPISettings(uint32_t&, BusIOBitOrder&, uint8_t&)'

   _spiSetting = new SPISettings(freq, dataOrder, dataMode);

                                                          ^

In file included from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.h:9:0,

                 from C:\Users\bhavesh\Documents\Arduino\libraries\Adafruit_BusIO-master\Adafruit_SPIDevice.cpp:1:

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note: candidate: SPISettings::SPISettings()

   SPISettings() { init_AlwaysInline(4000000, MSBFIRST, SPI_MODE0); }

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:64:3: note:   candidate expects 0 arguments, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note: candidate: SPISettings::SPISettings(uint32_t, BitOrder, uint8_t)

   SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {

   ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:55:3: note:   no known conversion for argument 2 from 'BusIOBitOrder {aka _BitOrder}' to 'BitOrder'

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(const SPISettings&)

 class SPISettings {

       ^~~~~~~~~~~

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note: candidate: constexpr SPISettings::SPISettings(SPISettings&&)

C:\Users\bhavesh\Documents\Arduino\hardware\MegaCoreX-master\megaavr\libraries\SPI\src/SPI.h:53:7: note:   candidate expects 1 argument, 3 provided

MCUdude, 2021-12-14:
You're probably using an old version of MegaCoreX, Adafruit SSD1306, Adafruit GFX, or Adafruit BusIO.

I'm able to compile [all SSD1306 examples](https://github.com/adafruit/Adafruit_SSD1306/tree/master/examples) without any errors.

bhaveshk023, 2021-12-14:
After updating the MegaCoreX I am able to compile the code. Thank you for your quick support.

closed 2021-12-14 by bhaveshk023

MCUdude, 2021-12-14:
Great to hear you figured it out. May I ask which version of MegaCoreX you had, or the date you cloned the repo?

bhaveshk023, 2021-12-15:
I had downloaded the Zip on approximately 29-Oct-2019.