MCUdude/MegaCoreX #136: Add Input Disable

med1306, 2021-12-17:
Is it possible to add in pinMode en mode witch disable input buffer?

in my code i use that
void disableInputBuffer(uint8_t pin){
    uint8_t bit_pos = digitalPinToBitPosition(pin); // Get pin bit position in PORT register
    PORT_t *port = digitalPinToPortStruct(pin);     // Get pin port register
    volatile uint8_t *pin_ctrl_reg = getPINnCTRLregister(port, bit_pos); // Calculate where pin control register is
    *pin_ctrl_reg = PORT_ISC_INPUT_DISABLE_gc;      // Disable input buffer on specified pin
}

but it can be greate to do that pinMode(pin, INPUT_DISABLE)

Regards

MCUdude, 2021-12-17:
Thanks for your input! I'll see what I can do 👍

SpenceKonde, 2021-12-17:
I consider pinMode to be a lost cause. And a bad precedent to start adding
more options to the existing pinMode.

I implemented this and a bunch more with the pinConfigure function on
megaTinyCore and DxCore. The code should be directly portable to MegaCoreX.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

rocketscream, 2021-12-17:
I implemented this on my [low power library](https://github.com/rocketscream/RocketScream_LowPowerAVRZero) specific for the 0-series as it doesn't exist when I needed it. This feature is useful especially when you want to isolate unused pins from floating. It does reduce some current in the nA range per pin.

MCUdude, 2022-03-16:
the pinConfigure function, that lets you disable a pin completely is available in the latest boards manager release, v1.0.10.

closed 2022-03-16 by MCUdude