MCUdude/MegaCoreX #139: Special protocol when uploading via Optiboot?

seisfeld, 2022-02-22:
Hi!

First let me congratulate for this very nice Core for the megaAVRs. We are using a custom made board with the ATmega4809 in our project and it works like charm. We flash Optiboot via UPDI and then onwards use Serial0 to upload sketches with the Arduinio IDE. So far so nice.
We have a not very technical user base though and for previous boards (328p based) we offer a webuploader so our users can flash their board using Chrome if they don't want to or can't deal with the Arduino IDE. Sadly this is not working with the ATmega4809. I wonder why though. Is Optiboot from your Core working differently? Is avrdude doing something different? From what I found checking how avrdude is called by the IDE everything looks normal. So I wonder why a web implementation of STK500 protocol is not working. It transfers the sketch (well it looks like at least) but when the Chip is reset afterwards the Sketch is not starting. As if the transfer never happened. The chips looks empty afterwards. Any clue what is going on here?

Second question, I found when I have avrdude verify the (successfull, as said avrdude works like charm) transfer it fails the verify. Is this normal?

Any help greatly appreciated!

cheers
Stephan

MCUdude, 2022-02-22:
Hi Stephan!

> First let me congratulate for this very nice Core for the megaAVRs

Thanks for the recognition! I've put quite a lot of work into this (and several other cores), so it's nice to hear it's working well and live up to its expectation.

There's nothing special with the version of Optiboot that's running on the ATmega4809. The baud rate in use is 115200, and there is a default 1 second timeout before starting to execute the application if no data is received.

I've re-compiled the bootloader(s) and increased the timput from 1s to 4s just to check if there is a timeout issue. Can you install the one that matches your hardware and report back if it works or not? If it works, we can decrease the timeout down to 2 seconds and check again.

If you turn on verbose output in Arduino IDE you can just copy the Avrdude bootloader command, modify the 
`-Uflash:w:[bootloader_file.hex]:i` part with the path where the new bootloader is located.

[OptibootX_115200_4s_timeout.zip](https://github.com/MCUdude/MegaCoreX/files/8119916/OptibootX_115200_4s_timeout.zip)

seisfeld, 2022-02-22:
> Thanks for the recognition! I've put quite a lot of work into this (and several other cores), so it's nice to hear it's working well and live up to its expectation.

You are welcome!

Unfortunately the modified bootloader does not make any difference. Beside that fact that it now takes 4s for the sketch to start. But you already explained why this is on #140 . 🙂

seisfeld, 2022-02-22:
> Does your custom board have the autoreset circuit wired up? Is it resetting and flashing the LED when chrome tries to upload?

Yes it can auto reset. We don't have any LED wired though. The 4809 board is still in early prototype status. We are currently migrating from a different MCU due to chip shortage and waaaaay more peripherals with the 4809. What pin is optiboot using for the LED? PA7? I could solder one in...

> I'd love to see a web uploader that uploads via STK500, are you able to share that?

We are using [this](https://github.com/dbuezas/arduino-web-uploader), done by a fried of mine. Behind the scenes it using [this](https://github.com/jacobrosenthal/js-stk500v1) afaik. 

> After a failed write, when you dump the contents of the flash via UPDI, what's there? Has it written anything?

I'm afraid I never tried that. I would need some hints how to do it actually. :D

> Can you post a hex file that manifests the problem? (I have a specific pitfall in mind that this would lay bare)

It manifestes with any hex I generate. From the simple blink sketch to out own application. I have a demo [website](https://hotr.de/aio-v2) setup which is ready do deliver a 4809 hex if you want to try. You probably can't do much with it with out the board but it does have serial output on SERIAL0...

seisfeld, 2022-02-23:
> Does your custom board have the autoreset circuit wired up? Is it resetting and flashing the LED when chrome tries to upload?

Small update. I wired in an LED now and I can see two things: 1) Yes it's blinking 3 times when Chrome starts the upload. 2) After Chrome says its done and power is kept on, it repeatedly blinks 3 times over and over again.

SpenceKonde, 2022-02-23:
Oh goddamnit, did my gigantic post of instructions on how to read the hex file back last night get eaten?

SpenceKonde, 2022-02-24:
So the led repeats the triple flash at the same rate both before and after the failed upload? Any diffference in behavior after power cycle. 

The next steps for you are, IMO:

1. Upload a hex file unsuccessfully. Read the contents back via UPDI. 
Compare with the original - what is different? Is the flash empty? filled with garbage? something else? 
For bet results, reinitialize the board to a fully known state (freshly burned bootloader). 

2 Are you able to get any debug output from the uploader tool? According to that site, it has a verification function. Surely if the sketch is not being uploaded correctly, it would not be able to verify either? 

3. Connect the target board normally. With a second serial adapter, connect it's ground to the common ground of programming serial adapter and target, and it's RX line to the TX line of the serial adapter being used to upload. (hence also the  Open connection at the upload baud rate in a high quality serial terminal program, for example, hTerm. Make attempt and and fail to upload. You now have a log of everthing the unsuccessful upload said to the chip , Save the log. Make sure you have it set up to show even the non-printing characters since the upload is performed as binary data. (hterm allows you to show data as hex, decimal - hex probably being the most useful
Repeat this step with a method that does upload successfully via optiboot. 

I predict you will find an obvious difference how it goes about the upload. I actually don't know off hand how the NVMCTRL behaves after and error, ie, what would it do if the core tried to start uploading unprogrammed bytes between 0x00 and 0x200 thwe

seisfeld, 2022-02-24:
Wow! Thanks for the extensive info. Let my try to work it step by step.

> So the led repeats the triple flash at the same rate both before and after the failed upload? Any diffference in behavior after power cycle.

I did verify this again and with the blink sketch as test file (see below), after flashing with Chrome it just blinks 3 times once after power on. Then it stops.

> 1. Upload a hex file unsuccessfully. Read the contents back via UPDI.
> Compare with the original - what is different? Is the flash empty? filled with garbage? something else?
> For bet results, reinitialize the board to a fully known state (freshly burned bootloader).

1. To init the test, I did burn Optiboot via UPDI and immediately dumped it via UPDI.
2. Flashed Blink Sketch via known working method (Arduino IDE / avrdude) via Serial. Dumped it via UPDI.
3. Flashed Blink Sketch via Chrome via Serial. Dumped it via UPDI. It makes no difference whether there was a sketch already loaded or there was just the Bootloader. The dump looks the same in both cases after Chrome is done. To me it looks as if the "header" is missing? Isn't that where all the vectors are stored? But that's just wild guess...

Here are the three dumps as well as the Blink hex used: [dumps.zip](https://github.com/MCUdude/MegaCoreX/files/8134005/dumps.zip)

> 2. Are you able to get any debug output from the uploader tool? According to that site, it has a verification function. Surely if the sketch is not being uploaded correctly, it would not be able to verify either?

The verify option mentioned has no effect. But I don't recall I ever used that with 328p based boards and if it was working there.

I have another observation though regarding verify. When using the Audio IDE / avrdude to flash via serial it apparently can only verify short sketches like Blink successfully. When I try to flash a larger sketch, the flashing works fine but the verify breaks after 94% with an error:

```
Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9651 (probably m4809)
avrdude: reading input file "test.ino.hex"
avrdude: writing flash (20380 bytes):

Writing | ################################################## | 100% 2.60s

avrdude: 20380 bytes of flash written
avrdude: verifying flash memory against test.ino.hex:
avrdude: load data flash data from input file test.ino.hex:
avrdude: input file test.ino.hex contains 20380 bytes
avrdude: reading on-chip flash data:

Reading | ###############################################    | 94% 1.85savrdude: ser_recv(): read error: Device not configured
avrdude: stk500_recv(): programmer is not responding
Reading | ################################################   | 95% 2.25savr_read(): error reading address 0x0200
    read operation not supported for memory "flash"
avrdude: failed to read all of flash memory, rc=-2

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: stk500_recv(): programmer is not responding
ioctl("TIOCMGET"): Device not configured
avrdude: ser_close(): can't reset attributes for device: Device not configured

```

I don't know if this could be related to our design with the CH340C (maybe we made a mistake there...) or this is a software/driver issue!? Normally verify is disabled with '-V' when uploading sketches using the Arduino IDE.

Reading your number 3, dumping the communication, please bear with me I need to see if I get that working. But maybe the above already sheds some light.

Thanks so much!
Stephan

SpenceKonde, 2022-02-24:
JUST AS I SUSPECTED! 

The hex file specifies the data starts at 0x0200.. After the bootkloader. The tool is starting the upload from 0x0000 The first page fails,, and then 512 bytes in, it starts writing now that it's in writable flash, but the first 512b are missing and everything is offsef -0x200 bytes.

SpenceKonde, 2022-02-24:
And that is why I was asking about the hex files, because I wanted to make sure that the files being uploaded were being compiled for use with a bootloader, which has to have -section-start parameter passed to the compiler to tell it to generate bootloader-compatible hex files instead of non-bootloader files 

This is a fundamental difference between a normal hex file and a bootloader-compatible hexfile - and making a tool to port between the two is impossible, because ijmp and icall use absolute addresses given by the r30 amd r31 register pair (Z pointer). The information needed to acheive that is only known before the linking step 

So now the issue is narrowed down to either:
* Uploader is failing to correctly interpret the hex file, and thinks tht files starting from 0x0200 start from 0x0000 and starting the write from there. OR
* Uploader is unsuccessfully communicating the start address to the bootloader. 

(that of course assumes that the pipeline involved here is taking a hex file and trying to upload it, not compiling another copy incorrectly without the -section-start parameter and uploading that) 

Also, this core is poorly configured, and and wastes flash for large sketches, it's not using -mrelax (you can tell from the fact that the vector table for a sketch that fits within the first 8k is still usingfull length jump instructions (0x940Cxxxx) instead of rjmps - you don't get to use the weird bytes in the vector table that you save to store anything, but it makes a difference on larger sketches - the compiler very rarely generates rjmp and rcall instructions on parts with >8k fash unless you tell the linker to -mrelax. Will create an issue for that

That's a very odd verify error you're reporting.... what I think is bizzare is that it gets to 94% and then barfs, but says it's doing 0x200 address then. That's not 94% of the way into a 20k file.

seisfeld, 2022-02-24:
Thanks for the interpretation of the hex files. Makes totally sense what you are saying.

My (wild) guess is that the javascript stk500 library is not interpreting the hex file correctly. I checked the example hex files that come with it and all start from address 0x0000. I think the assumption there is that it's always like this and hence its not reading the address and treating it correctly. Unfortunately I'm not a JS coder and can't interpret what exactly its doing to confirm this.

Based on [this](https://github.com/Optiboot/optiboot/wiki/HowOptibootWorks) "normally" on Arduino (like 328p based chips for example) the Bootloader is not at the beginning of the flash but further down depending on the exact chip. On reset a register of some sort makes the chip jump there and the Bootloader gets executed. On ATmega4809 the bootloader seems to be at the beginning of the flash, hence the Sketch starts at a later address. So if this is true, this needs to be fixed in the JS library which was likely never designed to support the 4809 and its bootloader location. Does this make sense?

I will try if I get the sniffing of the communication working and post you the outcome if I manage to get something useful. Thanks again for your advises. :)

MCUdude, 2022-02-24:
As @SpenceKonde pointed out, the issue is most likely caused by the 0x200 offset that all "bootloader compatible" hex files has to include (unlike older AVRs where the bootloader were located at the top op the flash memory).

(While I was typing, @seisfeld posted his reply)

I'm not able to reproduce this issue using Arduino IDE and Avrdude, so my guess is that [the uploader](https://github.com/jacobrosenthal/js-stk500v1) always starts writing data to address 0x0000, because that's how "classic" AVRs work.

> Also, this core is poorly configured, and and wastes flash for large sketches, it's not using -mrelax (you can tell from the fact that the vector table for a sketch that fits within the first 8k is [...]

The reason why I'm not able to keep up with you is this: Work family and kids takes up a large portion of my time. The other one is to try to keep up with the megaTinyCore and DxCore commit history. I know there's lots of great work in there, but without proper commit messages that explains what the commit does and _what_ if fixes, it's sometimes difficult for me to determine if this is something I should adapt or not.

seisfeld, 2022-02-24:
> I'm not able to reproduce this issue using Arduino IDE and Avrdude, so my guess is that [the uploader](https://github.com/jacobrosenthal/js-stk500v1) always starts writing data to address 0x0000, because that's how "classic" AVRs work.

Yap, me neither and exactly my guess. And it's good, that this way is working fine. I personally can live with that as well. :) But our project (a nfc controlled jukebox for little kids) has many non techie users (moms and dads who just want to "get it done" when it comes to firmware updates). Since we had this Webuploader thing in the past I was thinking, hey how hard can it be to port this to the new architecture. And here we are. ;)

But I'm glad I tried, because l also learned much due to this issue. Some really smart ppl here. Thank you guys!

SpenceKonde, 2022-02-25:
[testhexfiles.zip](https://github.com/MCUdude/MegaCoreX/files/8138070/testhexfiles.zip)
One of these is a test file to try uploading to the 4809 and to a classic 328p. In both cases burn bootloader immediately before the test to ensure that it works correctly. 
```
# Upload via optiboot. Observe the device.

# Pass: 
# On a freshly bootloaded classic AVR it should do nothing. 
# On a freshly bootloaded **modern** AVR, it should repeatedly enter the bootloader and generate the triple-blink. 

# Failure: 
# On a classic: repeatedly run the bootloader. 
# on a modern AVR: write pin PA7 high and do nothing further. 

# Under examination when you dump the flash, you should see:
# Modern AVR - data up to 0x204, then a bunch of 0's, 6 more bytes 512b further in, then 0's up until just past the 2k mark
# (0x22 bytes after specifically) - the point of this is just to bring out any failures to write all pages). 

# Classic AVR - EMPTY flash (0xFF) up to 0x200. 2 instructions, then a bunch of 0's, then another 4 bytes, then 0's up to 0x822. 
```

The second one is that blink hex file you gave me, with a bunch of 0's padding it. See if that will run when upoloaded to the 4809.

This should tell you what's going on with certaintyy.

SpenceKonde, 2022-02-25:
> The reason why I'm not able to keep up with you is this: Work family and kids takes up a large portion of my time. The other one is to try to keep up with the megaTinyCore and DxCore commit history. [...]

I'm not really sure how do a better job of my commit messages - you've said this before, but no matter what I do, there are going to be a shit-ton of commits. Many of which are tiny, a few of which are a really big deal. And there are problems that can only be detected (by me) once I commit the files and push to repo. I routinely lament the fact that there's no way in heaven and eartth to undo a commit because I slipped and clicked commit without finishing the commit messsage. Or if I committed a file I meant to go in a different commit added subsequently, and now when I commit the next fix, it's just the docs for the fix hat was accidentally committed with a previous one. Doesn't matter, it's set in stone now, and the only way to undo it is to make an even bigger mess to look through with commits that undo the previous commit and then redo it a ltittle differently. 
I realize they suck, but I don't really know how to improve that. 

I would suggest using the changelog, which I can curate to reflect reality accurately, while a commit message can only reflect what I believed to be the case when I wrote the code.

SpenceKonde, 2022-02-25:
[Altered_Blink.4809.zip](https://github.com/MCUdude/MegaCoreX/files/8138382/Altered_Blink.4809.zip)
Test with this hex file too - I noticed something that looked off about your blink hex file. Where did that file come from????

```
:0400000300000200F7
```
Record type 0x03, value 0x200. Record type 03 is:
" **Start Segment Address:**	For 80x86 processors, specifies the starting execution address). The byte count is always 04, the address field is 0000 and the first two data bytes are the CS value, the latter two are the IP value." 

When I export hex files, I don't have that record. I'm not sure it should be there. So if the "altered blink" file works, then the problem would be quite different; I'd call it "Upload tool neither ignores nor reports an error upon encountering record type 03 in hex file"

MCUdude, 2022-02-25:
> I'm not really sure how do a better job of my commit messages - you've said this before, but no matter what I do, there are going to be a shit-ton of commits. Many of which are tiny, a few of which [...]
I realize they suck, but I don't really know how to improve that. If you have any thoughts on this, or if you can point me to some resource that you think would be helpful... though I'm pessimistic about it ever being possible for commit messages to be an effective record of that someone could use to see which changes they would want to port. (I've stopped globbing toegether unrelated changes into single commits at least for the most part at least?)

The great thing about git is that nothing is set in stone. I use `git rebase` very often, and it lets you edit a commit (what's been commited, the code), change the commit message, delete the commit all together and more. When you're done, you simply fore-push your changes, and everything will line up perfectly.

Here's what I usually do:

* `git log --oneline`
* Copy a short hash, usually 5th or 6th from the top. Can't be the actual commit you want to change
* `git rebase -i [that hash]`
* Use the text editor to replace `pick` with `drop`, `edit` or `reword` for the commit you want to chage
* Save and exit
* Do the necessary changes
* `git add -A`
* `git rebase --continue`
* `git push --force`

You've now rewritten history!

seisfeld, 2022-02-26:
Hey Spence,

I did the tests with the 4809, here are the results:

**test_hex_start_not_zero.hex**

1. flash fresh bootloader via updi
2. flash test_hex_start_not_zero.hex
3. obversavation: LED continuously blinks 3 times with led lit between the phases 
4. dump: see zip

(Is this _kinda_ pass? Normally LED is off between the phases...)

**padded_blink.4809.hex**

1. flash fresh bootloader via updi
2. flash padded_blink.4809.hex
3. obversavation: LED blinks 3 times twice, then stops
4. dump: see zip

**Altered_Blink.4809.hex**

1. flash fresh bootloader via updi
2. flash Altered_Blink.4809.hex failed due to checksum error

> [Altered_Blink.4809.zip](https://github.com/MCUdude/MegaCoreX/files/8138382/Altered_Blink.4809.zip)
> Test with this hex file too - I noticed something that looked off about your blink hex file. Where did that file come from????

Arduino IDE->Sketch->Export compiled Binary - I'm using MegaCoreX v1.0.9.

> When I export hex files, I don't have that record. I'm not sure it should be there.

This is odd... what am I doing differently!?

I could not run all tests with classic AVR 328p because I can't restore the bootloader currently. I did flash _test_hex_start_not_zero.hex_ though and can confirm it does nothing. I included the dump anyways but it includes partially an old sketch because the flash was not erased.

[dumps-20220226-01.zip](https://github.com/MCUdude/MegaCoreX/files/8146426/dumps-20220226-01.zip)

I hope this all makes sense to you!

cheers
Stephan

seisfeld, 2022-02-26:
Hi Spence,

I managed to capture the serial communication now with a second serial2usb adapter. Please find a working and non working capture below. I also included the hex I did upload in both cases for reference.

[serial_captures-20220226-01.zip](https://github.com/MCUdude/MegaCoreX/files/8147140/serial_captures-20220226-01.zip)

cheers
Stephan

seisfeld, 2022-02-26:
Hi Spence,

My friend and I made progress today with the JS stuff. The library (actually two, one to read the hex and one doing the upload) he uses were indeed making "assumptions". The hex reading lib did report a starting address but the uploading one was always assuming zero as start. He modified it and we just did upload the blink sketch successfully! A UPDI dump afterwards looks 1:1 as when avrdude did upload it! :) He will release a fixed version soon and also make a PR to the source library.

Will close this issue after some more testing.

Thanks for your help Spence and Hans! Really appreciate it!

cheers
Stephan

SpenceKonde, 2022-02-27:
There's a trap here, set by whatever you are using to generate those weird hex files with "start segment address" record specifying 0x0200. This is not valid in an AVR .hex file, and your toolchain should not be generating it. In any event, you should not assume that when that's not present in the file, that it always starts at 0. 

Unless the library has a method for this, You might need to loop from 0...flashsize checking for non 0xFF values. Which strikes me as sort of silly, but I think the pyrhon lib is similar

seisfeld, 2022-02-27:
> There's a trap here, set by whatever you are using to generate those weird hex files with "start segment address" record specifying 0x0200, and the library probably maps that to This is not valid in an AVR .hex file (the start address is specified by the fuses - and while there is a standard (as in endorsed by microchip) way for specifying them, this is not it. ), and your toolchain should not be generating it. In any event, you should not assume that when that's not present in the file, that it always starts at 0.

The tool we use just wraps [this uploader](https://github.com/jacobrosenthal/js-stk500v1) (which uses [this library](https://github.com/bminer/intel-hex.js) to parse the hex file) into a website. Here is what the readme says:

```
The parse function takes 2 arguments:

data - Intel Hex file (string in ASCII format or Buffer Object)
bufferSize - the size of the Buffer containing the data (optional)

and returns an Object with the following properties:

data - data as a Buffer Object, padded with 0xFF where data is empty.
startSegmentAddress - the address provided by the last start segment address record; null, if not given
startLinearAddress - the address provided by the last start linear address record; null, if not given
```

> Export compiled binary from the IDE doesn't add that record.

Mine does for the 4809 (not for 328p). That's exactly how I generated the hex files I shared here.

SpenceKonde, 2022-02-27:
> Mine does for the 4809 (not for 328p). That's exactly how I generated the hex files I shared here.

Iiinteresting. I am seeing different behavior depending on how I process the file. Some tools don't retain it, python intel-hex moves it around..... Do you know if there's any documentation from an authoritative source (Microchip, Atmel, gcc. etc to back up that convention? 
Now if there was authoritative reference  material that said it was required, well, then it doesn't really matter as much

But that said, seeing as doing it in the general case for any hex that doesn't start at 0, regardless of how or why is sorta trivial.... 

I can think of like 5+ 1-2 line algorithms that should work in python or javascript with just adaptation to the language syntax.....

```
start = 0
while ((start < flashsize) && (data[start]== 0xFF)) {
start++;
}
start = start & (~(pagesize-1)); 
```
startpage = 0;
for ( var i = 0; i < flashsize; i++) {
  if (data[i] != 0xFF) {
    startpage =parseInt( i / pagesize);
    break;
  }
}
```

(am out of practice with js, but I think the algorithm is sound.

seisfeld, 2022-02-27:
> Do you know if there's any documentation from an authoritative source (Microchip, Atmel, gcc. etc to back up that convention?

Unfortunately I don't. It's all just tweaking and poking around, modifying libraries etc to make it work. Kinda frustrating but I'm also happy we have something working now. We did some more testing today and it's at a point where the dump via updi looks exactly like if flashed via avrdude. And it blinks along...

MCUdude, 2022-03-11:
I'm closing this issue because it's not really a bug cause by MegaCoreX, but rather an issue with how the upload tool handles hex files with address offsets.

closed 2022-03-11 by MCUdude