MCUdude/MegaCoreX #140: Startup delay caused by Optiboot?

seisfeld, 2022-02-22:
Hi,

We experience a shot 1s - 1.5s delay until the stored sketch is starting when using Optiboot. When we flash the ATmega4809 via UPDI the sketch starts instantaneous after applying power. Is this because Optiboot waits for a new sketch every time the MCU starts up? I'm not 100% sure, and not an expert, but - if its Optiboot causing this - shouldn't it only wait for new load after soft reset? Not power on? I'm just guessing here though. Maybe this is totally wrong thinking.

We are using the CH340C as USB2UART bridge if that is of any help. Is this a known issue with Optiboot or is this behavior normal? Or, third option, we have done something wrong with our design...

Any hin greatly appreciated!

cheers
Stephan

MCUdude, 2022-02-22:
Hi!

Optiboot adds a 1 second delay if a hardware reset "pulse" on the reset pin is applied. This is because when uploading new programs, the RTS/DTR line pulls the reset line low (though a 100n cap) , causing a reset, and starts executing the bootloader. When powering the board though a USB to serial adapter, the RTS/DTR lines has a tendency to toggle a few times, usually on when the USB to serial chip has been initialized by a Windows computer.

If you actually apply power externally, not though USB, the bootloader detects a power-on reset and starts the application immediately.

So, Optiboot is not causing this, it's the USB to serial chip when initialized by a computer.
Hope this makes sense!

seisfeld, 2022-02-22:
Thanks for the detailed explanation! We'll look into this, maybe there is a way to rectify this...

seisfeld, 2022-02-25:
To track this down we took step by step actions to see what actually triggers this. We went as far as removing power to the USB bridge and removing all connections between it and the 4809 except a pull-up resistor its reset line. Still we see Optiboot triggered, seeing his 3-times led flash and 1s delay.

Is there any other pin on the 4809 that could cause this?

seisfeld, 2022-02-26:
I did some more research on this issue. I checked in the [optiboot repo](https://github.com/Optiboot/optiboot) and found [optiboot_x.c](https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/optiboot_x.c) is way newer than in this repo here. The detection of power on reset seems to have been changed there as well. Is it possible that power on reset detection is not working with the version of optiboot in this repo here?

I tried to print out RSTCTRL.RSTFR - it shows 0. If POR was the cause it should read 1. I don't know if this might be cleared by the bootloader though. The more recent version does store the cause in GPIOR0 for use in the app.

@MCUdude Do you consider to update your bootloaders to the upstream version? I'd love to try the other version but I don't know how to compile it to work with this core...

MCUdude, 2022-02-27:
Yes, you're right. I just had a look at the source files, and the MegaCoreX bootloaders could need an update. I'm currently on vaccation, but I'll have a look at this in a week or so!

MCUdude, 2022-02-27:
I haven't published the sources yet, since the makefile need some work. But here are the pre-compiled bootloader files where the sketch starts immedeatly.
[OptibootX_POR.zip](https://github.com/MCUdude/MegaCoreX/files/8148341/OptibootX_POR.zip)

seisfeld, 2022-02-27:
Hey Hans!

Really appreciate the efforts during your vacation! I tried the one matching our UART layout and bingo, the POR issue is fixed. No wait after power on anymore. :)

I think we can close this one then as soon as you made the fix public.

Thanks again!

Cheers
Stephan

SpenceKonde, 2022-02-28:
Oh, nice, you're pulling in the GPIOR0 reset flag version? Glad to hear it. 

A while back (while doing optiboot for DxCore) I came to the conclusion that preserving reset flags in place violated one of the prime directives of a bootloader: The application shall not adversely impact the functioning of the bootloader
This is not possible if the reset flags are not cleared upon startup. 

Furthermore, it is not possible for the entry conditions to be honored if the reset flags are not reset upon startup. 

The method I use (which I suspect is the same as you've pulled in?)
1. read the reset flags into a variable. 
2. If that variable takes a value of zero, that means execution arrived at the reset vector *without* a reset, and we have no idea what the state of other peripheral registers are: We don't know what clock speed we are running at is, nor do we know anything about a dozen other factors that could result in the bootloader functioning incorrectly. On classic AVRs, lacking a software reset, it was plausible that a user might perform a manual jump to 0x0000 to enter the bootloader from the application. On modern AVRs, there is software reset, so doing that sort of dirty jump is incorrect code. In the event of such a dirty reset, I immediately issue a software reset
3. otherwise, I clear the reset flags by writing the value of that aforementioned variable to RSTFR.
4. I write the variable to GPIOR0
5. I test the value of the variable against the entry conditions and decide whether to jump to app or run bootloader. 

When compiling for a non-optiboot board, I have a little routine (weakly defined so users can override it) that runs in .init3 which checks reset flag register, if it's 0 at startup - that's a dirty reset, we cannot guarantee that init() will put the chip into a working state, and so we issue a software reset, otherwise we stash the value read from the reset flags in GPIOR0 just like the bootloader is. I initially wanted to try to get users to do that, but I very quickly realized that 99% of users are not going to even copy/paste reset cause flag resetting routine that I was recommending. So I made the less-broken behavior the default. 

Both of my cores discuss the reasoning for this at great length in extras/Ref_Reset.md, including the causes of unexpected "dirty resets" (you will get them if BAD_ISR is invoked, you overwrite the return address in the stack with a value in  the empty space at the end of the flash, or if the value used in an icall or ijmp operation is corrupted somehow. 

This is not unique to modern AVRs, though the problem is more visible there, because BAD_ISR is guaranteed to leave the application completely hosed. It does the same thing as a classic AVR - jump to 0x00. However, on a a classic AVR that will have a result more closely resembling a reset. The global interrupt flag is cleared when entering the ISR, but init(), right before calling setup() executes an sei instruction and interrupts work again, and the stack pointer was reinitialized back in .init2, to that's clean so it will only malfunction due to other registers being messed up. On a modern AVR, you end up in a guaranteed-broken state that only a reset can fix, because CPUINT.LVL0EX is set, and sei doesn't change that. No other interrupts (except the single one that can be set to higher priority) can run, and that bit can only be unset by executing RETI or by a reset. 

After analyzing a number of failures, I came to the conclusion that the vast majority of cases where the chip gets into a "bad state" and requires manual reset are in fact due to a dirty reset caused by some software bug. Since on a modern AVR, there is no good reason to ever jump to 0x0000 intentionally (instead, you should do a software reset if you do want to run the bootloader, or a watchdog reset if you don't), any time we're at the beginning of the program (optiboot on a board that uses it, otherwise the application), and RSTFR = 0, something went severely wrong,. and since both init() and optiboot rely on the assumption that all registers are in their reset state , IMO the only defensible actions at that point are to either a) issue a software reset, or b) while(true), and I think the reset is likely to be preferred  by a huge margin among users.