MCUdude/MegaCoreX #144: Fuses are not programmed correctly for 20MHz

SensorsIot, 2022-03-15:
I uploaded a sketch to a 4808 using 16MHz. Then I changed to 20MHz. This is the protocol:

Sketch uses 506 bytes (1%) of program storage space. Maximum is 49152 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 6140 bytes for local variables. Maximum is 6144 bytes.
Forcing reset using 1200bps open/close on port COM24
C:\Users\andre\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\andre\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/avrdude.conf -v -patmega4808 -cjtag2updi -PCOM24 -e -Uflash:w:C:\Users\andre\AppData\Local\Temp\arduino_build_30504/digitalReadFast.ino.hex:i {bootloader.fuse0} {bootloader.fuse1} {bootloader.fuse2} {bootloader.fuse4} {bootloader.fuse5} {bootloader.fuse6} {bootloader.fuse7} {bootloader.fuse8} {bootloader.lock} 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\andre\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/avrdude.conf"

         Using Port                    : COM24
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   0a:0b:0c:0d:0e:0f
Device ID:                       Thinary mkII
         AVR Part                      : ATmega4808
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
         Serial number:          0a:0b:0c:0d:0e:0f
         Vtarget         : 5.0 V

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e9650 (probably m4808)
avrdude: erasing chip
avrdude: reading input file "C:\Users\andre\AppData\Local\Temp\arduino_build_30504/digitalReadFast.ino.hex"
avrdude: writing flash (506 bytes):

Writing | ################################################## | 100% 0.14s

avrdude: 506 bytes of flash written
avrdude: verifying flash memory against C:\Users\andre\AppData\Local\Temp\arduino_build_30504/digitalReadFast.ino.hex:
avrdude: load data flash data from input file C:\Users\andre\AppData\Local\Temp\arduino_build_30504/digitalReadFast.ino.hex:
avrdude: input file C:\Users\andre\AppData\Local\Temp\arduino_build_30504/digitalReadFast.ino.hex contains 506 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.10s

avrdude: verifying ...
avrdude: 506 bytes of flash verified

avrdude done.  Thank you.

The digitalWriteFast frequency stayed the same.
A change to 12MHz also did not program fuses and the clock rate stayed the same.
At 8MHz I got half the toggle frequency
A change to 5MHz did not program fuses, but resulted in a 4 times lower toggle frequency

MCUdude, 2022-03-15:
Thank you for reporting Andreas! The issue can be seen in the upload command:

```
C:\Users\andre\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude
-CC:\Users\andre\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/avrdude.conf 
-v -patmega4808 -cjtag2updi -PCOM24 -e -Uflash:w:C:\Users\andre\AppData\Local\Temp\arduino_build_30504/digitalReadFast.ino.hex:i
{bootloader.fuse0} {bootloader.fuse1} {bootloader.fuse2} {bootloader.fuse4} {bootloader.fuse5}
{bootloader.fuse6} {bootloader.fuse7} {bootloader.fuse8} {bootloader.lock}
```

As you can see, the bootloader "place holders" doesn't get replaced by the actual fuse values. I'm able to reproduce this issue, and I can confirm that it's related to the Nano 4808 pinout, and does not affect the other chips and their pinouts.
I'll push a fix for this soon, and I'll try to issue a new boards manager release within this week.

MCUdude, 2022-03-16:
A new release, MegaCoreX v1.0.10 is now available though Boards manager, where the issue has been fixed. Thanks for reporting!

closed 2022-03-16 by MCUdude