MCUdude/MegaCoreX #145: Please document #define to identify MegaCoreX

technoblogy, 2022-04-13:
Is there a `#define` that identifies that the MegaCoreX core is being used, and if so, could it be documented?

Something like `if defined(DXCORE)` and `if defined(MEGATINYCORE)` that Spence Konde provides for his cores?

It would be easier than doing:

    #if defined(__AVR_ATmega808__) || defined(__AVR_ATmega1608__) || defined(__AVR_ATmega3208__) || \
    defined(__AVR_ATmega4808__) || defined(__AVR_ATmega809__) || defined(__AVR_ATmega1609__) || \
    defined(__AVR_ATmega3209__) || defined(__AVR_ATmega4809__)

MCUdude, 2022-04-13:
> Is there a #define that identifies that the MegaCoreX core is being used, and if so, could it be documented?

Yes, there is. It's quite hidden, but it's there. I should probably document this. They are all located in the pins_arduino.h files.

https://github.com/MCUdude/MegaCoreX/blob/531beba634cdb05d6f8bd67b9ed45c646a0ac4d8/megaavr/variants/48pin-standard/pins_arduino.h#L26-L30

https://github.com/MCUdude/MegaCoreX/blob/531beba634cdb05d6f8bd67b9ed45c646a0ac4d8/megaavr/variants/32pin-standard/pins_arduino.h#L21-L25

technoblogy, 2022-04-13:
Thanks! Exactly what I was looking for.

closed 2022-04-13 by technoblogy