MCUdude/MegaCoreX #146: I2C compile error for 4809

vjkolaventy, 2022-04-24:
Hi

I am getting the following error when I compile the OLED test program using I2C. The same program compiles and runs flawlessly on Arduino Uno. I am using the Arduino Uno with JTAG2UPDI to program the 4809.

In file included from C:\Arduino\libraries\Adafruit_BusIO\Adafruit_I2CDevice.h:5:0,
                 from C:\Arduino\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp:1:
C:\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.10\libraries\Wire\src/Wire.h: In member function 'bool Adafruit_I2CDevice::_read(uint8_t*, size_t, bool)':
C:\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.10\libraries\Wire\src/Wire.h:77:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int, int)
     uint8_t requestFrom(int address, int quantity, int sendStop);
             ^~~~~~~~~~~
C:\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.10\libraries\Wire\src/Wire.h:76:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, size_t, bool)
     uint8_t requestFrom(uint8_t address, size_t quantity, bool sendStop);

MCUdude, 2022-04-24:
Is it a real error, or is it just a warning? I'm just getting a warning when trying to compile similar code.
Can you post the entire output from Arduino IDE?

vjkolaventy, 2022-04-24:
I am not sure if its an error or warning. I think its an error because the sketch is not  working on 4809 while the same is working on Uno. It also compiles without these errors/warnings on Uno. Below is the entire output.

C:\Arduino\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::_read(uint8_t*, size_t, bool)':
C:\Arduino\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp:191:79: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                               ^
In file included from C:\Arduino\libraries\Adafruit_BusIO\Adafruit_I2CDevice.h:5:0,
                 from C:\libraries\Adafruit_BusIO\Adafruit_I2CDevice.cpp:1:
C:\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.10\libraries\Wire\src/Wire.h:77:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int, int)
     uint8_t requestFrom(int address, int quantity, int sendStop);
             ^~~~~~~~~~~
C:\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.10\libraries\Wire\src/Wire.h:76:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, size_t, bool)
     uint8_t requestFrom(uint8_t address, size_t quantity, bool sendStop);

Also noticed that the IDE throws up message the first time, but if I run it again, I dont get this warning/error. However, the program does not work no matter how many times I upload it.

SpenceKonde, 2022-11-13:
I solved this tarpit of bullshit by getting rid of all the aliases for requestfrom except the one that took uint8_t as all arguments.

closed 2023-01-10 by MCUdude