MCUdude/MegaCoreX #147: MegaCoreX ATMega4809 pinout is different from datasheet

vjkolaventy, 2022-04-27:
Hi

The pinout shown on MegaCoreX for ATMega4809 is different from the datasheet. Nearly all the pin numbers in pins_arduino.h are different from the datasheet. I have got a custom PCB with ATMega4809 chip. Should I change the pinout as per the datasheet in pin_arduino.h for it to work? No program that I uploading to the 4809 using UPDI is working. I am not even able to get the I2C OLED screen to work.

It even more confusing, that of the many times I uploaded the sketch to the 4809, the OLED screen randomly worked once. No matter how many time I reloaded the software, I could not get the OLED to work. Thanks in advance.

Regards
Vijay

MCUdude, 2022-04-27:
> The pinout shown on MegaCoreX for ATMega4809 is different from the datasheet.

What? There are physical pins, "port pins" (PB0 for instance), and Arduino pins. You can't expect a physical pin to match the Arduino pin number. The pinout is fine, just use it as it is. If you get confused by the Arduino pin numbers, just use the PIN macros instead (PIN_PA0, PIN_PB1, etc).

vjkolaventy, 2022-04-27:
I am sorry if I upset you. I just read that certain boards like Uno WiFi needed pin swap. Since the board I am using is custom made and I am not able to get the I2C OLED screen to work, I thought I might need to change the pin numbers. Just ruling out if I am making a mistake

MCUdude, 2022-04-28:
> I am sorry if I upset you

No worries! But the physical pinout of the chip is something that has to be taken into account when designing a PCB.

Do you have a schematic and perhaps a picture of the board I can have a look at?

vjkolaventy, 2022-04-28:
Thanks. I got this PCB made from this [Solder Reflow Plate](https://github.com/AfterEarthLTD/Solder-Reflow-Plate) project. I am not sure if there is problem uploading the program or if its the pin numbers. I am using jtag2updi to program the board with Arduino Uno. I also posted the problem on jtag2updi (https://github.com/ElTangas/jtag2updi/issues/62)

[Schematic_Solder Reflow 70_50mm ATMega4809.pdf](https://github.com/MCUdude/MegaCoreX/files/8586016/Schematic_Solder.Reflow.70_50mm.ATMega4809.pdf)

MCUdude, 2022-04-28:
Well, the board designer hasn't really tried to make troubleshooting easy. No on-board LEDs to blink, and neither of the four serial ports has been broken out.

Try to upload the modified blink sketch below and measure the voltage on the MOSFET gate with a multimeter to see if it turns on and off every second:

```cpp

void setup()
{
  pinMode(PIN_PC3, OUTPUT);
}

void loop()
{
  digitalWrite(PIN_PC3, HIGH);
  delay(1000);
  digitalWrite(PIN_PC3, LOW);
  delay(1000);
}

```

If it does, it proves that the chip is working correctly, and a working program is running.

vjkolaventy, 2022-04-28:
Thank you. I am getting 4.9V on the mosfet. Looks like the board is working. Any advise on why I am not able to get the OLED SSD1306 I2C display to work? Both I2C SDA and SCL pins have 3.3V. I am trying this simple code.

#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128 // OLED display width,  in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

// declare an SSD1306 display object connected to I2C
Adafruit_SSD1306 oled(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);

void setup() {
  delay(2000);         // wait for initializing
  oled.clearDisplay(); // clear display

  oled.setTextSize(1);          // text size
  oled.setTextColor(WHITE);     // text color
  oled.setCursor(0, 10);        // position to display
  oled.println("Hello World!"); // text to display
  oled.display();               // show on OLED
}

void loop() {
}

MCUdude, 2022-05-01:
> Thank you. I am getting 4.9V on the mosfet. Looks like the board is working

Are you getting a steady 5V on the MOSFET, or is it toggling on and off as we would expect from the example sketch you loaded?

> Any advise on why I am not able to get the OLED SSD1306 I2C display to work? Both I2C SDA and SCL pins have 3.3V. I am trying this simple code.

I have no idea really. Without any serial pins broken out on the board, it's difficult to debug the code to figure out what's going on. I have no experience with the Adafruit GFX library, but I've used the [u8g2](https://github.com/olikraus/U8g2_Arduino) library, in the past. Which pinout are you using? The 48pin standard, Uno Wifi or nano Every?

JoaquimBoavida, 2022-07-06:
Taking advantage of this open thread, to ask another question, also pinout related:

I want to use Ethernet module WIZ850IO, that uses standard SPI.  Can I use the standard pinout?  I had some trouble in  the past regarding the SS pin. Thanks!

SpenceKonde, 2022-11-13:
If the wire library has the bug that my SPI library used to have (I don't know offhand whether it does), SS needs to be set output, because my SPI library was clearing the SSD (Slave Select Disable) bit, and if the pin is input and goes low, it puts the chip into slave mode; The arduino library does not support this mode at all, and that's certainly not the mode you want - in that case the SS pin of the SPI pinset being used must be set output. If it doesn't have the bug, and SSD is always set while using the Arduino library, then there is nothing magic about the SS pin). 

I don't think his library ever had that bug (I think it was a bug I added when I did the new attach mode (and ripped out the specific per-pin using interrupt option) and didn't realize and didn't realize that some code elsewhere woudl result in it being unset by SPI.endTransfer())

closed 2023-01-10 by MCUdude