MCUdude/MegaCoreX #148: whats now?

xinnie-the-pooh, 2022-05-22:
Why this project canceled？(arduino always cut some useful function on their boards,such like 'can'bus on due..so i love the 3rd-part board lib)
One other thing:  does this chip more expensive than before？in 2021 i bought one for 3$,but now need 5$ ,damn..

seisfeld, 2022-05-23:
> What makes you think this project is canceled?

Wild guess: The Travis CI button at the top of the readme says "canceled".

> I still accept PRs, follow up on reported bugs/issues, and add new functionality from time to time

Keep up the awesome work! Really appreciate it.

MCUdude, 2022-05-23:
> Wild guess: The Travis CI button at the top of the readme says "canceled".

Travis CI has been a pain to use ever since they changed their policy for open source project usage. I'm planning to migrate over to Github Actions, but it is a lot of work and takes a lot of time; Something I don't have at the moment.

> Keep up the awesome work! Really appreciate it.

Thanks! It's great to hear people find my projects useful.

closed 2022-05-23 by MCUdude

seisfeld, 2022-05-23:
> Thanks! It's great to hear people find my projects useful.

Anytime. If I can contact you beside GitHub somehow to exchange address I'll send you one of our next PCBs where we use your core with. Only if you are interested of course.

MCUdude, 2022-05-23:
Thanks for the offer @seisfeld!

I do have a lot of hardware lying around that I don't use that often, but if the board you're working on has some interesting features that may be useful, It may come in handy. Please send me a picture of the board to 
`hansibull {A} googles mail service {dot} com`. Thanks!