MCUdude/MegaCoreX #149: TCA register probleme

amchiri, 2022-05-24:
Hi I am a student (from french sorry for my expression) and i try to to use the waveform generator for my project but it doens't work i think that there might be a probleme with the the register CTRLA from TCA it's like we cannot set the TCA in FRQ generator because whatever I write in that register the period always depend about the register PER from TCA could you try to check that please? (below here my code copied from https://www.avrfreaks.net/forum/atmega4808-wave-form-generator)

/*
*
*/

void Clock_Init(void){
// Main Clock Configuration

CPU_CCP = CCP_IOREG_gc;
// Select 20MHz Internal Oscillator
CLKCTRL_MCLKCTRLA = CLKCTRL_CLKSEL_OSC20M_gc;

CPU_CCP = CCP_IOREG_gc;
// Disable Prescaler
CLKCTRL_MCLKCTRLB = ~CLKCTRL_PEN_bm;

while(CLKCTRL_MCLKSTATUS & CLKCTRL_SOSC_bm);
}
void Port_Init(void){
PORTA_DIR |= PIN0_bm;
}
void TimerA_Init(void){
// Set PA[0:5] as output for the wave form
PORTMUX_TCAROUTEA = PORTMUX_TCA0_PORTA_gc;

//Frequency Mode and Compare Channels enabled
TCA0_SINGLE_CTRLB = 0x01 | TCA_SINGLE_CMP0EN_bm;

//Disable Event count. Ticks counter instead
TCA0_SINGLE_CTRLA &= ~(TCA_SINGLE_CNTEI_bm);

// TOP --> 4000Hz = (16e6)/(2*1(CMPn + 1)) <=> CMPn = 1999
TCA0_SINGLE_CMP0 = 0x7cf;

//Select DIV1 for Timer prescaler and Enable peripheral
TCA0_SINGLE_CTRLA =  (TCA_SINGLE_CLKSEL_DIV1_gc | TCA_SINGLE_ENABLE_bm); 
}

void setup() {
//http://ww1.microchip.com/downloads/en/DeviceDoc/40002015A.pdf
Clock_Init();
Port_Init();
TimerA_Init();

for ( ; ;){

      }
}

SpenceKonde, 2022-06-06:
I think this core sets the timer up initially in SPLIT MODE in order to get 3 extra 8-bit PWM channels, since analogWrite only supports 8-bit PWM anyway. To put it back into non-split mode, before and of your TCA code:
 
TCA0_SPLIT_CTRLA=0; //stop TCA0.
TCA0_SPLIT_CTRLESET=0x33 // hard reset TCA0.

mcuee, 2023-05-12:
In this case, I guess this issue can be closed?

closed 2023-05-12 by MCUdude