MCUdude/MegaCoreX #152: Unable to retain EEPROM

Hamza-arch, 2022-07-06:
I have been trying to retain EEPROM. I am using JTAG2UPDI programmer using arduino uno. What I have understand from the documentation is that, we need to burn the bootloader once and then EEPROM gets enabled. Then we can program with programmer. Am I right??  When I burn the bootloader, I got the following message. I think it is not bootloading. Can anyone guide.

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Muhammad Hamza\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATmega4809
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
         Serial number:          00:00:00:00:00:00
         Vtarget         : 5.0 V

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions
[... 210 lines cut ...]
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude: Device signature = 0x1e9651 (probably m4809)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "C:\Users\Muhammad Hamza\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/bootloaders/empty/empty.hex"
avrdude: writing flash (0 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 0 bytes of flash written
avrdude: verifying flash memory against C:\Users\Muhammad Hamza\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/bootloaders/empty/empty.hex:
avrdude: load data flash data from input file C:\Users\Muhammad Hamza\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/bootloaders/empty/empty.hex:
avrdude: input file C:\Users\Muhammad Hamza\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/bootloaders/empty/empty.hex contains 0 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

avrdude: verifying ...
avrdude: 0 bytes of flash verified

avrdude done.  Thank you.

MCUdude, 2022-07-06:
If you look at the Avrdude output, you can see that `fuse5` is set to 0xC9. This means that the EEPROM retain fuse is already set.

(BTW you should update to the latest MegaCoreX boards manager version, which bundles Avrdude 7.0, which is way better for UPDI programming in general)

From the datasheet:
[image]

> When I burn the bootloader, I got the following message. I think it is not bootloading. Can anyone guide.

Again, if you look at the Avrdude output, you can see that it's writing `empty.hex`, which means that you have selected "Bootloader: Not bootloader" in the tools menu. However, if you're going to use JTAG2UPDI (or any other UPDI programmer), you don't need a bootloader. The "Burn Bootloader" button is still doing its thing though. It's setting the correct fuse bits according to your settings in the Tools menu.

```
avrdude: erasing chip
avrdude: reading input file "C:\Users\Muhammad Hamza\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.9/bootloaders/empty/empty.hex"
avrdude: writing flash (0 bytes):
```

Bottom line: The EEPROM retain fuse it set, and EEPROM should not be wiped when you upload your code. Can you confirm that the EEPROM gets erased after uploading a new program?

Hamza-arch, 2022-07-07:
I understand the issue now. If I save data to EEPROM, it is saved if I turn off or on the board. However, when I upload a new program, it erases the old EEPROM data. Suppose I want to upload EEPROM vales using one progam and use it on other, is it possible??

SpenceKonde, 2022-07-07:
@MCUdude  You still haven't ditched the "Empty.hex" thing? I got rid of that years ago - see the boards and platform.txt for either of my cores. 

If you're not using optiboot, no "bootloader" is involved. Burn bootloader is just required to set fuses (the menu option should say "Set fuses and install bootloader, if any", but the arduino team picked a phrasing that causes tons of confusion instead. 

Does the new program you're uploading, say, write to the eeprom without checking that something isn't there first? Blank EEPROM cells hold the value 0xFF. 

Some people have had problems with the EEPROM when it is accessed **immediately** after startup on megaTinyCore. Even a delay of a few milliseconds before interacting with the EEPROM made it go away, as did using brown-out detection. We speculated that slow rising power was to blame.

mcuee, 2022-10-15:
> I understand the issue now. If I save data to EEPROM, it is saved if I turn off or on the board. However, when I upload a new program, it erases the old EEPROM data. Suppose I want to upload EEPROM vales using one progam and use it on other, is it possible??

If the EEPROM retain fuse is set, EEPROM contents will not be erased when you upload a new sketch. If your sketch change the EEPROM contents, it can be used elsewhere.

MCUdude, 2022-10-15:
> MCUdude You still haven't ditched the "Empty.hex" thing? I got rid of that years ago - see the boards and platform.txt for either of my cores.

Nope, but I probably should.

> If the EEPROM retain fuse is set, EEPROM contents will not be erased when you upload a new sketch. If your sketch change the EEPROM contents, it can be used elsewhere.

Correct. I'm closing this issue because it's really not an issue

closed 2022-10-15 by MCUdude