MCUdude/MegaCoreX #153: Nano 4808 pinout vs. 32-pin standard pinout

rspy76, 2022-07-08:
Hi

First of all, many thanks for your work!
I'm starting with the Atmega4808 and I'm studying your library and pinouts. There is one thing which confuses me a bit.
To start with, I will use a board such as the Thinary Nano Every which uses a Atmega 4808 but then I will like to move to a custom PCB with a Atmega4808 chip which I then will program with UPDI. 
So, when I program for the Thinary Nano Every, the Arduino pins are mapped differently to the Port names in comparison to the 32-pin standard pinout. I don't understand why this is.

For example: 

- On 32-pin standard pinout the PORT PIN PA3 is arduino pin 3.
- On Nano 4808 pinout the PORT PIN PA3 is arduino pin 5

Is there a reason you chose different arduino pin numbers in these two pinouts even though the Thinary Nano Every uses the 32 pin Atmega4808?

Many thanks and best regards

[image]
[image]

SpenceKonde, 2022-07-08:
It's because the standard pinout was chosen to make logical sense, it
starts at PA0 and proceeds counterclockwise, and the ports and pins within
the ports are in order, while the thinary one matches the numbers of the
pinsprinted on the board. which were located and numbered to be more nano
like, Unfortunately, the nano 3.0 puts pins in a dumb order. Like almost
every other Arduino board :-/
I believe MCUDude's cores supply the PIN_Pxn macros, so you can refer to
pins by port and bit and eliminate all ambiguity. I am trying to slowly
reduce the presence of user visible numbers in my core documentation, move
toward exclusively calling pins by the PIN_xn macros (eg, PIN PA1, PIN_PD7
etc) especially since oon modern full size AVRs, special functions almost
never get moved around between families. Note that  the PIN_An macros are
something totally different. My cores go even further and expect
analogRead() to be passed a digital pin number or a special constant for
internal channels

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

MCUdude, 2022-09-15:
What @SpenceKonde wrote!

closed 2022-09-15 by MCUdude