MCUdude/MegaCoreX #154: Incorrect checksum format in paxkage index JSON

jgromes, 2022-07-17:
Hi,

when running a CI on one of my projects, I'm getting the following error when attempting to install MegaCoreX using arduino-cli:

```
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 installed
Error during install: Cannot install tool MegaCoreX:avrdude@7.0.0-arduino2: testing local archive integrity: testing archive checksum: invalid checksum format: 03f29bae3b53ac1a0e469c1e79f1696a572d7be632838484526674501de0248e
```

Looking at the package index JSON file, I noticed that one of the checksums format is different:

```
```

MCUdude, 2022-07-17:
Thank for reporting! The missing `SHA-256:` is absolutely the issue. I've fixed this in [1d4bebc](https://github.com/MCUdude/MegaCoreX/commit/1d4bebc755392f0e7fb2b5749adb5645c728ab92). Please delete the downloaded JSON boards manager file and try again

jgromes, 2022-07-17:
```
```

closed 2022-07-17 by jgromes