MCUdude/MegaCoreX #155: Error installing/updating MegaCoreX 1.1.0

wittend, 2022-07-21:
I have been trying to install MegacoreX 1.1.0.  I am currently using Arduino IDE 2.0.0-rc9.  But I had this problem with rc8 as well.  I am using Linux Mint 20.3 Cinnamon using Kernel 5.4.0-122-generic.

I cleaned my system of Arduino related stuff and did a fresh install of the system.  Everything works well, but installing 1.1.0 via the boards manager gives the following output:

> Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
> Tool arduino:arduinoOTA@1.3.0 already installed
> Downloading packages
> MegaCoreX:avrdude@7.0.0-arduino2
> Installing MegaCoreX:avrdude@7.0.0-arduino2
> Failed to install platform: MegaCoreX:megaavr.
> Error: 13 INTERNAL: Cannot install tool MegaCoreX:avrdude@7.0.0-arduino2: testing local archive integrity: testing archive checksum: invalid checksum format: 03f29bae3b53ac1a0e469c1e79f1696a572d7be632838484526674501de0248e
> 

I get 'Error: 13' whether doing a clean install or attempting to update from 1.0.10.  Version 1.0.10 installs and works happily.
The problem seems to be down to avrdude 7.0.  but I have been reluctant to track this down to rip-and-replace.

MCUdude, 2022-07-21:
> This sounds related to https://github.com/MCUdude/MegaCoreX/issues/154. Is the [package_MCUdude_MegaCoreX_index.json](https://github.com/MCUdude/MegaCoreX/blob/1d4bebc755392f0e7fb2b5749adb5645c728ab92/package_MCUdude_MegaCoreX_index.json) up to date?

Yes, the issue has been fixed. Please delete the downloaded `package_MCUdude_MegaCoreX_index.json` file and retry.

wittend, 2022-07-21:
Thank you!

I had to:

   1. Remove all vestiges of MegacoreX that I could find on this machine
   manually.
   2. Delete the package_MCUdude_MegaCoreX_index.json file from the
   .arduino15 folder.
   3. Remove an old instance of avrdude that seemed to be installed via
   Synaptic, or some other.
   4. Shut down the machine.
   5. Clean the monitors with Windex.
   6. Restart the system and the Arduino 2.0rc9 environment.
   7. Then reinstall MegacoreX 1.10 via the boards manager.

*Mirabilis Dictu!*
Now it works!

Dave

per1234, 2022-07-21:
The persistence of the issue even after the package index had been corrected days ago might be related to this bug in Arduino CLI: https://github.com/arduino/arduino-cli/pull/1810

Arduino IDE 2.x uses Arduino CLI for the Boards Manager installations, so a bug in Arduino CLI also affects Arduino IDE 2.x.

closed 2022-07-27 by MCUdude