MCUdude/MegaCoreX #157: How config PlatformIO with MegaCoreX with Arduino Nano Every?

mikrocoder, 2022-08-17:
Hello,

I don't know what else I have to set. With the .ini I get 4 usart error messages. I do not know where the takes the 4 USARTs.

The example seems to be incorrect. How would PlatformIO know that I want to use the MegaCoreX package?

It seems to me that the MegaCoreX is not yet integrated although it should be included by default. How does that work?

How do I get the Arduino Nano Every Board to work with the MegaCoreX under PlatformIO?

``````cpp
#include <Arduino.h>

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW); 
  delay(1000);
}
``````

.ini File
``````cpp
; Parameters used for all environments
[env]
platform = atmelmegaavr
framework = arduino

; Chip in use
board = ATmega4809
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L
; Oscillator type (internal or external)
board_hardware.oscillator = internal
; Arduino pinout variant
board_build.variant = nano_every

; Unflag build flags
build_unflags =
; Extra build flags
build_flags =

; Monitor port is auto detected. Override here
;monitor_port =
; Serial monitor baud rate
monitor_speed = 9600

; Run the following command to upload with this environment
; pio run -e Upload_UPDI -t upload
[env:Upload_UPDI]
; Upload protocol for UPDI upload
upload_protocol = atmelice_updi
upload_flags =

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
; Upload protocol for serial uploads (using Optiboot)
upload_protocol = arduino
upload_flags =
upload_port = /dev/COM7

; run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
[... 415 lines cut ...]

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\fuses_bootloader\FrameworkArduino\UART2.cpp.o] Error 1
*** [.pio\build\fuses_bootloader\FrameworkArduino\UART3.cpp.o] Error 1
============================================= [FAILED] Took 1.46 seconds =============================================

Environment       Status    Duration
----------------  --------  ------------
Upload_UPDI       FAILED    00:00:01.442
Upload_UART       FAILED    00:00:01.484
fuses_bootloader  FAILED    00:00:01.464

#include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\nano_every\FrameworkArduino\UART0.cpp.o] Error 1
*** [.pio\build\nano_every\FrameworkArduino\UART1.cpp.o] Error 1
*** [.pio\build\nano_every\FrameworkArduino\UART3.cpp.o] Error 1
*** [.pio\build\nano_every\FrameworkArduino\UART2.cpp.o] Error 1
======================= [FAILED] Took 1.49 seconds =======================
Processing Upload_UPDI ()
--------------Error: Please specify platform for 'Upload_UPDI' environment
``````

MCUdude, 2022-08-17:
Try using `board_build.variant = nano-every` (with dash instead of underscore).

Also, you need the following "environment" if you want to upload using the on-board USB connector on the Nano Every:

```ini
[env:Upload_onboard_usb]
upload_protocol = jtag2updi
board_upload.use_1200bps_touch = true
upload_flags =
```

mikrocoder, 2022-08-17:
Thanks, I'll try it out later. And how does PlatformIO even know that I want to use your MegaCoreX package?

mikrocoder, 2022-08-17:
Somehow it can flash, but with error messages. How do you even know what to set everything? The documentation does not help me. There are many lines that are not documented.

``````cpp
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

; Parameters used for all environments
[env]
platform = atmelmegaavr
framework = arduino

; Chip in use
board = ATmega4809
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L
; Oscillator type (internal or external)
board_hardware.oscillator = internal
; Arduino pinout variant
board_build.variant = nano-every

; Unflag build flags
build_unflags =
; Extra build flags
build_flags =

; Monitor port is auto detected. Override here
;monitor_port =
; Serial monitor baud rate
monitor_speed = 9600

; Run the following command to upload with this environment
; pio run -e Upload_UPDI -t upload
[env:Upload_UPDI]
; Upload protocol for UPDI upload
upload_protocol = atmelice_updi
upload_flags =

[env:Upload_onboard_usb]
upload_protocol = jtag2updi
board_upload.use_1200bps_touch = true
upload_flags =

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
; Upload protocol for serial uploads (using Optiboot)
upload_protocol = arduino
upload_flags =
upload_port = /dev/COM[7]

; run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader
[env:fuses_bootloader]
; Upload protocol for used to set fuses/bootloader
upload_protocol = ${env:Upload_UPDI.upload_protocol}
upload_flags =
; Hardware settings
board_hardware.bod = 2.7v
board_hardware.eesave = yes
board_hardware.uart = no_bootloader
board_hardware.rstpin = reset
[... 303 lines cut ...]
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.1% (used 4 bytes from 6144 bytes)
Flash: [          ]   1.6% (used 804 bytes from 49152 bytes)
Building .pio\build\fuses_bootloader\firmware.hex
Configuring upload protocol...
AVAILABLE: atmelice_updi
CURRENT: upload_protocol = atmelice_updi
Looking for upload port...
Uploading .pio\build\fuses_bootloader\firmware.hex
avrdude: usbhid_open(): No device found
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2141

avrdude done.  Thank you.

*** [upload] Error 1
================================================================= [FAILED] Took 4.56 seconds =================================================================

Environment         Status    Duration
------------------  --------  ------------
Upload_UPDI         FAILED    00:00:04.490
Upload_onboard_usb  SUCCESS   00:00:06.283
Upload_UART         FAILED    00:00:03.803
fuses_bootloader    FAILED    00:00:04.559
============================================================ 3 failed, 1 succeeded in 00:00:19.135 =====
``````

MCUdude, 2022-08-17:
>  And how does PlatformIO even know that I want to use your MegaCoreX package? Because that is not in any setting.

MegaCoreX is selected when you choose `board = ATmega4809`, `platform = atmelmegaavr` and `framework = arduino`.

Which programmer are you using for uploading? The on-board USB connector on the Nano Every?

mikrocoder, 2022-08-17:
Thank you for the explanation. That helps already once for the understanding.

I am using the original Arduino Nano Every normal with the USB cable.

If this all works I would like to configure this for an AVR128DB64. For this the DxCore is to be used. I have a selfmade board which works with the Arduino IDE. There is a FTDI232RL on it for the standard serial and for flashing with bootloader. UPDI with Atmel ICE is also possible. 

But first I have to concentrate on the Nano Every.

What are the many names in [env ...] in the platformio.ini for? I don't have a menu to choose from like in the Arduino IDE. How do I know which name to use?

In comparison, the configuration in Microchip Studio is totally simple.

This is my 5th attempt to try PlatformIO for months now.

MCUdude, 2022-08-17:
PlatformIO is fantastic once you can wrap your head around it. If you're using VSCode, you can select which environment you want to use. For other projects, I have one environment for uploading over UART, and another environment for uploading using an ISP programmer. You can easily change the environment you're using in VSCode:

[image]

Here's a working PlatformIO.ini I recently used with an Arduino Nano Every:

```ini
[platformio]
default_envs = Upload_UPDI ; Default build target

; Common settings for all environments
[env]
platform = atmelmegaavr
framework = arduino

; TARGET SETTINGS
; Chip in use
board = ATmega4809
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L
; Current pinout
board_build.variant = nano-every

; BUILD OPTIONS
build_unflags =
build_flags = 

; SERIAL MONITOR OPTIONS
; Serial monitor port defined in the Upload_UART environment
;monitor_port = 
; Serial monitor baud rate
monitor_speed = 9600

[env:Upload_UPDI]
; Upload protocol for UPDI upload
upload_protocol = jtag2updi
board_upload.use_1200bps_touch = true
upload_flags =
```

mikrocoder, 2022-08-17:
Thank you very much. Now I know what the [env] are for. That is, all the settings of a [env] are valid until the next [env] definition and so on? I will definitely have some more questions in the near future.

MCUdude, 2022-08-17:
> That is, all the settings of a [env] are valid until the next [env] definition and so on?

Environments are "isolated", but you can extend an environment like so:

```ini
[env:Upload_UPDI]
; Upload protocol for UPDI upload
upload_protocol = jtag2updi
board_upload.use_1200bps_touch = true

[env:Upload_UPDI_57600_baud]
extends = env:Upload_UPDI
upload_flags =
  -b57600
```

I'll recommend you to read through the PlatformIO docs. There's lots of valuable information there.

mikrocoder, 2022-08-17:
Thanks. My problem is that I do not understand the logic and syntax of the .ini. It is not logical for me. This also shows me your new example. A flag for a new baudrate? I don't have the logic for that.

I wanted to include my toolchain first.
platform_packages = C:/avrToolchain/avr-gcc-11.3.0_mingw32_binutils2.38/bin
Does not work either.

Tool Manager: Installing bin @ *
Error: Could not find the package with 'bin @ *' requirements for your system 'windows_amd64'.

MCUdude, 2022-08-17:
> My problem is that I do not understand the logic and syntax of the .ini. It is not logical for me

I do agree that the syntax might be a bit confusing. I try to keep various templates around because it is difficult to remember all the keywords. There's lots of info here though: https://docs.platformio.org/en/latest/projectconf/index.html

> This also shows me your new example. A flag for a new baudrate? I don't have the logic for that.

It was just a dumb example to show you how `extends` works. There's no reason to not use 115200 baud. `upload_flags` is just flags passed directly to Avrdude, and I happen to remember the correct flag to specify the Avrdude baud rate. Each flag is supposed to be on a new line, indented as shown in my example.

> How do you even know that you have to set this with these values for example?
board_upload.use_1200bps_touch = true

The Nano Every requires a 1200bps "touch" in order to leave UART mode and enter jtag2updi mode. This line needs to be there for the 1200bps thing to happen. All I searched for was "PlatformIO ini 1200bps touch", and I found it quite quickly.

> I wanted to include my toolchain first.
platform_packages = C:/avrToolchain/avr-gcc-11.3.0_mingw32_binutils2.38/bin
Does not work either.

Sorry, I've never used my own toolchain with PlatformIO before, so I can't help you there...

mikrocoder, 2022-08-17:
Thank you for everything. Let's see how I get on. Am with the x th attempt with your help already very far come.

mikrocoder, 2022-08-18:
Hello,

got this working with the toolchain. Because you helped me so kindly I show you this.

[image]
Picture 01 shows the hidden folder where PlatformIO stores all its toolchains according to my knowledge.

In the folder 'packages" I copied 2 of my toolchains. The 10.2 and the 11.3 version. With a path where they are actually located, I did not get it, therefore the copying. The folder names of my two toolchains I have renamed like the other existing.

``````
toolchain-atmelavr@1.1020.0
toolchain-atmelavr@1.1130.0
``````

The crucial thing for it to work at all are the two files .piopm and package.json in these folders. See picture 02.
[image]

You can copy both files from the existing toolchains and change only the version in both files, everything else in it you leave alone. So for me once on "version": "1.1020.0", and the other on "version": "1.1130.0",

The .ini currently looks like this.
``````
[platformio]
default_envs = Upload_UPDI

[env]
platform = atmelmegaavr
framework = arduino
board = ATmega4809
board_build.f_cpu = 16000000L
board_build.variant = nano-every
build_unflags = 
    -std=gnu++11
    ;-std=gnu++14
    ;-std=gnu++17
build_flags = 
    --verbose
    -flto
    -std=c++20
monitor_speed = 250000
;platform_packages = toolchain-atmelavr@1.1020.0  ; ok
platform_packages = toolchain-atmelavr@1.1130.0  ; ok
;platform_packages = toolchain-atmelavr@C:/avrToolchain/avr-gcc-12.1.0_mingw32_binutils2.38  ; path or anything else not in okay

[env:Upload_UPDI]
upload_protocol = jtag2updi
board_upload.use_1200bps_touch = true
upload_flags = 
lib_deps = mikalhart/Streaming@^1.0.0
``````

How would you change the .ini to switch between the default toolchain and the other two toolchains?

MCUdude, 2022-09-21:
Sorry, but I've never used custom toolchains with PlatformIO. But you may get help with this over at the [PlatformIO community forums](https://community.platformio.org/).

closed 2022-10-15 by MCUdude