MCUdude/MegaCoreX #158: Garbled serial communication on UART1 when using exported hex file for ATmega4808, using internal clock other than 16MHz

prchrmn0507, 2022-09-09:
I have a board with an ATmega4808 chip on it, and it has a serial port connected to UART1.  When I program my board from the IDE, everything works as expected with any internal clock speed I pick. If I export a hex file, with the internal clock set to 16MHz, and upload that using avrdude, everything still works as expected. However, If I pick any internal clock speed other than 16MHz, export a hex file, and upload it using avrdude, the serial communications are garbled. I suspect that an incorrect baud rate is being set, since it is tied to the internal clock speed.

MCUdude, 2022-09-10:
Sorry, but I can't reproduce your issue when using an ATmega4808.

The base clock 16 or 20 MHz is set by the fuse bits, but the clock divider is set in the init() function, which runs before setup().

If your base clock is 16 MHz, you can run sketches that are based on a 16 MHz clock. These frequencies are 16, 8, 4, 2, and 1 MHz. For a 20 MHz clock, it is 20, 10, and 5 MHz.

If I upload a sketch that runs at 16 MHz in Arduino IDE, and then export one that's compiled for 8 MHz I can upload this program using Avrdude directly and it works perfectly and without a garbled serial output.

prchrmn0507, 2022-09-10:
Hmm... I wonder what is happening with my setup? I tried 8MHz, exported it and uploaded it with avrdude, and it was garbled. I had my baud rate set to 19200 in the sketch; when that didn't work I decided to try 9600 and 38400 to see if the baud might be off by 2x, but those didn't work either. I uploaded the sketch from the IDE (with the internal clock set to 8MHz) directly to my board, and it worked correctly. I then read it out to a file. I uploaded the resulting file to another board with avrdude, and it works correctly as well. I'll have to compare the hex files and see if I can find the difference. Thank you for looking into it; the problem must be on my end. If I can figure out what I'm doing wrong, I'll report back in case it might help someone.

closed 2022-09-10 by prchrmn0507

prchrmn0507, 2022-09-12:
When I compared the hex files, they were identical, so that wasn't the problem. When I looked at the avrdude command line executed by the IDE, and compared it to what I had been doing, I found the problem. I wasn't passing the fuse settings to avrdude! I was under the mistaken impression that the fuse data was contained within the hex file. When I added the fuse settings to my avrdude command when uploading the hex file, everything functioned as expected. I'm sorry for the false alarm!