MCUdude/MegaCoreX #160: Arduino IDE V2.0, NanoEvery, Windows 10 Environment

cattledogGH, 2022-09-20:
When using MegaCoreX with Arduino IDE 2.0 there is an issue with uploading fails showing 0xc0000135 errors. This is due to a missing libusb0.dll file. In the Windows environment Avrdude has a dependency on this file and it is not included with the avrdude version bundled with MegaCoreX.

This was not an issue with previous 1.8.x versions of the Arduino IDE because the libusb0.dll was included in the root installation file of the IDE, and avrdude could find it for the upload.
This root installation of the dll file is not included with V2.0.

The simple solution is to add libusb0.dll to the bin file folder of the avrdude file in tools.

Pending a release of MegaCoreX which this addition you can add the file to your current MegaCoreX installation at C:\Users\your name\AppData\Local\Arduino15\packages\MegaCoreX\tools\avrdude\7.0.0-arduino2\bin

I think this has to be a 32bit version of the .dll file. I copied the file used in the Arduino 1.8.x ide and pasted it in the bin folder.
Alternatively, you can download libusb-win32 from SourceForge.net and rename it to libusb0.dll.
([url]https://sourceforge.net/projects/libusb-win32/(url))

MCUdude, 2022-09-21:
@per1234 is this recognized as a known IDE 2.0 issue?
I'm bundling Avrdude 7.0 with MegaCoreX using the official Arduino Avrdude 7.0 release.

per1234, 2022-09-21:
This is not really an IDE issue. It is true that the Arduino IDE 1.x installation comes with a copy of the DLL dependency, but as @cattledogGH reported at https://github.com/MCUdude/MegaCoreX/issues/156, that only helps when the IDE was started from the installation folder.

The issue is either with the AVRDUDE or with the AVRDUDE package produced by Arduino. That is tracked here: https://github.com/arduino/avrdude-packing/issues/15

There are two possible solutions:

- Adjust the build system to avoid the dependency on the DLL
- Bundle the DLL with the AVRDUDE Windows package, as has been done with all previous AVRDUDE packages distributed by Arduino

>the official Arduino Avrdude 7.0 release.

Keep in mind that these are pre-releases that were created to allow beta testing. It wasn't really expected that they would be distributed in production platform releases yet. But it certainly gets us more beta testers even if they aren't aware of that status!

mcuee, 2022-10-12:
I encountered the issue today and indeed the issue is with the packaging of Arduino avrdude Windows packaging (https://github.com/arduino/avrdude-packing/issues/15). Simple workaround of copying libusb0.dll (x86 version) works fine.

mcuee, 2022-10-16:
> Keep in mind that these are pre-releases that were created to allow beta testing. It wasn't really expected that they would be distributed in production platform releases yet. But it certainly gets us more beta testers even if they aren't aware of that status!

It is kind of slow for Arduino to adopt avrdude 7.0 release...

mcuee, 2022-10-29:
For those who want to try out for latest build of avrdude using arduino-pack project, you can try the ming32 binary[ here](https://github.com/arduino/avrdude-packing/files/9889835/avrdude-main-test-libusb-compat.zip). It works quite well for me.

Reference:
* https://github.com/arduino/avrdude-packing/issues/15#issuecomment-1295147283

mcuee, 2022-10-30:
I have also built Linux and macOS binaries here (32/64bit Intel/AMD Linux, 32bit/64bit ARM Linux, macOS x86_64).
https://github.com/mcuee/avrdude-packing/actions/runs/3354027563

mcuee, 2022-10-30:
Just one example using Nano 4808 and the blinky sketch. I do not know how to properly replace the tools. So I just overwrite the default `avrdude `and `avrdude.conf` with the 64bit Linux binary (need to `chmod +x`) and the corresponding `avrdude.conf` file.

```
/home/mcuee/.arduino15/packages/MegaCoreX/tools/avrdude/7.0.0-arduino2/bin/avrdude -C/home/mcuee/.arduino15/packages/MegaCoreX/hardware/megaavr/1.1.0/avrdude.conf -v -patmega4808 -cjtag2updi -P/dev/ttyUSB0 -Uflash:w:/tmp/arduino_build_467866/Blink.ino.with_bootloader.hex:i -Ufuse0:w:0x00:m -Ufuse1:w:0x54:m -Ufuse2:w:0x01:m -Ufuse4:w:0x00:m -Ufuse5:w:0b11001001:m -Ufuse6:w:0x06:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Ulock:w:0xC5:m 

avrdude: Version 7.0-20221029 (b569966)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is /home/mcuee/.arduino15/packages/MegaCoreX/hardware/megaavr/1.1.0/avrdude.conf
         User configuration file is /home/mcuee/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   0a:0b:0c:0d:0e:0f
Device ID:                       Thinary mkII
         AVR Part                      : ATmega4808
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_UPDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU HW version: 1
         M_MCU FW version: 6.00
         S_MCU HW version: 1
         S_MCU FW version: 6.00
         Serial number   : 0a:0b:0c:0d:0e:0f
         Vtarget         : 5.0 V

avrdude: AVR device initialized and ready to accept instructions
[... 50 lines cut ...]
avrdude: 1 byte of fuse6/syscfg1 written
avrdude: verifying fuse6/syscfg1 memory against 0x06
avrdude: 1 byte of fuse6/syscfg1 verified
avrdude: reading input file 0x00 for fuse7/append
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse7/append ...
avrdude: 1 byte of fuse7/append written
avrdude: verifying fuse7/append memory against 0x00
avrdude: 1 byte of fuse7/append verified
avrdude: reading input file 0x00 for fuse8/bootend
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse8/bootend ...
avrdude: 1 byte of fuse8/bootend written
avrdude: verifying fuse8/bootend memory against 0x00
avrdude: 1 byte of fuse8/bootend verified
avrdude: reading input file 0xC5 for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xC5
avrdude: 1 byte of lock verified

avrdude done.  Thank you.

```

MCUdude, 2023-01-11:
Fixed in the MegaCoreX v1.1.1 boards manager release!

closed 2023-01-11 by MCUdude