MCUdude/MegaCoreX #162: Nano Every Servo issue - servo.read() and servo.readMicroseconds() return incorrect value

Cobalt6700, 2022-10-28:
Hi, 

Not sure if I am doing something wrong here, but have reached the limits of my understanding. 

Using MegaCoreX board manger, programming using Arduino IDE 1.8.19 over USB. Board settings: 
Board: ATmega 4809
Clock: Internal 16MHz
BOD: BOD 2.6v
EEPROM: EEPROM retained
Pinout: Nano Every
Reset Pin: Reset
Bootloader: No bootloader

**For context:** 
I have written a sketch to drive  Servos on a 328P Nano using the Var Speed Servo Library and it works great. I am trying to move this to a 4809.
I have forked and been trying to update the VarSpeedServo Library ( https://github.com/netlabtoolkit/VarSpeedServo ) to work with a Nano Every. 

Part of my journey has been learning how the official Arduino Servo Library and the MegaCoreX Servo Library control servos and how to port that over to VarSpeedServo.    

I made some changes to my sketch to use the standard Arduino Servo Library and that seems to work okay (without any speed control), however it does seem to be 'skipping' 1 or 2 degrees when I use servo.write(). For example - read current angle at 80 deg, writing 81 deg does not move the servo, have write angle of 82 deg, read new angle at 81 deg. 

This is not my main issue - but guessing it could be related so have included it.

**Main issue:** 

When using the unmodified megaavr Servo Library the servo.read() and servo.readMicroseconds() appear to report ~1/2 of the written value. 
The .write() function seems to be working okay - I have confirmed with a scope the expected write values in us.

Writing to a servo and calling .read() and .readMicroseconds() reports: 
Write 90 deg / Read: 18 deg and 738us, scope 1465us.
Write 45 deg / Read -3 and 506us, scope 995us. 
Write 135 deg / Read 41 and 970us, scope 1925us. 

I added a line to the megaavr Servo.cpp wich multiplies the pulsewidth value by 2: 

```
uint16_t Servo::readMicroseconds() {
  uint16_t pulsewidth;
  if (this->servoIndex != INVALID_SERVO) {
    pulsewidth = ticksToUs(servos[this->servoIndex].ticks + TRIM_DURATION);
    pulsewidth = pulsewidth * 2; //added line to correct reported value
  } else {
    pulsewidth  = 0;
  }
  return pulsewidth;
}
```

Read angles are now reported correctly, and read us values are within 10us of the scope reading (from a quick glance). 
When I now write to a servo, I can increment/decrement by 1 deg - the skipping deg's issue seems to have been corrected too. 

I'm not suggesting this as a fix, I am guessing there is something happening with ticksToUs() that is causing the reported value to be 1/2 of what it should be. 

Happy to provide further info if required.

Cobalt6700, 2022-11-13:
@SpenceKonde

SpenceKonde, 2022-11-13:
Does the issue reproduce on my cores? 

MegaCoreX libraries != DxCore libraries....Mi

SpenceKonde, 2022-11-13:
If this does not manifest in my cores, this is an @MCUdude issue; there is some divergance between the libraries and MegaCoreX is not my department. 

If it does manifest in my cores cant you do a bad read of positions and then (after demonstrating teh bogus values) then print to serial via Serial.printHexLn:

Serial.printHexLn((uint8_t*)(&TCB1),16)

That will print out the register values so I can compare expected and actual behaviror
A 2*1 discrepancy suggests that the prescaling of the is being handled wrong.

SpenceKonde, 2022-11-13:
I checked - this doesn't reproduce on my cores. I think @MCUdude can just steal my version of wire and it'll work. I didn't dig to figure out exactly the source of the problem.

cattledogGH, 2022-11-14:
I can not confirm the issue reported by @Cobalt6700 where the readMicroseconds() don't match the writeMicroseconds().

I am using his settings with a Nano Every and MegaCoreX 1.1.0
programming using Arduino IDE 1.8.19 over USB.
 
Board: ATmega 4809
Clock: Internal 16MHz
BOD: BOD 2.6v
EEPROM: EEPROM retained
Pinout: Nano Every
Reset Pin: Reset
Bootloader: No bootloader

`#include <Servo.h>
Servo myservo;  // create servo object to control a servo
void setup() {
  myservo.attach(9);
  Serial.begin(115200);
}

void loop() {
  uint16_t val[] = {600, 1200, 1800};
  for (byte i = 0; i < 3; i++)
  {
    myservo.writeMicroseconds(val[i]);
    delay(1000);
    Serial.println(myservo.readMicroseconds());
  }
}`

Output
```
597
1197
1797
```

The same holds for reading and writing degrees or writing degrees and reading microseconds. I don't see any anomalies.

Cobalt6700, 2022-11-14:
@SpenceKonde Thanks very much for having a look into this. 

I will have a look at your version of Wire + have a go with your version of Servo later and see what happens. 

I used the code that @cattledogGH posted to confirm that it wasn't my sketch that was doing something silly - with the standard version of MegacoreX Servo I get the following output: 

uS
Write 600, Read 302
Write 1200, Read 602
Write 1800, Read 902

Deg
Write 60, Read 3
Write 90, Read 18
Write 120, Read 33. 

I'm wondering if there is something happening with my Arduino IDE install and using an incorrect library file? 
I will try removing Arduino IDE from my machine, reinstall and report back.

cattledogGH, 2022-11-14:
@Cobalt6700 
I am confused about which version of the servo library was compiled into my sketch. I now think it was Servo 1.1.8 which has support for the megaavr architecture. Downloaded from the Library Manager. The test code was not using Servo library packaged with MegaCoreX

```
Multiple libraries were found for "Servo.h"
Used: C:\Users\myName\Documents\Arduino\libraries\Servo
Not used: C:\Program Files (x86)\Arduino\libraries\Servo
Not used: C:\Users\myName\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.0\libraries\Servo
Using library Servo at version 1.1.8 in folder: C:\Users\myName\Documents\Arduino\libraries\Servo 
```
I will need to review the the version of the library used and how it differs from the MegaCoreX library with the timer set up, the clock source used, and the ticks to microsecond conversions.

You may be correct about the MegaCoreX Servo Library and the discrepency between written and read values.

Like you, I'll do some more checking.

Cobalt6700, 2022-11-14:
Had a go with a few different things tonight, will try to list logically. 

1. Re-tested on MegacoreX Servo Library, Confirmed that values are not written and read correctly on any move.

2. Re-tested on MegacoreX Servo Library with my added line of code `pulsewidth = pulsewidth * 2;` (from https://github.com/MCUdude/MegaCoreX/issues/162#issue-1426869980). 
Confirmed that values written and read correctly on all moves. I can write a change of one degree, read it back and use that value to increment. (from code in: https://github.com/MCUdude/MegaCoreX/issues/162#issuecomment-1313902161)

4. Replaced MegacoreX Servo Library with MegaTinyCore Servo Library. (had to comment out `#include <core_devices.h>` from Servo.h) Values written and read correctly on moves greater than 1 deg.

5.  Keeping MegaTinyCore Servo, Replaced MegacoreX core wiring files with with MegaTinyCore. Values written and read correctly on moves greater than 1 deg.

6. Re-installed Arduino megaavr from board manager. Values written and read correctly on moves greater than 1 deg. I think I'm going to open this 'missed degree' as an issue on the Arduino Servo Github.

7. Changed test code to write values in uS to increment servo by 1 deg - works on Arduino megaavr, MegaTinyCore and MegacoreX (with my added line).

MCUdude, 2023-04-13:
Sorry for not responding earlier. I found out that the issue was caused by a missing parenthesis in `ticksToUs`.

Can @Cobalt6700 and/or @cattledogGH give the updated library a try, please? 🙂 

```diff
diff --git a/megaavr/libraries/Servo/src/megaavr/Servo.cpp b/megaavr/libraries/Servo/src/megaavr/Servo.cpp
index 32c9693..2699c15 100644
--- a/megaavr/libraries/Servo/src/megaavr/Servo.cpp
+++ b/megaavr/libraries/Servo/src/megaavr/Servo.cpp
@@ -5,7 +5,7 @@
 
 #if (F_CPU > 10000000L)
   #define usToTicks(_us)    ((( (_us) / 2) * clockCyclesPerMicrosecond()))          // converts microseconds to tick
-  #define ticksToUs(_ticks) (((unsigned) _ticks * 2) / clockCyclesPerMicrosecond()) // converts from ticks back to microseconds
+  #define ticksToUs(_ticks) (((unsigned) (_ticks) * 2) / clockCyclesPerMicrosecond()) // converts from ticks back to microseconds
   #define TRIM_DURATION  37                                                         // compensation ticks to trim adjust for digitalWrite delays
 #else
   #define usToTicks(_us)    ((( _us ) * clockCyclesPerMicrosecond()))               // converts microseconds to tick
```

@SpenceKonde what's the deal with `TRIM_DURATION`? I didn't write this library, so my knowledge here is limited. When running an ATmega4809 at 16 MHz, I'm getting the absolute most accurate timing if I set `TRIM_DURATION` to zero. The higher the number, the quicker the on-time is.

* `myservo.writeMicroseconds(600)` leaves me with an on-time between 590 and 593us when setting `TRIM_DURATION` to 51 (same value as megaTinyCore uses)
* `myservo.writeMicroseconds(600)` leaves me with an on-time between 596 and 599us when setting `TRIM_DURATION` to 0

SpenceKonde, 2023-04-14:
> @SpenceKonde what's the deal with `TRIM_DURATION`? I didn't write this library, so my knowledge here is limited

And you think I did?! I have no clue what that does, so I avoided touching it with great care. you're saying it's wrong? Have you  tested this on mTC or just MCX - there are reasons why the correct trim value would be different!!

SpenceKonde, 2023-04-14:
As a serious answer, I think it's supposed to compesate for the mindblowingly long time that digitalWrite() takes to execute on AVR. But we may well have worked around this in one way or another.

cattledogGH, 2023-04-15:
I find that the library fix  does correct the value.
```
-  #define ticksToUs(_ticks) (((unsigned) _ticks * 2) / clockCyclesPerMicrosecond()) // converts from ticks back to microseconds
+  #define ticksToUs(_ticks) (((unsigned) (_ticks) * 2) / clockCyclesPerMicrosecond()) // converts from ticks back to microseconds
```
What's not clear is if the library version will change from 1.2.0 to 1.2.1 in the Arduino IDE without an update to the core version from 1.1.1

MCUdude, 2023-04-15:
I haven't released a boards manager version yet, so the MegaCoreX version number hasn't changed yet.

The library version number will change from 1.2.0 to 1.2.1, and the MegacoreX version number will change from 1.1.1 to 1.1.2

MCUdude, 2023-05-02:
Fixed in the latest boards manager release!

closed 2023-05-02 by MCUdude