MCUdude/MegaCoreX #163: Programming does not work with MegaCoreX

sfr-droid, 2022-11-20:
After compiling finished, at the point where it would start programming an error accour:

exit status 0xc000007b

I can not program one single board in the whole library because each time that error accours

per1234, 2022-11-20:
Maybe it is the same as https://github.com/MCUdude/MegaCoreX/issues/160

sfr-droid, 2022-11-21:
I'm pretty sure that its not more than some missing files and yes as you said in post #160, adding the libusb0.dll to the bin folder of the MegaCoreX Library will cure the error  0xc0000135 but not the error 0xc000007b. I also got the error 0xc0000135 before.

sfr-droid, 2022-11-24:
Finally I found a solution that works for the issues 0xc0000135 and 0xc000007b. The reason for this error is that avrdude is obviously missing some dll files. 

So all I did was download libusbk from: [https://sourceforge.net/projects/libusbk/files/libusbK-release/3.0.7.0/](url)

And the errors where gone. Hope this helps others that suffer also from this nasty problem

Cheers

closed 2022-11-24 by sfr-droid