MCUdude/MegaCoreX #165: I suggest giving the timer usage in the documentation

technoblogy, 2022-12-17:
It would be good to have a sentence in the documentation stating which timers are used by default for Arduino functions; I believe it's TCB2 for millis() and delay(), but it would be good to have a statement to that effect.

SpenceKonde, 2022-12-17:
Is someone spoiled by my essay-length documentation?

https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/Ref_Timers.md

technoblogy, 2022-12-17:
I'd settle for something less definitive than that!

cattledogGH, 2022-12-17:
The Timer used for millis() is given in the variants files with pins_arduino.h.

> I believe it's TCB2 for millis() and delay()

Depends on the processor/board. For the Nano Every and UnoWiFi using the 4809 its TCB3.

MCUdude, 2023-01-10:
Resolved in [8dca08](https://github.com/MCUdude/MegaCoreX/commit/8dca088f6f287fa83e43d2a85b883e3838c5318b)

closed 2023-01-10 by MCUdude