MCUdude/MegaCoreX #166: Do you need to do Burn Bootloader to change any of the options?

technoblogy, 2022-12-20:
It's not clear whether you have to do **Burn Bootloader** to change the fuses for any of the options, such as the clock frequency.

The only case where the documentation explicitly says you have to do **Burn Bootloader** is for the **EEPROM** option, so can I assume all the other options don't need it?

MCUdude, 2023-01-10:
Sorry for the late reply. You don't need to click "Burn Bootloader" as long as you're using a UPDI programmer. I'll remove the part in the README that states this is necessary for the EEPROM option.

MCUdude, 2023-01-10:
Resolved in [dabd52d](https://github.com/MCUdude/MegaCoreX/commit/dabd52d4bf78f91b96acbec94c9fd8f7449ed09c)

closed 2023-01-10 by MCUdude