MCUdude/MegaCoreX #167: ATmege4808: Arduino auto-detect libraries finds wrong Servo library -> Error: multiple definition of __vector_25

ArminJo, 2022-12-31:
I try to compile Servo code for a ATmega4808, as can be found [here](https://github.com/ArminJo/ServoEasing/actions/runs/3812060983/jobs/6485073837):
This is the error i get:
```
(.text+0x0): multiple definition of `__vector_25'
...
 Multiple libraries were found for "Servo.h"
  Used: /home/runner/Arduino/libraries/Servo
  Not used: /home/runner/.arduino15/packages/MegaCoreX/hardware/megaavr/1.1.0/libraries/Servo
Using library ServoEasing at version 3.1.0 in folder: /home/runner/work/ServoEasing/ServoEasing 
```
I get the same error on my windows Arduino 1.8 and 2.0 IDEs.
The compile error vanishes if I enable `USE_TIMERB1 `instead of `USE_TIMERB2 `in the Arduino Servo megaavr/ServoTimers.h file.

Do you know any workaround apart from deleting the Arduino Servo library manually?
And what is the use case of having a Servo library not used by Arduino? Or is this an Arduino library detection bug (which will never be fixed, I assume)?

Thanks for caring
Armin

per1234, 2022-12-31:
>is this an Arduino library detection bug

No. This is the expected result from the system working as designed:

https://arduino.github.io/arduino-cli/latest/sketch-build-process/#dependency-resolution

The two libraries have the same score for "architecture matching", "library name", and "folder name" priorities, so the tie-breaker is the "[location priority](https://arduino.github.io/arduino-cli/latest/sketch-build-process/#location-priority)" score:

> The "location priority" is determined as follows (in order of highest to lowest priority):
[...]
>3. The library is under the libraries subfolder of the IDE's sketchbook or Arduino CLI's user directory
>4. The library is bundled with the board platform/core ([`{runtime.platform.path}/libraries`](https://arduino.github.io/arduino-cli/0.28/platform-specification/#global-predefined-properties))

Something to note is most users will have an additional copy of the global Servo library installed in a separate location. This location has lower priority than the platform bundled library:

> 6. The library is bundled with the Arduino IDE (this location is determined by the Arduino CLI configuration setting `directories.builtin.libraries`)

This means that after uninstalling the Servo library from the sketchbook folder, a global installation of Servo library would still be available when compiling for other boards without a platform bundled Servo library, while the platform bundled library gets priority when compiling for a MegaCoreX board.

>Do you know any workaround apart from deleting the Arduino Servo library manually?

For [your GitHub Actions workflow](https://github.com/ArminJo/ServoEasing/blob/master/.github/workflows/LibraryBuild.yml), simple remove `Servo` from [the `env.REQUIRED_LIBRARIES` key](https://github.com/ArminJo/ServoEasing/blob/fc34e8d1b60ffb25fd4a1b160c4d500bb3058aab/.github/workflows/LibraryBuild.yml#L36) and add it to the `matrix.required-libraries` key ([example](https://github.com/ArminJo/ServoEasing/blob/fc34e8d1b60ffb25fd4a1b160c4d500bb3058aab/.github/workflows/LibraryBuild.yml#L74)) of the boards that don't have a platform bundled version of the Servo library.

As for normal human usage of MegaCoreX, the workaround would be to add a header file with a unique name (e.g., `ServoMegaCoreX.h` to the platform bundled library). If the program contains an `#include` directive for this header before the `#include` directive for `Servo.h`, this will cause the intended platform bundled library to be discovered instead of the one in the sketchbook. The header could be empty, or even act as a substitute for including `Servo.h`:

**ServoMegaCoreX.h**
```cpp
#include <Servo.h>
```

ArminJo, 2022-12-31:
> https://arduino.github.io/arduino-cli/latest/sketch-build-process/#dependency-resolution

Thanks for the link, I searched for it for half an hour, before starting this issue.

> For [your GitHub Actions workflow](https://github.com/ArminJo/ServoEasing/blob/master/.github/workflows/LibraryBuild.yml), simple remove Servo from [the env.REQUIRED_LIBRARIES [...]

This indeed works for me 😀 👍  T H A N K S ! ! !

@MCUdude 
How can I decide, that the MEGAX kernel is used?
I am looking for something like [this](https://github.com/ArminJo/DigistumpArduino/blob/master/digistump-avr/platform.txt#L15)
#if defined MEGAXCORE

@ All
Happy new year

MCUdude, 2023-01-10:
> How can I decide, that the MEGAX kernel is used?

It's all there in the README:
https://github.com/MCUdude/MegaCoreX#identifying-megacorex

closed 2023-01-10 by MCUdude

ArminJo, 2023-01-11:
Thanks for adding ServoMegaCoreX.h 👍