MCUdude/MegaCoreX #168: MegaCoreX v1.1.1 boards manager release not installable

MCUdude, 2023-01-10:
@per1234 maybe you can help here? I've just released MegaCoreX v1.1.1, where I've also updated the toolchain in the boards manager json file to Avrdude 7.1. I've triple checked the Avrdude 7.1 checksums, and the _should_ be correct. However, I'm still getting a "CRC doesn't match" error when trying to install v1.1.1. Do you know what the issue might be?

[image]

ArminJo, 2023-01-11:
Here https://github.com/ArminJo/ServoEasing/actions/runs/3892044344/jobs/6642994213
I get another error
`Error during install: Cannot install tool MegaCoreX:avrdude@7.1-arduino.1: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 1169286 != 3696640`

MCUdude, 2023-01-11:
For some reason, it seems like the size and checksum for Avrdude 7.1 I calculated on my mac last night are incorrect for some reason. I'll look into it later today

MCUdude, 2023-01-11:
Everything should be good now. When I downloaded the avrdude tar.gz files, my mac automatically extracted them into .tar files. But I finally figures out what I did wrong. Enjoy the v1.1.1 release!

closed 2023-01-11 by MCUdude

ArminJo, 2023-01-11:
Thanks for the 1.1.1 release.
I can confirm, it works now!

cattledogGH, 2023-01-11:
I just tried to update from 1.1.0 to 1.1.1 with Arduino 2.0.3 and received the following error message.
`Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
Tool arduino:arduinoOTA@1.3.0 already installed
Downloading packages
MegaCoreX:avrdude@7.1-arduino.1
MegaCoreX:megaavr@1.1.1
Installing MegaCoreX:avrdude@7.1-arduino.1
Failed to install platform: MegaCoreX:megaavr.
Error: 13 INTERNAL: Cannot install tool MegaCoreX:avrdude@7.1-arduino.1: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 1591591 != 4945920`

MCUdude, 2023-01-11:
> fetched archive size differs from size specified in index: 1591591 != 4945920

I've fixed a size and checksum issue in the MegaCoreX boards manager JSON file, so you should find the `Arduino15` folder on your computer and delete the MegaCoreX boards manager JSON file. Make sure to delete both of them. The next time you open the IDE, it will download a fresh JSON file, and then it should work 🙂

cattledogGH, 2023-01-14:
>  find the Arduino15 folder on your computer and delete the MegaCoreX boards manager JSON file. Make sure to delete both of them.

In the Arduino 15 folder I certainly find
`package_MCUdude_MegaCoreX_index   JSON File`

What other file is the "both of them" referencing?