MCUdude/MegaCoreX #171: Analogue ports reading

ednieuw, 2023-02-07:
When the analogue port are defined with a number the readings of this port number is always 255.
When defining with an A before the port number results are fine.

The Arduino Every board compiles fine without the A.

In the coding example below the results of analogue port 2 is always 255
Like   ` const int sensorPin = 2; `

When 
  ` const int sensorPin = A2; `
readings are fine 

`
const int sensorPin = 2; //A2;         // Readings are now always 255
       // const int sensorPin = A2;    // Readings are now fine
int sensorValue = 0;                       // the sensor value

void setup() {  
 Serial.begin(9600);                     
 Serial.println("\n*********\nSerial started"); 
 #if defined(MEGACOREX)
      Serial.println("Compiled with MegaCore X"); 
 #else 
     Serial.println("Compiled with Arduino MegaAVR");   
#endif
}

void loop() {
sensorValue = analogRead(sensorPin);
Serial.print("LDR: ");
Serial.println(sensorValue);
delay(100);
}`

MCUdude, 2023-02-07:
I'll assume you're using the Nano Every?

> When the analogue port are defined with a number the readings of this port number is always 255.
When defining with an A before the port number results are fine.

It's always to either use the digital pin number or use the A constants. The reason why you can't use 0 to 15 as analog pin values with the Nano Every makes perfect sense if you look at the pinout below.

Since MegaCoreX enables _all_ available analog inputs to the user (unlike the official Arduino core), there are suddenly an overlap between the digital and analog pin number.

Does `analogRead(3)` means digital pin 3 or analog input 3? This is impossible for the compiler to determine.

[image]

However, on other pinouts where the analog pin numbers don't overlap the digital ones, like when using the 48-pin standard pinout, you can use `analogRead(3)` or any other number below 16, because none of the digital pins below 16 has analog input capabilities.

I hope this makes sense 🙂

ednieuw, 2023-02-07:
Thanks Hans,Yes, a Nano Every. This was a nice pitfall. I can live with it. Thanks for the quick replyEd

closed 2023-02-07 by MCUdude