MCUdude/MegaCoreX #174: Bootloader burns just fine, uploaded simple program is not working

yeayer, 2023-03-28:
``Hello, I am trying to get ATmega4809 on Arduino Nano Every working with serial programmer. Thing I have done:
1. Removed level shifting transistors that connected SAMD11 used as UPDI programmer to Atmega4809, also removed pullup R on UPDI
2. Created a working programmer from Arduino Nano with Atmega328P
3. Burned bootloader Optiboot with serial3 on alternative pins (to match the Every Nano so I can use the PCB traces)

So my issue is - I can burn bootloader via Arduino Nano to the ATmega4809. When I dump the memory to a file, it is exactly what the bootloader hex from your code looks like. When I try uploading through serial (FT231X connected to mega4809 on Serial3 alt pins), it uploads just fine, reads just fine but the code simply does not run. I checked the fuses, dumped the hex (bootloader is still here, tried to upload also with no bootloader, whatever...). When I tried a sketch as simple as this:

`#include <Arduino.h>
bool test;
void setup() 
{
  pinMode(5,OUTPUT);
}

void loop() 
{
  delay(500);
  test = !test;
  digitalWrite(5,test);
}`

, it simply will not do anything. Tried using serialPrint, SSD1306 display, also nothing. Also I tried uploading with PlatformIO on VScode, it deletes the bootloader but the sketch gets uploaded, also does not work. When I purposefully "messed" the pin operations up in the code, on some versions the LED lit up with a very low current (I guess something got really wrong). Is there anything I forgot? I tried also not using the Arduino pinMode and digitalWrite functions, I just set the registers directly, with also no luck. Is there anything I am forgetting from the HW point? Do I need to add some resistors, etc? It drives me crazy as I have already written a long code that works on Nano Every, but I want to use my own boards and I do not want another chip (SAMD11) here. Thanks in advance!

MCUdude, 2023-03-28:
Can you upload the hex file you're using?

When using a bootloader, the program needs to start at address 0x200, because the bootloader uses the first 512 bytes in the flash memory.

yeayer, 2023-03-28:
Here it is.
[readtest_blink.txt](https://github.com/MCUdude/MegaCoreX/files/11087865/readtest_blink.txt)

MCUdude, 2023-03-28:
The problem is that the code isn't compiled for use with a bootloader. If you look at how [Intel hex files works](https://en.wikipedia.org/wiki/Intel_HEX), you can see that your program starts at 0x0000 and not 0x0200.

You need to compile your program with the bootloader option enabled in Arduino IDE.

BTW, are you using Arduino IDE or PlatformIO to compile your code?

yeayer, 2023-03-28:
For this one I used Arduino IDE with following settings:
Board: "ATmega4809"
Clock: "Internal 16MHz"
BOD: "BOD 2.6V"
EEPROM: "EEPROM retained"
Pinout: "Nano Every" or I use "48pin", does not change the outcome
Reset pin: "Reset"
Bootloader: "Optiboot (UART3 alternative pins)"

So how do I make the program start on 0x200 please? I thought that when I just choose the bootloader it will automatically start here, I can see that the hex file I sent in previous post contains the bootloader in the beginning...

yeayer, 2023-03-28:
If I understand it correctly, the first lines are the bootloader and then the code appended starts on :2002000050C0000068C0000066C0000064C0000062C0000060C000005EC000005CC00000E0, which by the link you sent means address 0x200, right?

MCUdude, 2023-03-28:
Can you post the actual hex file the compiler spits out, not the flash dump?

yeayer, 2023-03-28:
[test.ino.hex.txt](https://github.com/MCUdude/MegaCoreX/files/11088309/test.ino.hex.txt)
[test.ino.with_bootloader.hex.txt](https://github.com/MCUdude/MegaCoreX/files/11088317/test.ino.with_bootloader.hex.txt)

Should be these, one with and one without bootloader.

MCUdude, 2023-03-28:
Does the sketch that doesn't contain the bootloader run if you upload it using a UPDI programmer?

Also, I'd like to see the current fuse settings.

Use `-U fuses:r:-:h` to dump all fuses

yeayer, 2023-03-28:
-patmega4809 -cjtag2updi -P/dev/ttyUSB2 -U fuses:r:-:h

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: reading fuses memory ...
avrdude: writing output file <stdout>
0x0,0x54,0x1,0xff,0x0,0xc9,0x6,0x0,0x2

avrdude done.  Thank you.

----------------------------------
When I switch from board ATmega4809 to Arduino Every and upload it using UPDI (Arduino Nano as jtag2updi), the sketch works and LED blinks.

yeayer, 2023-03-28:
Now I did a test. I have tried to compile the code as for Arduino Every, took that one and manually created a .hex file with fuse setting for bootend 0x02. The code alone works, but as soon as the bootloader preceeds it, it does not work.

yeayer, 2023-03-28:
Sorry for confusion, by the way, I really appreciate your support, thanks for the immediate replies!
Here are the steps I have done to get to this point:
1: I have modified the original Arduino Nano Every with ATmega4809 like this:
https://content.arduino.cc/assets/NANOEveryV3.0_sch.pdf?_gl=1*17exlmw*_ga*MTk3NTMzOTE2MS4xNjQ2Mzk2ODUy*_ga_NEXN8H46L5*MTY3OTU2Nzc1MC4xMzQuMS4xNjc5NTY3NzYxLjAuMC4w

I have removed Q1 and Q2 to disconnect UART and UPDI between ATmega4809 and SAMD11. Then, I have added a serial resistor to UPDI with a value of 4k7. After checking the signal with a scope, I found out that UPDI does not work so I tried removing R19 (4k7 PULLUP on UPDI) and it started working. I did this after step 2, but if you have Arduino Nano Every and Arduino Nano to try this, I have summed it in here so it can be done at once.

2. I have created a JTAG2UPDI programmer per this tutroial: https://daumemo.com/diy-updi-usb-programmer-which-can-be-made-with-cheap-hardware/
3. Then I tried burning bootloader Optiboot with UART3 on alt. pins using the programmer from step2 and it worked, bootloader was burned successfully in ATmega4809.
4. After this, I tried uploading a blink sketch - and it did not work - upload and fuse programming was successful, but the LED did not blink.
5. I thought my Arduino Nano makeshift programmer was at fault, so I tried switching the Board option back to Arduino Nano Every. I was thinking it will not work, because it could have different VID/PID than the original SAMD11, but it worked and the LED started blinking, so the blink sketch program was being executed correctly. So I found out that my programmer works. Then I burned the bootloader again to ATmega4809 (because it got overwritten by the previous programming of the blink sketch using a standard procedure to upload a normal non-modified Nano Every).
6. After this, I found out that if I compile the blink sketch with these options:

Board: "Arduino Nano Every"
Registers Emulation: "None(ATMEGA4809)"

And sketch with options using MegaCoreX with no bootloader:

Board: "ATmega4809"
Clock: "Internal 16MHz"
BOD: "BOD 2.6V"
EEPROM: "EEPROM retained"
Pinout: "Nano Every" or I use "48pin", does not change the outcome
Reset pin: "Reset"
Bootloader: "No Bootloader"

will result in two different hex files.

So I took the original work
ing blink sketch hex compiled with default Nano Every options 
[Blink_test.ino.hex.txt](https://github.com/MCUdude/MegaCoreX/files/11089809/Blink_test.ino.hex.txt) ,
then I took the bootloader itself 
[Optiboot_mega0_UART3_ALT_115200_A7.hex.txt](https://github.com/MCUdude/MegaCoreX/files/11089818/Optiboot_mega0_UART3_ALT_115200_A7.hex.txt)
and combined the into one hex (sorry I was using the dumps which were written in shorter code, because I needed to change the checksum for each line). I have just appended the working code and changed the starting address to 0x200 and the for every next line I changed the address accordingly and also the checksum, so it looks like this 
[test_with_bootloader.hex.txt](https://github.com/MCUdude/MegaCoreX/files/11089850/test_with_bootloader.hex.txt)
. However, with this one, the bootloader works (Iam able to upload it using the serial3, but it will not blink the LED). My theories are these:
1. The serial3 is somehow through some interrupt, voltage level on TX or RX keeping the ATmega4809 in the programming mode and it will not somehow reset it or make it jump to the application code.
2. There is an issue with the bootloader itself on this ATmega
3. My CPU is somewhat broken or the FT231X is not a good IC to choose as a serial programmer for the ATmega4809

I am using an FT231X as a USB-Serial adapter for ATmega4809, so I might try changing it to CH340 or something like that if it solves the issue.

MCUdude, 2023-03-28:
I managed to get it to work.

I'm using the built-in jtag2updi programmer on the Nano Every, since I didn't want to modify the board. I'm using `stty` to trigger a "1200bps touch" in order to enter "jtag2updi mode". I've also connected a USB to serial adapter to UART1 ALT (Nano every pins 0 and 1).

I've build the latest Avrdude on my computer, which lets me connect the DTR/RTS line on my USB to serial adapter directly to the Nano Every reset pin without the need for a capacitor. This feature will be available in Avrdude 7.2.

(I know this because I'm one of the active Avrdude developers at the moment. Pro tip: Play around with Avrdude terminal mode in Avrdude 7.1 and read the docs. It's an incredible tool that gives you r/w byte access to any memory, and it's also very powerful.  Data can also be piped into Avrdude terminal mode)
```
$ avrdude -cjtag2updi -P /dev/ttyUSB0 -patmega4809 -t
$ echo "read eeprom 0 0x200" | avrdude -cjtag2updi -P /dev/ttyUSB0 -patmega4809 -t
```

Burn the bootloader to the Nano Every using the on-board jtag2updi and set the correct fuses:
```
$ stty -f /dev/cu.usbmodem14101 1200 && ./avrdude -patmega4809 -cjtag2updi -P/dev/cu.usbmodem14101 -e -Ufuses:w:0x00,0x54,0x01,0xff,0x00,0b11001001,0x06,0x00,0x02:m -Ulock:w:0xC5:m -Uflash:w:/Users/hans/Documents/Arduino/MegaCoreX/megaavr/bootloaders/optiboot/bootloaders/mega0/115200/Optiboot_mega0_UART1_ALT_115200_A7.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: erasing chip
avrdude: reading input file 0x00,0x54,0x01,0xff,0x00,0b11001001,0x06,0x00,0x02 for fuses
         with 9 bytes in 1 section within [0, 8]
         using 1 page and 1 pad byte
avrdude: writing 9 bytes fuses ...
avrdude: 9 bytes of fuses written
avrdude: verifying fuses memory against 0x00,0x54,0x01,0xff,0x00,0b11001001,0x06,0x00,0x02
avrdude: 9 bytes of fuses verified
avrdude: reading input file 0xC5 for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xC5
avrdude: 1 byte of lock verified
avrdude: reading input file /Users/hans/Documents/Arduino/MegaCoreX/megaavr/bootloaders/optiboot/bootloaders/mega0/115200/Optiboot_mega0_UART1_ALT_115200_A7.hex for flash
         with 468 bytes in 2 sections within [0, 0x1ff]
         using 4 pages and 44 pad bytes
avrdude: writing 468 bytes flash ...

Writing | ################################################## | 100% 0.38 s 

avrdude: 468 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Documents/Arduino/MegaCoreX/megaavr/bootloaders/optiboot/bootloaders/mega0/115200/Optiboot_mega0_UART1_ALT_115200_A7.hex

Reading | ################################################## | 100% 0.06 s 

avrdude: 468 bytes of flash verified

avrdude done.  Thank you.
```

Test program flashed using the bootloader:
```
$ ./avrdude -carduino -P /dev/cu.usbserial-1420 -patmega4809 -D -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_146207/Blink.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_146207/Blink.ino.hex for flash
         with 730 bytes in 1 section within [0x200, 0x4d9]
         using 6 pages and 38 pad bytes
avrdude: writing 730 bytes flash ...

Writing | ################################################## | 100% 0.06 s 

avrdude: 730 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_146207/Blink.ino.hex

Reading | ################################################## | 100% 0.03 s 

avrdude: 730 bytes of flash verified

avrdude done.  Thank you.
[... 61 lines cut ...]
:1003B000290A80932C0A80932A0A8093280A8BE0CA
:1003C0008093000A83E08093E505E0E8FAE037E0F7
:1003D0002EEF90E885E0318324879587808370969F
:1003E000E03C4AE0F407B9F71092B10A8FE79EE3C8
:1003F0008093BC0A9093BD0A8091B50A8160809376
:10040000B50A1092B00A8091B00A81608093B00A58
:10041000789484E08093810401E014E0C0E0D0E0AF
:10042000809104288027809304289091800492FF73
:100430002AC0811125C01093860443DF6B017C0123
:1004400020E2C20E21EAD21E27E0E21EF11C40E2A9
:10045000C41641EAD40647E0E406F10430F431DF83
:10046000C616D706E806F906D0F32BDF6C157D0516
:100470008E059F05D0F3209799F20E940000D0CFFF
:1004800010938504DACF9FB7F894811107C080914B
:100490009204877F809392049FBFCFCF8091920474
:1004A0008860F8CF09D0A59F900DB49F900DA49FB0
:1004B000800D911D11240895A29FB001B39FC0012A
:1004C000A39F700D811D1124911DB29F700D811D80
:0A04D0001124911D0895F894FFCF48
:0400000300000200F7
:00000001FF

```

</details>

MCUdude, 2023-03-28:
> Also, I found out that I can not read fuses through the serial3:

Correct. Fuses can only be read/written using a UPDI programmer.

yeayer, 2023-03-29:
Unfortunately, when I choose pinout "Nano Every", it throws an error, I was always using "48 pin".

`.arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/avrdude.conf -v -patmega4809 -carduino -P/dev/ttyUSB4 -e -Ufuses:w:0x00,0x54,0x01,0xff,0x00,0b11001001,0x06,0x00,0x02:m -Ulock:w:0xC5:m -Uflash:w:/tmp/arduino_build_518265/spiffs_test.ino.hex:i 

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/tlukac/.arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/avrdude.conf
         User configuration file is /home/tlukac/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB4
         Using Programmer              : arduino
         AVR Part                      : ATmega4809
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         64   64      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   64      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      49152  128      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 3
         Firmware Version: 9.0

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9651 (probably m4809)
avrdude: erasing chip
avrdude stk500_chip_erase() error: chip erase instruction not defined for part ATmega4809

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected
`

yeayer, 2023-03-29:
I have tried your solution, connected the programmer to UART1 alt pins and it works. So I think there must be something wrong with using UART3 ALT pins for programming for sure. Can you please try to reproduce my steps? You can just remove the two dual transistors and solder RX and TX to the exposed pads. For some reason, it is not working. Also, I would like to ask - how to get the serial communication working now? Serial.begin and Serial.println will dimm the LED when I try using it on pin D7 (PIN_PA1) and Serial1.println() doesnt print anything...

yeayer, 2023-04-13:
Hello, so I am currently using it as per your information. For anyone reading this in the future I will add some info:
Use UART1 alternative pins for bootloader, also use the same for RX/TX for USB serial console. However, in the code, when using Serial.print(), use Serial1.print(); to get it working. for init, you need to use Serial1.swap(1); . When you want to use same SPI pins as on Arduino Nano Every with mega4809, you need to write SPI.swap(2); in the beginning of your code. 

Also I have two last questions for @MCUdude : 
1. When will be the avrdude 7.2 ready with implementation for the DTR/RTS to reset pin ready?
2. How does the HW implementation of DTR/RTS pin to MEGA4809 RESET pin look like? Is it just straight connected, or is tehre any pullup/pulldown/cap?

In the end, I would like to thank both of you guys responding there in the thread for the information an in the future, if there is a possibility, please try to test the UART3 alt pins for bootloader - I am still not able to get it working.

mcuee, 2023-05-12:
@yeayer 

If you want avrdude 7.2 release to be sooner than later, please help to carry out tests aginst the chips/programmers you have.

You can check the discussion here.
* https://github.com/avrdudes/avrdude/discussions/1257

mcuee, 2023-05-12:
Similar discussion of Nano Every here.
* https://github.com/MCUdude/MegaCoreX/issues/156

I was able to use the Optiboot-x bootloader provided by @MCUdude last time on the Nano Every (same here as I also do not want to change it). I have not tried it for a while though.

gemforce, 2025-02-01:
And what if I simply unsolder the chip from the arduino every and connect the updi rx tx gnd vcc pins to the external chip?