MCUdude/MegaCoreX #175: TimerOne library not working

Hamza-arch, 2023-04-04:
I am working on automating interrupts with time interval. I found that TimerOne is a good library for it but it is not working with Atmega 4809. Here is link to the library (https://github.com/PaulStoffregen/TimerOne). 
The error given by arduino compiler is given below. I want to activate functions after certain time intervals. Can you kindly guide me.
"

WARNING: library TimerOne-master claims to run on avr architecture(s) and may be incompatible with your current board which runs on megaavr architecture(s).
In file included from C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\atmega4809\meanfunction_inside_loop\meanfunction_inside_loop.ino:1:0:
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h: In member function 'void TimerOne::initialize(long unsigned int)':
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:184:2: error: 'TCCR1B' was not declared in this scope
  TCCR1B = _BV(WGM13);        // set mode as phase and frequency correct pwm, stop the timer
  ^~~~~~
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:184:2: note: suggested alternative: 'TCB1'
  TCCR1B = _BV(WGM13);        // set mode as phase and frequency correct pwm, stop the timer
  ^~~~~~
  TCB1
In file included from c:\users\arshad computer lab\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:99:0,
                 from c:\users\arshad computer lab\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\pgmspace.h:90,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/String.h:30,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/IPAddress.h:25,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/Client.h:22,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/ArduinoAPI.h:29,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/Arduino.h:23,
                 from sketch\meanfunction_inside_loop.ino.cpp:1:
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:184:15: error: 'WGM13' was not declared in this scope
  TCCR1B = _BV(WGM13);        // set mode as phase and frequency correct pwm, stop the timer
               ^
In file included from C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\atmega4809\meanfunction_inside_loop\meanfunction_inside_loop.ino:1:0:
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:185:2: error: 'TCCR1A' was not declared in this scope
  TCCR1A = 0;                 // clear control register A
  ^~~~~~
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:185:2: note: suggested alternative: 'TCB1'
  TCCR1A = 0;                 // clear control register A
  ^~~~~~
  TCB1
In file included from c:\users\arshad computer lab\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:99:0,
                 from c:\users\arshad computer lab\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\pgmspace.h:90,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/String.h:30,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/IPAddress.h:25,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/Client.h:22,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/ArduinoAPI.h:29,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/Arduino.h:23,
                 from sketch\meanfunction_inside_loop.ino.cpp:1:
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h: In member function 'void TimerOne::setPeriod(long unsigned int)':
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:191:25: error: 'CS10' was not declared in this scope
   clockSelectBits = _BV(CS10);
                         ^
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:191:25: note: suggested alternative: 'A10'
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:195:25: error: 'CS11' was not declared in this scope
   clockSelectBits = _BV(CS11);
                         ^
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:195:25: note: suggested alternative: 'A11'
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:199:25: error: 'CS11' was not declared in this scope
   clockSelectBits = _BV(CS11) | _BV(CS10);
                         ^
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:199:25: note: suggested alternative: 'A11'
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:199:37: error: 'CS10' was not declared in this scope
   clockSelectBits = _BV(CS11) | _BV(CS10);
                                     ^
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:199:37: note: suggested alternative: 'A10'
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:203:25: error: 'CS12' was not declared in this scope
   clockSelectBits = _BV(CS12);
                         ^
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:203:25: note: suggested alternative: 'A12'
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:207:25: error: 'CS12' was not declared in this scope
   clockSelectBits = _BV(CS12) | _BV(CS10);
                         ^
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:207:25: note: suggested alternative: 'A12'
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:207:37: error: 'CS10' was not declared in this scope
   clockSelectBits = _BV(CS12) | _BV(CS10);
                                     ^
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:207:37: note: suggested alternative: 'A10'
[... 180 lines cut ...]
  ^~~~~~
  TIMERB1
In file included from c:\users\arshad computer lab\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:99:0,
                 from c:\users\arshad computer lab\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\pgmspace.h:90,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/String.h:30,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/IPAddress.h:25,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/Client.h:22,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/api/ArduinoAPI.h:29,
                 from C:\Users\ARSHAD COMPUTER LAB\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.1.1\cores\coreX-corefiles/Arduino.h:23,
                 from sketch\meanfunction_inside_loop.ino.cpp:1:
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:281:15: error: 'TOIE1' was not declared in this scope
  TIMSK1 = _BV(TOIE1);
               ^
In file included from C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\atmega4809\meanfunction_inside_loop\meanfunction_inside_loop.ino:1:0:
[... cut ...]
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:288:2: error: 'TIMSK1' was not declared in this scope
  TIMSK1 = 0;
  ^~~~~~
C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:288:2: note: suggested alternative: 'TIMERB1'
  TIMSK1 = 0;
  ^~~~~~
  TIMERB1
exit status 1
Error compiling for board ATmega4809.
Invalid library found in C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\Elemyo-library-master_nano: no headers files (.h) found in C:\Users\ARSHAD COMPUTER LAB\Documents\Arduino\libraries\Elemyo-library-master_nano

MCUdude, 2023-04-04:
The TimerOne library doesn't support the ATmega4809, but you could try this instead 
https://github.com/Kees-van-der-Oord/Arduino-Nano-Every-Timer-Controller-B

Hamza-arch, 2023-04-04:
thanks for the support. I am getting this error when I used this library. Can you help?
wiring.c.o (symbol from plugin): In function `initVariant':
(.text+0x0): multiple definition of `__vector_25'
libraries\EveryTimerB\EveryTimerB.cpp.o (symbol from plugin):(.text+0x0): first defined here
c:/users/arshad computer lab/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ATmega4809.

SpenceKonde, 2023-04-04:
TimerOne is a Paul Stoffregen project. He's either lost interest in the whole thing, or more likely has a chip on his shoulder about the modern AVRs, I've had a PR that would fix another library of his for all current forseeable modern AVR devices with up to 56 port pins and on devices with more, if they were ever released, my code would probably still work fine for the first 56 pins. 

Re: second problem,
You have two ISR routines defined for a TCB. My guess is that one of them is from that EveryTimerB library, and the other is probably the millis timekeeping isr. 
If EveryTimerB does what it sounds like it says, that library may be fundamentally incompatible with MegaCoreX because he only supported using a TCB for millis, and so a library that defines all TCBs can never work there. 

That's why in my modern AVR cores, among other things, I provide the user the option to select the millisecond timekeeping timer (TCAs or TCBs only on Dx-series, TCD0 orthe RTC as well on megaTinyCore - the TCD0 because of the added clocking options is a much more complicated peripheral in practice on Dx, and RTC because I am not happy with the result on mTC (what's going to happen is my SmeepLib.h will be released (for all modern AVRs, assuming I don't find anything that would make some of the tricks unviable to implement in that context), and it will combine the ability to use micros (you lose that on mtc if you go RTC) with the timekeeping in sleep, which was the point of RTC millis, achieved by wrappers that would be called around the sleep related functions. and upon wakeup, if it was not woken by the thing that was supposed to put it into active mode we just increment a count of ovf's and and reset the flag that says to go back to sleep, otherwise, when it's finally time to awaken (say a pin interrupt on a specific port happened, that ISR would run normally, wouldn't set our go back to sleep flag, causing us to instead calculate the number of milliseconds that was spent sleeping and add that to millis count. smeepLib docs will note that micros will get out of sync with millis, but that if this matters in your code, or if you have a tmeout implemented with micros() while you're going to sleep, that's a bug in your code. Nobody ever said micros = 1000* millis, and in fact even under normal operation many cores will violate this depending on clock speed)

Hamza-arch, 2023-04-04:
Thank you for the detailed answer and support. But pardon me as I am not an advance level porgrammer and could not understand many points. What I have understand is that:

1. EveryTimerB library is not compatible with MegaCoreX
2. SpenceKonde is working on a new library smeeplib which will make it possible to use with MegaCoreX and other modern AVR boards. 
3. For now, we need to use sophisticated codes if we want to use timer interrupts.

Can you kindly guide me how I can use timer interrupt in my code for now. I need to read value of a sensor while doing other activities and decisison making in loop.

cattledogGH, 2023-04-05:
I have often uses this library for timer interrupts on a Nano Every (at4809).
[https://github.com/khoih-prog/megaAVR_TimerInterrupt available through library manager](url)

The library example for periodic Timer interrupts is
TimerInterruptTest.ino

You can specify the timers to be used.
I have used this library to set up Timer interrupts to drive stepper  motors at different rpm.

Hamza-arch, 2023-04-05:
@cattledogGH it worked. Thanks.