MCUdude/MegaCoreX #176: VENDOR_ID and CORE_ID ?

mbmorrissey, 2023-04-30:
Hi - I'm trying to make a custom board specification based on MegaCoreX.  Sorry, I'm sure this isn't an 'issue' with MegaCoreX, but it is with my understanding!

If I understand right from:
[https://arduino.github.io/arduino-cli/0.32/platform-specification/](url)
(most of the way down, under the header " Referencing another core, variant or tool")
I should be able to just use a board specification and use .build.core to reference MegaCoreX for the rest.

I'm struggling to actually link to MegaCoreX.  I think it may be that I don't know what to put for the VENDOR_ID and CORE_ID.

I currently have 
```
cadalogger.build.core=MCUdude:MegaCoreX
```
but that is a bit of a guess.

This is my first attempt at a board specification, and with many false starts I seem to be getting my .json file recognised and it is linking to my .zip archive and thinks it is installing successfully, but on compiling I get
```
FQBN: cadalogger:avr:cadalogger
Using board 'cadalogger' from platform in folder: /Users/mbm5/Library/Arduino15/packages/cadalogger/hardware/avr/0.0.2
Using core 'coreX-corefiles' from platform in folder: /Users/mbm5/Library/Arduino15/packages/cadalogger/hardware/avr/0.0.2

Detecting libraries used...

recipe.preproc.macros pattern is missing

Compilation error: recipe.preproc.macros pattern is missing
```

So, I think I am somwhat linking to MegaCoreX, but somehow it is expecting it as part of my package, rather than referring to my installation of MegaCoreX?  Could this be a matter of me having VENDOR_ID and CORE_ID wrong?  Or is there something else I may be doing wrong?

(n.b., I also don't have any programmer options in the tools menu, which I was hoping would come thorugh courtesy of MegaCoreX)

Help massively appreciated!!

Michael

per1234, 2023-04-30:
Hi @mbmorrissey 

>I currently have
>
>```text
>cadalogger.build.core=MCUdude:MegaCoreX
>```

It should be:

>```text
>cadalogger.build.core=MegaCoreX:coreX-corefiles
>```

"**VENDOR_ID**" is set by the folder name where the referenced platform is installed. If you [installed it via Boards Manager](https://github.com/MCUdude/MegaCoreX#boards-manager-installation), then the folder name is set by the value of the `packages[*].name` key of the [package index file](https://arduino.github.io/arduino-cli/latest/package_index_json-specification/):

https://github.com/MCUdude/MegaCoreX/blob/dfd2d119783c52f49f61a3ada5c79e5394c549e9/package_MCUdude_MegaCoreX_index.json#L4

If you [installed the platform manually](https://github.com/MCUdude/MegaCoreX#manual-installation) then you might have chosen any arbitrary folder name to install it under (e.g., `MegaCoreX-master`).

"**CORE_ID**" is set by the folder name of the core:

https://github.com/MCUdude/MegaCoreX/tree/master/megaavr/cores/coreX-corefiles

> I also don't have any programmer options in the tools menu, which I was hoping would come thorugh courtesy of MegaCoreX

They will:

https://arduino.github.io/arduino-cli/latest/platform-specification/#core-reference

> The [programmers](https://arduino.github.io/arduino-cli/dev/platform-specification/#programmerstxt) from the referenced platform are made available

mbmorrissey, 2023-04-30:
Thanks!  I'll try it now.

mbmorrissey, 2023-04-30:
That was a massive help, thanks.  Not quite there, but a lot of progress!  With your advice and a couple of more rounds of trial and error, I quite clearly have it talking to my installation of MegaCoreX!  It now makes much more progress toward compiling, and the programmer options show now too.

Now, when compiling a blank program (which works if I select MegaCoreX in 'tools'), when I try to use my board

```
FQBN: cadalogger:megaavr:cadalogger
Using board 'cadalogger' from platform in folder: /Users/mbm5/Library/Arduino15/packages/cadalogger/hardware/megaavr/0.0.4
Using core 'coreX-corefiles' from platform in folder: /Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1

Detecting libraries used...
/Users/mbm5/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -I/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/api/deprecated -I/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles /private/var/folders/hs/k40w7dn54r7fvvk_7ktssbfr0000gp/T/arduino/sketches/47740E45E6F354FA7BA26B2405BDC3CB/sketch/sketch_apr30a.ino.cpp -o /dev/null
In file included from /Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/Arduino.h:127:0,
                 from /private/var/folders/hs/k40w7dn54r7fvvk_7ktssbfr0000gp/T/arduino/sketches/47740E45E6F354FA7BA26B2405BDC3CB/sketch/sketch_apr30a.ino.cpp:1:
/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/UART.h:28:10: fatal error: pins_arduino.h: No such file or directory
 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
Alternatives for pins_arduino.h: []
ResolveLibrary(pins_arduino.h)
  -> candidates: []

exit status 1

Compilation error: exit status 1
```

Not sure if it is helpful, but here is my present boards.txt file:
```

menu.clock=Clock
menu.BOD=BOD
menu.eeprom=EEPROM
menu.bootloader=Bootloader

cadalogger.name=cadalogger
cadalogger.build.core=MegaCoreX:coreX-corefiles
cadalogger.upload.tool=avrdude
cadalogger.upload.maximum_data_size=6144
cadalogger.upload.speed=115200
cadalogger.bootloader.tool=avrdude
cadalogger.build.board=AVR_ATmega4808
cadalogger.build.mcu=atmega4808
cadalogger.build.extra_flags={build.oscillator}

# Fuses we don't need to modify in the tools menu
cadalogger.bootloader.WDTCFG=0x00
cadalogger.bootloader.TCD0CFG=0x00
cadalogger.bootloader.SYSCFG1=0x06
cadalogger.bootloader.APPEND=0x00
cadalogger.bootloader.LOCKBIT=0xC5

# menu item in megacoreX, not here
cadalogger.pinout.32pin_standard=32 pin standard
cadalogger.pinout.32pin_standard.build.variant=32pin-standard

# EEPROM
cadalogger.menu.eeprom.keep=EEPROM retained
cadalogger.menu.eeprom.keep.bootloader.eesave_bit=1
cadalogger.menu.eeprom.erase=EEPROM not retained
cadalogger.menu.eeprom.erase.bootloader.eesave_bit=0

# Reset pin --  menu item in megacoreX but not here
cadalogger.resetpin.reset=Reset
cadalogger.resetpin.reset.bootloader.SYSCFG0=0b1100100{bootloader.eesave_bit}

# Brown out detection
cadalogger.menu.BOD.2v6=BOD 2.6V
cadalogger.menu.BOD.2v6.bootloader.BODCFG=0x54
cadalogger.menu.BOD.4v3=BOD 4.3V
cadalogger.menu.BOD.4v3.bootloader.BODCFG=0xF4
[... 60 lines cut ...]
cadalogger.menu.clock.internal_1MHz.bootloader.OSCCFG=0x01
cadalogger.menu.clock.internal_1MHz.build.oscillator=
cadalogger.menu.clock.internal_1MHz.build.f_cpu=1000000L

# Bootloader -  subset of options from megacoreX
cadalogger.menu.bootloader.no_bootloader=No bootloader
cadalogger.menu.bootloader.no_bootloader.upload.maximum_size=49152
cadalogger.menu.bootloader.no_bootloader.upload.extra_params=
cadalogger.menu.bootloader.no_bootloader.build.text_section_start=.text=0x0
cadalogger.menu.bootloader.no_bootloader.build.export_merged_output=false
cadalogger.menu.bootloader.no_bootloader.bootloader.file=empty/empty.hex
cadalogger.menu.bootloader.no_bootloader.bootloader.BOOTEND=0x00

cadalogger.menu.bootloader.uart2_default=Optiboot (UART2 default pins)
cadalogger.menu.bootloader.uart2_default.upload.maximum_size=48640
cadalogger.menu.bootloader.uart2_default.upload.protocol=arduino
cadalogger.menu.bootloader.uart2_default.upload.port=UART2_DEF
cadalogger.menu.bootloader.uart2_default.upload.extra_params=
cadalogger.menu.bootloader.uart2_default.build.text_section_start=.text=0x200
cadalogger.menu.bootloader.uart2_default.build.export_merged_output=true
cadalogger.menu.bootloader.uart2_default.bootloader.file=optiboot/bootloaders/mega0/{upload.speed}/Optiboot_mega0_{upload.port}_{upload.speed}_A7.hex
cadalogger.menu.bootloader.uart2_default.bootloader.BOOTEND=0x02
cadalogger.menu.bootloader.uart2_default.bootloader.SYSCFG0=0b1100100{bootloader.eesave_bit}

```

per1234, 2023-04-30:
>```text
># menu item in megacoreX, not here
>cadalogger.pinout.32pin_standard=32 pin standard
>cadalogger.pinout.32pin_standard.build.variant=32pin-standard
>```

If the `32pin-standard` variant is in MegaCoreX then you need to [reference it](https://arduino.github.io/arduino-cli/latest/platform-specification/#variant-reference):

```text
cadalogger.pinout.32pin_standard.build.variant=MegaCoreX:32pin-standard
```

mbmorrissey, 2023-04-30:
Again, thanks! Will give it a try...

mbmorrissey, 2023-04-30:
Now I've broken it!  It won't install:
```
Downloading packages
cadalogger:megaavr@0.0.6
Installing platform cadalogger:megaavr@0.0.6
Failed to install platform: cadalogger:megaavr.
Error: 13 INTERNAL: Cannot install platform: installing platform cadalogger:megaavr@0.0.6: searching package root dir: no unique root dir in archive, found '/Users/mbm5/Library/Arduino15/tmp/package-781919629/__MACOSX' and '/Users/mbm5/Library/Arduino15/tmp/package-781919629/hardware'
```

Probably healthy to take a break before coming back and grinding my way through this.

Thanks for stepping me through things that are in the platform specification documentation - I've been pouring through it for the last couple of days, and have made some progress using it, but also I'm finding it rather terse for a non-specialist to fully understand.

Again, thanks.

mbmorrissey, 2023-04-30:
OK, I've at least corrected the last issue (an OS specific thing of adding files to zip archives).

However, the pins issue is not fixed.  

```
FQBN: cadalogger:megaavr:cadalogger
Using board 'cadalogger' from platform in folder: /Users/mbm5/Library/Arduino15/packages/cadalogger/hardware/megaavr/0.0.4
Using core 'coreX-corefiles' from platform in folder: /Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1

Detecting libraries used...
/Users/mbm5/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_MegaCoreX:AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -I/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/api/deprecated -I/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles /private/var/folders/hs/k40w7dn54r7fvvk_7ktssbfr0000gp/T/arduino/sketches/4F358E33F193002A37BE62B9660B934C/sketch/sketch_apr30a.ino.cpp -o /dev/null
Alternatives for pins_arduino.h: []
ResolveLibrary(pins_arduino.h)
In file included from /Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/Arduino.h:127:0,
                 from /private/var/folders/hs/k40w7dn54r7fvvk_7ktssbfr0000gp/T/arduino/sketches/4F358E33F193002A37BE62B9660B934C/sketch/sketch_apr30a.ino.cpp:1:
/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/UART.h:28:10: fatal error: pins_arduino.h: No such file or directory
 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.

  -> candidates: []
exit status 1

Compilation error: exit status 1
```

If you have time, can you see anything else in my boards.txt that is the issue?

```
menu.clock=Clock
menu.BOD=BOD
menu.eeprom=EEPROM
menu.bootloader=Bootloader

cadalogger.name=cadalogger
cadalogger.build.core=MegaCoreX:coreX-corefiles
cadalogger.upload.tool=avrdude
cadalogger.upload.maximum_data_size=6144
cadalogger.upload.speed=115200
cadalogger.bootloader.tool=avrdude
cadalogger.build.board=MegaCoreX:AVR_ATmega4808
cadalogger.build.mcu=atmega4808
cadalogger.build.extra_flags={build.oscillator}

# Fuses we don't need to modify in the tools menu
cadalogger.bootloader.WDTCFG=0x00
cadalogger.bootloader.TCD0CFG=0x00
cadalogger.bootloader.SYSCFG1=0x06
cadalogger.bootloader.APPEND=0x00
cadalogger.bootloader.LOCKBIT=0xC5

# menu item in megacoreX, not here
cadalogger.pinout.32pin_standard=32 pin standard
cadalogger.pinout.32pin_standard.build.variant=MegaCoreX:32pin-standard

# EEPROM
cadalogger.menu.eeprom.keep=EEPROM retained
cadalogger.menu.eeprom.keep.bootloader.eesave_bit=1
cadalogger.menu.eeprom.erase=EEPROM not retained
cadalogger.menu.eeprom.erase.bootloader.eesave_bit=0

# Reset pin --  menu item in megacoreX but not here
cadalogger.resetpin.reset=Reset
cadalogger.resetpin.reset.bootloader.SYSCFG0=0b1100100{bootloader.eesave_bit}

# Brown out detection
cadalogger.menu.BOD.2v6=BOD 2.6V
cadalogger.menu.BOD.2v6.bootloader.BODCFG=0x54
cadalogger.menu.BOD.4v3=BOD 4.3V
cadalogger.menu.BOD.4v3.bootloader.BODCFG=0xF4
[... 59 lines cut ...]
cadalogger.menu.clock.internal_1MHz.upload.speed=115200
cadalogger.menu.clock.internal_1MHz.bootloader.OSCCFG=0x01
cadalogger.menu.clock.internal_1MHz.build.oscillator=
cadalogger.menu.clock.internal_1MHz.build.f_cpu=1000000L

# Bootloader -  subset of options from megacoreX
cadalogger.menu.bootloader.no_bootloader=No bootloader
cadalogger.menu.bootloader.no_bootloader.upload.maximum_size=49152
cadalogger.menu.bootloader.no_bootloader.upload.extra_params=
cadalogger.menu.bootloader.no_bootloader.build.text_section_start=.text=0x0
cadalogger.menu.bootloader.no_bootloader.build.export_merged_output=false
cadalogger.menu.bootloader.no_bootloader.bootloader.file=empty/empty.hex
cadalogger.menu.bootloader.no_bootloader.bootloader.BOOTEND=0x00

cadalogger.menu.bootloader.uart2_default=Optiboot (UART2 default pins)
cadalogger.menu.bootloader.uart2_default.upload.maximum_size=48640
cadalogger.menu.bootloader.uart2_default.upload.protocol=arduino
cadalogger.menu.bootloader.uart2_default.upload.port=UART2_DEF
cadalogger.menu.bootloader.uart2_default.upload.extra_params=
cadalogger.menu.bootloader.uart2_default.build.text_section_start=.text=0x200
cadalogger.menu.bootloader.uart2_default.build.export_merged_output=true
cadalogger.menu.bootloader.uart2_default.bootloader.file=optiboot/bootloaders/mega0/{upload.speed}/Optiboot_mega0_{upload.port}_{upload.speed}_A7.hex
cadalogger.menu.bootloader.uart2_default.bootloader.BOOTEND=0x02
cadalogger.menu.bootloader.uart2_default.bootloader.SYSCFG0=0b1100100{bootloader.eesave_bit}
```

mbmorrissey, 2023-05-01:
Hi again.  I'm still a bit stuck.  But the issue that I opened is solved, so I'll close this thread, and if I can't figure it out with another day of trial and error, I'll open a new thread more directly about what's outstanding.

Thanks so much for the insightful help yesterday.

closed 2023-05-01 by mbmorrissey