MCUdude/MegaCoreX #177: megacorex referencing in custom board definition

mbmorrissey, 2023-05-02:
Hi,

I got some great help the other day and made some good progress in making a custom board definition.  I still don't fully understand how to set up my boards.txt file.

I use the 32pin standard Atmega4808 pin configuration and am trying to specify it like this
```
cadalogger.pinout.32pin_standard.build.variant=MegaCoreX:32pin-standard
```
however, I get an error on compiling a simple sketch of:
```
/Users/mbm5/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=atmega4808 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_MegaCoreX:AVR_ATmega4808 -DARDUINO_ARCH_MEGAAVR -I/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/api/deprecated -I/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles /private/var/folders/hs/k40w7dn54r7fvvk_7ktssbfr0000gp/T/arduino/sketches/955679A75F30D53F66BC7C66FC017475/sketch/set_time.ino.cpp -o /dev/null
In file included from /Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/Arduino.h:127:0,
                 from /private/var/folders/hs/k40w7dn54r7fvvk_7ktssbfr0000gp/T/arduino/sketches/955679A75F30D53F66BC7C66FC017475/sketch/set_time.ino.cpp:1:
/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/UART.h:28:10: fatal error: pins_arduino.h: No such file or directory
 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
Alternatives for pins_arduino.h: []
ResolveLibrary(pins_arduino.h)
  -> candidates: []

exit status 1

Compilation error: exit status 1
```

This seems to be despite the fact that I have at least (with help the other day) made some progress in referenceing MegaCoreX and the megaavr architecture, as I think evidenced by this message at the start of attempting to compile:
```
FQBN: cadalogger:megaavr:cadalogger
Using board 'cadalogger' from platform in folder: /Users/mbm5/Library/Arduino15/packages/cadalogger/hardware/megaavr/0.0.8
Using core 'coreX-corefiles' from platform in folder: /Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1
```

Any points massively appreciated!  I've been trying to understand the documentation at [https://arduino.github.io/arduino-cli/0.32/](url), and I'm sure everything I need to know is there, but I'm finding it very terse for a first-timer!

Thanks,

Michael

MCUdude, 2023-05-02:
Sorry, but I've never used arduino-cli, so I don't now how I can help here. @per1234 seems to know a lot, and he was the one that helped your in an earlier.

Personally I use PlatformIO for larger project where Arduino IDE isn't suitable. And MegaCoreX works very, very well with PlatformIO.

mbmorrissey, 2023-05-02:
Hi - thanks very much having a look at it nonetheless.  If its OK I'll leave the issue open for a couple of days before closing in case anyone happens upon it who can see where I'm going wrong?

Again, thanks also to @per1234 for the help the other day.  And also for having raised the point about the value of an introductory tutorial!

Michael

per1234, 2023-05-02:
>```text
>/Users/mbm5/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.1.1/cores/coreX-corefiles/UART.h:28:10: fatal error: pins_arduino.h: No such file or directory
> #include "pins_arduino.h"
>          ^~~~~~~~~~~~~~~~
>```
Change this line:

```text
cadalogger.pinout.32pin_standard.build.variant=MegaCoreX:32pin-standard
```

To this:

```text
cadalogger.build.variant=MegaCoreX:32pin-standard
```

The problem is your previous configuration defined the `build.variant` property via a [custom board option](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options) with the menu ID `pinout`. However, there is no such menu because you didn't define a `menu.pinout` property in `boards.txt` so that property definition can't take effect.

Alternatively, you could add the `menu.pinout` property definition to `boards.txt`, but since you only have one option in the menu (`32pin_standard`) there is no point in having a menu.

>I've never used arduino-cli, so I don't now how I can help here.

The tool is not relevant. The same Arduino boards platform framework is used by Arduino IDE 1.x, Arduino IDE 2.x, Arduino CLI, and even Arduino Cloud (though less interesting to this discussion since it is not possible to install arbitrary 3rd party platforms in Arduino Cloud).

The [Arduino Platform Specification](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options) is hosted in the [`arduino/arduino-cli`](https://github.com/arduino/arduino-cli) repository because the platform handling code is in that repository, but all the other tools use Arduino CLI under the hood so the information in the specification applies to all of them.

mbmorrissey, 2023-05-03:
Thanks @per1234 !
I think I'm getting closer.

Can I please ask you guys about the culture of issues/help requests? Am I posting these questions in the wrong place?  Are 'issues' sections of github repositories for this kind of thing, or more narrowly for bug reports?  I'm now understanding that my questions may be less about MegaCoreX, and more appropriate to direct to the issues section of the arduino-cli github?  I'm guessing that the main arduino forums won't necessarily have a lot of expertise for this kind of thing?

Anyway, I now have an issue with (but happy to close this issue and re-post more approriate depending on your advice for above):
```
chmod: /Users/mbm5/Library/Arduino15/packages/cadalogger/hardware/megaavr/0.0.2/scripts/create_disassembler_listing.sh: No such file or directory

```
I guess my boards.txt file needs to point somehow to /MegaCoreX/megaavr/scripts/ .  But as before, pouring over the platform specification documentation leaves not much wiser.  I tried a few variants on the theme of adding `cadalogger.build.system.path = MegaCoreX:megaavr`, but honestly that's grasping at straws.

per1234, 2023-05-03:
>Can I please ask you guys about the culture of issues/help requests?

As regarding this repository, that is for MCUdude to define. I'm only an interested party who makes some effort to monitor the activity here and make some infrequent trivial contributions.

>more appropriate to direct to the issues section of the arduino-cli github?

As one of the maintainers of the `arduino/arduino-cli` repository, I can authoritatively answer the question as it applies to that repository: The support requests you are making via issues here in the `MCUdude/MegaCoreX` repository would not be considered appropriate in the `arduino/arduino-cli` repository. We are strict about using issues only for formal bug reports and feature requests about the content hosted in that repository.

We redirect any support requests and general discussion to the Arduino Forum:

https://forum.arduino.cc/

mbmorrissey, 2023-05-03:
Thanks @per1234 - I'll try posting the current issue to the arduino forum.  I see that @MCUdude does answer a certain amount of troubleshooting via the 'issues' here, but I didn't/don't know how much that is imposing on his goodwill vs how it is supposed to work.

I'll close this now.  Thanks for the help.

closed 2023-05-03 by mbmorrissey