MCUdude/MegaCoreX #180: Fast ADC  (AnalogReadFast)

chsumanbabu, 2023-07-02:
Hi
I want to increase the ADC sampling rate. I have tried the below mentioned ADC Prescaler code in wiring.c file. but it was shown any improvement in ADC sampling rate.
Please suggest.

Actual 
"#elif (F_CPU >= 16000000L) // 16 MHz / 128 = 125 kHz
  ADC0.CTRLC |= ADC_PRESC_DIV128_gc; "

Modified to
"#elif (F_CPU >= 16000000L) // 16 MHz / 16= 1MHz
  ADC0.CTRLC |= ADC_PRESC_DIV16_gc; "

chsumanbabu, 2023-07-02:
Hi
I want to increase the ADC sampling rate. I have tried the below mentioned ADC Prescaler code in wiring.c file. but it was not showing any improvement in ADC sampling rate.
Please suggest.

Actual
"#elif (F_CPU >= 16000000L) // 16 MHz / 128 = 125 kHz
ADC0.CTRLC |= ADC_PRESC_DIV128_gc; "

Modified to
"#elif (F_CPU >= 16000000L) // 16 MHz / 16= 1MHz
ADC0.CTRLC |= ADC_PRESC_DIV16_gc; "

MCUdude, 2026-10-13:
Sorry, a faster ADC sample rate is not currently supported by the `analogRead()` function. However, you're free to implement this yourself.

closed 2026-10-13 by MCUdude