MCUdude/MegaCoreX #186: Optiboot bootloader does not work on UART2

i-g-g-y, 2023-09-16:
Config and programmed via UPDI and bootloader does not respond on pin 20, 21 for Atmega4808, TQFP32.

[image]

i-g-g-y, 2023-09-17:
This should be moved to discussion, probably HW issue.

MCUdude, 2023-09-18:
Well, you didn't provide any error messages, so I don't know what's going on.

I designed a board for 32-pin parts when I first developed this core, and the on-board USB connector is connected to UART2 default pins (PF0 and PF1) through a USB to serial adapter. And this board works perfectly fine with Optiboot. So check your hardware. I'm sure that's where the problem is.

[image]

[image]

i-g-g-y, 2023-09-18:
I am also 99% sure it is HW related, namely I think the reset circuit does not work properly, thus the bootloader is not even entered.

Thanks for feedback!