MCUdude/MegaCoreX #187: fork/exec C:\Python27\Lib\cmd.py: %1 is not a valid Win32 application

AGM1968, 2023-10-09:
Hello, 

After installing MegaCoreX 1.1.2, am having difficulty compiling for arduino nano every 4809, on an old 32bit win 7, arduino IDE 1.8.19.

I cant seem to find where/if the problem might resides with MegaCoreX.

Am aware that similar problem was noted with older XP IDE compatibility and esp8266 python2 - 3 tools.

Appreciate any pointers should this be related to MegaCoreX .... 

```
cmd /C echo. && "C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_708209/4809_hardware_serial_test.ino.elf" > "C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_708209/4809_hardware_serial_test.ino_atmega4809_16000000L.lst"

```

ISSUE...

```

fork/exec C:\Python27\Lib\cmd.py: %1 is not a valid Win32 application.
Error compiling for board ATmega4809.

```

MCUdude, 2023-10-09:
I have no idea why you're getting a Python error. MegaCoreX doesn't install or utilize Python at all.

MegaCoreX uses the Windows command line `cmd.exe` to copy files, so maybe there's an issue with `cmd` or `cmd.exe`?
Usually, @per1234 has been really good at diagonizing issues like this, related to the IDE.

AGM1968, 2023-10-09:
Thanks for quick reply, have been completely stumped as to where python has decided to 'intervene'. Appreciated

AGM1968, 2023-10-14:
Fixed internally, hence unrelated and closing

closed 2023-10-14 by AGM1968