MCUdude/MegaCoreX #188: Atmega4809 bootloader

ITstreet1, 2023-11-04:
How to burn the bootloader to an Atmega4809?

I have a design on which I can upload a sketch using jtag2updi. I wonder if I can add a UART and use it just like Uno. My guess is to have a bootloader on it first.
Can it be done using this programmer, or it needs SPI pins as Uno? What settings should I need to do, if any?

MCUdude, 2023-11-04:
> I wonder if I can add a UART and use it just like Uno. My guess is to have a bootloader on it first.

Yes, that would require a bootloader.

The procedure is the same as with all AVR based Arduinos. Connect your programmer, select the preferred bootloader option and click "Burn bootloader". You can use the jtag2updi for this.

ITstreet1, 2023-11-04:
Works!!!

Huge thank you, for your time, and your work for the community.

Dejan

closed 2023-11-04 by ITstreet1