MCUdude/MegaCoreX #189: ATmega4809 weird behavior

ITstreet1, 2023-12-25:
Hi,

I made a few boards using the ATmega4809 (even 3208), and everything went fine. I burned the bootloader easily with a jtag2updi programmer. The onboard bridge was usually the CP2102, on UART0. Zero problems.

Now I have a design with pretty much a barebone setup and I have a problem.
Now I use the CH340C and 4809. I burned the bootloader successfully. First, I uploaded the Blink sketch. No problems. Second, I uploaded the same sketch, but with a smaller delay. The sketch couldn't be uploaded. It spits out:

> Arduino: 1.8.19 (Windows 10), Board: "ATmega4809, 48 pin standard, EEPROM retained, Reset, BOD 2.6V, Internal 16 MHz, Optiboot (UART0 default pins)"
> 
> Sketch uses 752 bytes (1%) of program storage space. Maximum is 48640 bytes.
> 
> Global variables use 4 bytes (0%) of dynamic memory, leaving 6140 bytes for local variables. Maximum is 6144 bytes.
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 1 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 2 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 3 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 4 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 5 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 6 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 7 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 8 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 9 of 10: not in sync: resp=0x00
> 
> avrdude error: programmer is not responding
> 
> avrdude warning: attempt 10 of 10: not in sync: resp=0x00
> 
> avrdude error: unable to open programmer arduino on port COM13
> 
> avrdude error: unable to open programmer arduino on port COM13
> 
> 
> 
> 
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
> 

It turns out I have a problem with a UART bridge. I changed both the CH340C and the 4809. Nothing. I burned the bootloader again. It works. One sketch, ok. The second sketch, it doesn't. And so on...

This is the schematic:
[image]

I soldered only the components around the 4809 and the CH340C. I haven't soldered any LED or a button. Only an LED on D36.

Any idea what is wrong here?

rocketscream, 2023-12-25:
Try pressing the reset button manually after pressing upload. If that
works, the auto reset circuitry is not working. Check whether capacitor is
correct value or the DTR connection is proper.

ITstreet1, 2023-12-25:
LOL. I somehow soldered the wrong cap value on DTR. It is 0603, damn small.

closed 2023-12-25 by ITstreet1

SpenceKonde, 2023-12-27:
How'd you even read the value of smd cap? None of mine have any indication on them of their value whatsoever. When you put caps back into the wrong bin (this happened to me - a whole new load of 1uF's into 2-300 0.1's. They looked the same...) there is only one thing to do. Throw the capacitors away, wipe the parts and solder paste off the boards you were assembling, and try again when you order of new caps has arrived. 
Cost, ignoring time: $3
Cost, counting time: hundreds to thousands.

Curious - what was the value of the cap that failed to work? I am under the impression that an exceptionally wide range of values will "work" well enough.

ITstreet1, 2023-12-27:
I have no idea. Didn't even bother to measure it. I just noticed that the color of the soldered cap is not the one on the other 100nF and not the same size. And, of course, I throw it away. 

I have never experienced this failure. But, you learn something new every day.