MCUdude/MegaCoreX #192: Modbus Compatibility Issues.

treepoint2, 2024-02-12:
I am currently trying to use the mega core with ArduinoModBus library, but when I go to build the project for the ATmega4809 I end up with a couple errors. Any chance I could get some help? 

Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp: In function 'int modbus_tcp_accept(modbus_t*, arduino::Client*)':
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp:772:23: error: cannot convert 'arduino::Client*' to 'Client*' in assignment
     ctx_tcp->client = client;
                       ^~~~~~
In file included from Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp:82:0:
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.h:19:7: note: class type 'arduino::Client' is incomplete
 class Client;
       ^~~~~~
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp: In function 'modbus_t* modbus_new_tcp(arduino::Client*, arduino::IPAddress, int)':
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp:925:70: error: 'ip_address' has incomplete type
 modbus_t* modbus_new_tcp(arduino::Client* client, arduino::IPAddress ip_address, int port)
                                                                      ^~~~~~~~~~
In file included fromDocuments\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp:82:0:
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.h:20:7: note: forward declaration of 'class arduino::IPAddress'
 class IPAddress;
       ^~~~~~~~~
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp: In function 'modbus_t* modbus_new_tcp(arduino::Client*, arduino::IPAddress, int)':
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp:965:23: error: cannot convert 'arduino::Client*' to 'Client*' in assignment
     ctx_tcp->client = client;
                       ^~~~~~
In file included from Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.cpp:82:0:
Documents\Arduino\libraries\ArduinoModbus\src\libmodbus\modbus-tcp.h:19:7: note: class type 'arduino::Client' is incomplete
 class Client;
       ^~~~~~
Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPServer.cpp: In member function 'int ModbusTCPServer::begin(int)':
Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPServer.cpp:40:63: error: parameter 2 of 'modbus_t* modbus_new_tcp(arduino::Client*, arduino::IPAddress, int)' has incomplete type 'arduino::IPAddress'
   modbus_t* mb = modbus_new_tcp(NULL, IPAddress(0, 0, 0, 0), 0);
                                                               ^
Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPServer.cpp: In member function 'void ModbusTCPServer::accept(Client&)':
Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPServer.cpp:55:37: error: cannot convert 'Client*' to 'arduino::Client*' for argument '2' to 'int modbus_tcp_accept(modbus_t*, arduino::Client*)'
   if (modbus_tcp_accept(_mb, &client) == 0) {
                                     ^
Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp: In member function 'int ModbusTCPClient::begin(IPAddress, uint16_t)':
Documents\Arduino\libraries\ArduinoModbus\src\ModbusTCPClient.cpp:41:50: error: cannot convert 'Client*' to 'arduino::Client*' for argument '1' to 'modbus_t* modbus_new_tcp(arduino::Client*, arduino::IPAddress, int)'
   modbus_t* mb = modbus_new_tcp(_client, ip, port);
                                                  ^`

DrItanium, 2024-03-16:
It looks like the MegaCoreX does not place Client and IPAddress classes in the arduino namespace. This is the cause of the error. arduino-modbus assumes that these two classes are in the arduino namespace since api/ArduinoAPI.h exists.

treepoint2, 2024-03-25:
> It looks like the MegaCoreX does not place Client and IPAddress classes in the arduino namespace. This is the cause of the error. arduino-modbus assumes that these two classes are in the arduino namespace since api/ArduinoAPI.h exists.

I am unsure of how to fix this as I am relatively new to programing could you explain the process?

DrItanium, 2024-03-26:
It requires modification to MegaCoreX as the simplest way forward. Otherwise, you will run into issues if you ever clear your library cache in platformio or arduino. Let me see what I can do to fix this.

DrItanium, 2024-03-26:
I was able to fix the issue with minimal modifications to the MegaCoreX library. I'll make a pull request for this.

MCUdude, 2024-10-10:
Closed by #198

closed 2024-10-10 by MCUdude